sparkfun/SparkFun_ICM-20948_ArduinoLibrary #1: Will there be a chance to get DMP supported in this library?

larryyyyy, 2019-04-18:
Thank you very much to create this very good and useful lib! This helps a lot to test out the new imu sensor. Will there be a chance to get DMP supported in this library? As the sensor claims to have a better DMP performance and I really want to see how it goes. Thank you so much!

oclyke, 2019-04-18:
Hi @larryyyyy, glad you like it. Yes, we definitely want to add DMP support, but to achieve it we may need outside help. I'm going to rename this issue so that it can be the official 'DMP Support' issue. (Thanks for bringing it up!)

I've personally looked into it and found this:
1. the DMP image is easily found in Invensense example code
1. there is an 'undercover' FIFO register through which you can read/write the DMP image into the chip

Unfortunately I have had very little success in trying to figure out how to interpret the results once the DMP is enabled. Some questions are:
1. Where will the results be placed? (I suspect it will be in the main FIFO, but cannot confirm)
1. What will the results mean? (quaternions? MRPs? etc...? )
1. Can you enable virtual sensors such as the step counter? If so how do you differentiate those results from the orientation results?

I can't put as much time into this as may be needed right now, so if anyone is knowledgeable in this area or simply willing to dig a little deeper we'd certainly appreciate some input.

ericalbers, 2019-07-19:
If you need to see how to parse the DMP output, I think this repo's example.c might show you the way...
https://github.com/OneMillionDolar/libIDD

ericalbers, 2020-01-09:
I've uploaded a version with DMP working with arduino, it runs on the sparkfun esp32 board
See
https://github.com/ericalbers/ICM20948_DMP_Arduino/tree/master

larryyyyy, 2020-01-09:
> I've uploaded a version with DMP working with arduino, it runs on the sparkfun esp32 board
> See
> https://github.com/ericalbers/ICM20948_DMP_Arduino/tree/master

Thank you so much! Will test it out later!

ruffner, 2020-01-10:
@ericalbers  you are the man, this even has the body activity classification features... wow! I'm getting ~40Hz yaw/pitch/roll over i2c no prob

pidloop, 2020-01-10:
Outstanding, many thanks.

drtritmwin, 2020-11-16:
> I've uploaded a version with DMP working with arduino, it runs on the sparkfun esp32 board
> See
> https://github.com/ericalbers/ICM20948_DMP_Arduino/tree/master

Dear ericalbers, I build it with arduino IDE on my arduino uno board, but the IDE keeps saying that "Sketch too bigs". The global variable use 19113 bytes. Please help me. Thanks in advance

PaulZC, 2021-02-15:
Hi @larryyyyy / @ericalbers / All,
We have - at long last - added limited DMP support to this library. We'd be grateful if you could take a look and let us know what you find.
- You will need to manually download the code from the front page of this repo (look for the green Code icon) as the Arduino Library Manager won't pick up the changes until the next release.
- We have been working from an InvenSense Application Note "Programming Sequence for DMP Hardware Functions" which they kindly shared with us. They have asked us not to share the App Note further as it is not 100% complete. So, before you ask, sorry, no, we cannot send you a copy. We have put as much information into the code as we can - please have a look at the comments in the new Example5 and ICM_20948_DMP.h.
- Think of the new code as a toolkit or skeleton for further development. It works. Kind of. You can download the DMP into the chip, configure it, request (e.g.) quaternion orientation and extract the DMP data from the FIFO. I ran into problems when I started to convert the quaternions into Euler angles. The Yaw appeared to have a scaling error and I suspect I either haven't configured the compass "mount multiplier" correctly - or the sensors need calibrating.
- The DMP functionality is disabled by default - to save program memory and prevent backward-compatibility problems. Please see [the comments in Example5](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example5_DMP_Quat9_Orientation/Example5_DMP_Quat9_Orientation.ino#L13-L20) for further details.
- We have deliberately written our own functions to work with the DMP - @ericalbers will confirm how opaque the InvenSense example code is and our functions are not a clone of those examples.

Please let us know what you find.
Best wishes,
Paul

PaulZC, 2021-02-26:
Hi @larryyyyy / @ericalbers / All,
The new DMP code now seems to be working nicely. I figured out what I'd done wrong with the magnetometer... The quaternions now convert into well-behaved Euler angles.
I've added a few extra configuration steps based on a by-hand decode of all the I2C traffic created by @ericalbers's port of the InvenSense code. (Thanks again Eric for sharing that!) If you want all of the gory detail, scroll down to the bottom of issue #34 .
I plan to add a quick markdown summary of the things I've learned along the way, up-issue the library and then I think we can finally close this issue (after _almost_ two years!).
There were a few surprises along the way, like the DMP Gyro data is actually Gyro + Bias (6x2 bytes, not 3x2...). And each DMP packet is terminated with an undocumented two byte 'footer' (gyro count)...
All the best,
Paul

PaulZC, 2021-02-27:
Resolved in [Version 1.2.0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0)
Closing!!

closed 2021-02-27 by PaulZC

FedericoPacini, 2021-05-06:
Good afternoon,
I've read this discussion and I want to thank you about precious work.
Anyway, I want to make this question since I've seen that even using your examples (Example6_DPM and Example_7_DPM) and making some rotations to calibrate the sensors, I got however some bias (for instance, on X axes, I've a roll of 5° wheareas on Y pitch of 10° while Yaw behaves as it wants but it's not so important for my application - detecting an overturning on an electric wheelchair).

I'd like to ask if it's normal (I don't think so reading your previous discussion) and eventually how to provide manual calibration as the other IMUs allow to do.
Moreover, I've read the code and I've encountered this sentence that makes me astonished (maybe I've not understood the real meaning).
[image]

Now, considering that you calculate 
   q0 = sqrt(1 - (q1^2 + q2^2 + q^3))
Then it's obvious (by construction) that q0^2 * q1^2 + q2^2 + q3^2 = 1, so it's nonsense to specify "in case of the sum is not equal 1" and so for you it's never needed a calibration.

Thanks in advance.

PaulZC, 2021-05-07:
Hi Federico (@FedericoPacini ),

Welcome to the strange and wonderful world of the ICM-20948 DMP!

The bias offset you are seeing is not unusual. Please see issues #40, #50, #59 and #60 for more details.

For your electric wheelchair application, it might be better to use the standard accelerometer data from the sensor? I think you should be able to calculate the tip and tilt from the X and Y accelerometer values?

We do not yet understand how to manually calibrate the sensor. Please see those same issues for more details.

The quote "the sum will not add to 1" comes directly from the InvenSense DMP documentation. I see that the powers are missing in the code comment, it should say: Q0^2 + Q1^2 + Q2^2 + Q3^2 = 1. But I agree, when calculating Q0 I do assume the squares sum to 1. Until we understand the 'drift' better, that is the best I can do.

Very best wishes - and good luck with your application,
Paul

FedericoPacini, 2021-05-07:
Hi Paul,

Thank you very much for your fast reply.
"that is the best I can do".. You have done even too much!! You've done what InveSense hasn't done and you were not required to do it! So, at least for what it's worth, I want to THANK you!

I'll be reading that issues and I'll find a way.. at the end we're engineers, we always find a way to make something work ^.^

Greetings and good luck also to you!,
Federico.

PaulZC, 2021-05-07:
Thank you Federico - nice to meet you too!
Best wishes,
Paul

FedericoPacini, 2021-05-07:
I did not know this since I've joined hust few days ago.
Hope InveSense will wake up very soon from their drowsiness.

Have a nice weekend too,
Federico

PaulZC, 2021-05-27:
Hi Federico (@FedericoPacini ),

Please try version 1.2.6 of the library. I have made a small but important change to allow the DMP to work correctly. Please see issue #59 for more details. Hopefully the library will now behave as expected for you?

Best wishes,
Paul