sparkfun/SparkFun_ICM-20948_ArduinoLibrary #2: Will this driver also work with the ICM 20648?

sslupsky, 2019-05-02:
This looks interesting.  I have starting working with the ICM 20648 which I understand is similar to the ICM 20948 but without the magnetometer.  Do you know if this library will also work with the ICM 20648?

oclyke, 2019-05-03:
I can't say for sure, but I have a suspicion that you are correct. At least for the accelerometer and gyroscpe i could see it being identical. As far as features like the Digital Motion Processor, and the I2C master I don't know enough about the 20648 to say. 

As a starting point try out some of the verification functions like reading the WHOIAM register, and maybe trying to start up and get readings from the Accel/Gyro. Remember to double-check the I2C address that you are using - the 20948 is either 0x68 or 0x68 (7-bit).

If that doesn't work right away then as a last step I'd check the register lists in the two datasheets to see if the locations and functions match. If they don't then you could at least use this library as an example of how to write one for the 20648.

Please let me know if it works out!

closed 2019-05-03 by oclyke

sslupsky, 2019-05-03:
I made a mistake regarding the part number.  The part I am using is the ICM-20649.  I think there is still a lot of similarity.  The I2C address is the same.  The WHOAMI value is different.

I changed the WHOAMI address in the library to 0xE1 (this is the value for the 20649) and I get the "All is well" message indicating the initialization appears to have worked.

Interesting.  I'll keep going.

oclyke, 2019-05-04:
Neat! Though don't rely on those messages entirely. Right now the error checking is only basic - for example on initialization I think it will only error out if WHOIAM is mismatched or if there is a detectable communication problem (such as not enough bytes being transferred)