sparkfun/SparkFun_ICM-20948_ArduinoLibrary #3: Compile error when SPI_INTERFACES_COUNT 0

sslupsky, 2019-05-06:
### Subject of the issue
Compile generates an error when SPI_INTERFACES_COUNT 0

### Your workbench
Custom variants.h has no SPI ports defined.

### Steps to reproduce
Edit variants.h:

// SPI Interfaces
// --------------
#define SPI_INTERFACES_COUNT 0

### Expected behaviour
Code compiles with SPI disabled

### Actual behaviour
Sketch does not compile.

error: 'SPI' was not declared in this scope
     ICM_20948_Status_e begin( uint8_t csPin, SPIClass &spiPort = SPI );

sslupsky, 2019-05-06:
Should there be a compile directive?

#if SPI_INTERFACES_COUNT > 0
#endif

and similarly for the Wire interface, 

#if WIRE_INTERFACES_COUNT > 0
#endif

oclyke, 2019-05-06:
Interesting... yes let's fix this.

So previously I'd assumed that every board+variant combination would have the ```Wire``` and ```SPI``` interfaces defined. From your example it is clear that that will not always be the case. However that also makes me cautious to use ```SPI_INTERFACES_COUNT``` as a switch because I can imagine some Arduino cores not using that name.

In general I don't have a great idea of what (beyond the built-in standard functions like pinMode, digitalWrite, etc... ) is required from a minimal core. If you have any information on that I'd definitely like to know. 

It would be nice if we could switch on either ```SPI``` or ```Wire``` however I don't know any preprocessor directives that can detect the existence of a particular variable (seems pretty impossible since the linker could link in anything later...)

In the meantime how would it be if we used an extern declaration of SPI and Wire. That way it should compile and link as long as you don't attempt to use the undefined interface.

oclyke, 2019-05-06:
@sslupsky Can you pull this branch and tell me if the changes suit your needs? 

Branch: [Arduino Interface FWD Declarations](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/tree/arduino_interface_fwd_declarations)
Commit: cb2e346

sslupsky, 2019-05-06:
SPI_INTERFACES_COUNT is defined in variants.h.  From what I can see, I think you pretty much have to set this properly or you are going to encounter issues with conflicting resources.  SPI_INTERFACES_COUNT and WIRE_INTERFACES_COUNT are used in their respective SPI and Wire libraries to define the Wire (Wire, Wire1, Wire2, etc) and SPI (SPI, SPI1, SPI2, etc) objects.

I am not sure how wide spread the use of that define is in other architectures (AVR, ESP32, etc) as I am only familiar with the Arduino SAMD architecture at the moment.  So, you might be correct on the point about other cores.

Yesterday I added the switch and successfully compiled the library.  I'll give the forward declarations a try too.

I suppose the advantage of using the switch is that if SPI or Wire doesn't exist, there is really no need to include the code for the missing interface so you can save code space by leaving it out.

oclyke, 2019-05-06:
Yeah, being able to exclude code and save that space could be useful. Generally I see the Arduino side of this library as being geared toward working 'out of the box' for as many people as possible. For the purpose of being minimal and extensible the C-based backbone can do that. 

As a possible alternative to the forward declarations here's another possibility that doesn't use the preprocessor: split up the Arduino header file into ```_BASE.h```, ```_SPI.h```, ```_I2C.h```, and the current named file ```.h```. So in that case you'd include the normal header for default behavior, and it would in turn include both ```_SPI.h``` and ```_I2C.h``` which would each include the base header. (For most typical Arduino users with SPI and Wire)

The advanced use case would mean simply including either ```_SPI.h``` or ```_I2C.h```. That way if you don't have any I2C interfaces you won't get a compile error about ```Wire```. Of course that's not an automatic solution - requiring the user to know about the ability to do that and also appending the desired interface to the include path, however it seems completely portable.

Don't get me wrong either, I'm not strongly opposed to using some preprocessor tools such as:
```
#ifdef SPI_INTERFACES_COUNT
#if SPI_INTERFACES_COUNT  == 0
// begin function declaration without a default object
#endif
#else 
// begin function with default object
#endif
```
Which would keep the current behavior unless SPI_INTERFACES_COUNT is defined. Just taking a moment to consider other (prettier?) options first.

sslupsky, 2019-05-06:
The forward declaration seems to work fine.  I have not done much testing yet beyond initialization of the device.

There are other libraries that have this problem as well.  I ran across this same issue with the Adafruit BMP280 library today.  It was a little more difficult to deal with because there were some methods involved.  For now I simply created an SPI object with a bunch of null values.  The pointer to the sercom object cannot be null though so I just sercom0.

A solution that could work across libraries would be helpful I think for the other maintainers' libraries.

For the moment, I am using the forward declaration branch you created.  I'll think about this a little more and see if I come up with any other ideas.

oclyke, 2020-01-06:
@sslupsky

closed 2020-01-06 by oclyke