sparkfun/SparkFun_ICM-20948_ArduinoLibrary #7: Initialization of the sensor returned: Data Underflow

limgm, 2019-07-09:
Thank you for creating this useful library.

I have just bought a new set of SparkFun ICM-20948 and SparkFun Thing Plus - ESP32 WROOM, after successfully compiling and uploading the Example1_Basics, I got the error on the Serial Monitor: 

```
Initialization of the sensor returned: Data Underflow
Trying again... 
```

Did I miss out anything? 

Thank you for your help.

oclyke, 2019-07-09:
@limgm Thanks for reaching out! This question would fit better on the SparkFun forums because it is not necessarily an issue with the software. Here's a link to the [Inertial Measurement Units subforum](https://forum.sparkfun.com/viewforum.php?f=83) where you can (and should) continue this inquiry. 

Please first search that forum to see if there are similar topics, and if there is not please make a new post. When you do it would be useful to know more details about your situation such as:

1. The *exact* code you are running, 
1. How you have the Arduino IDE set up (i.e. which board definition are you using?)
1. How is your ICM-20948 breakout board configured (e.g. have you soldered any jumpers together?)
1. How is your wiring? Are you using a Qwiic cable or wiring individually? Do you have good contact?

From what I can see ("Initialization of the sensor returned: Data Underflow") I suspect I2C communications trouble. Perhaps wiring or using the wrong I2C address?

I'd be happy to keep helping you on the forums -- that way the information can be easily accessible to all SparkFun customers. If we end up finding out that this is caused by a problem with the library then we can move back here and fix it. Thanks!

closed 2019-07-09 by oclyke

limgm, 2019-07-11:
@oclyke Thank you for your suggestions, I tried to remove and reinstall the board definition for esp32 by espressif and now the sensor is able to output the data correctly.  

Thanks for your help!

EstevesDouglas, 2021-01-12:
I am having the same problem described.

Using the correct library and example code.

To use the I2C feature, do I keep the ICM-20948 breakout (original state without touching the jumpers) correct?

[image]

svenakis, 2021-01-30:
to make it work on an arduino NANO the AD0_VAL has to be 0. AD0_VAL = 1 does not work on an NANO:

#define AD0_VAL   0     // The value of the last bit of the I2C address. 
                                       // On the SparkFun 9DoF IMU breakout the default is 1, and when 
                                      // the ADR jumper is closed the value becomes 0

EstevesDouglas, 2021-02-02:
thank you very much 

I ended up also using the sparkfun esp32 thing plus

9DoF IMU Breakout // Sparkfun ESP32 Thing Plus
SLC  >>>    22 SCL
SDA  >>>  21 GPIO 21
GND >>>  GND
VIN   >>>  3v3

Just using the Qwiic cable didn't work

svenakis, 2021-02-03:
This works to read both, the accel, gyro and calculates roll pitch ... temp
You have to comment out what you need
I use it with processing to visualize this values
Have fun
Sven

// Basic demo for accelerometer readings from Adafruit ICM20948

#include <Adafruit_ICM20X.h>
#include <Adafruit_ICM20948.h>
#include <Adafruit_Sensor.h>
#include <Wire.h>

Adafruit_ICM20948 icm;
uint16_t measurement_delay_us = 65535; // Delay between measurements for testing
// For SPI mode, we need a CS pin
#define ICM_CS 10
// For software-SPI mode we need SCK/MOSI/MISO pins
#define ICM_SCK 13
#define ICM_MISO 12
#define ICM_MOSI 11

float GyroErrorX, GyroErrorY, GyroErrorZ;
float AccErrorX, AccErrorY;
float GyroX, GyroY, GyroZ;
float AccX, AccY, AccZ;
float accAngleX, accAngleY;
unsigned long elapsedTime, currentTime, previousTime;
float gyroAngleX, gyroAngleY, gyroAngleZ;
float roll, pitch, yaw;

int c = 0;

void setup(void) {

  Serial.begin(9600);
  //Serial.begin(115200);
  while (!Serial)
    delay(10); // will pause Zero, Leonardo, etc until serial console opens

    //Serial.println("Adafruit ICM20948 test!");

  // Try to initialize!
  // i2c address of cjmcu-20948 board 9 axis accelerometer gyr temp 
  if (!icm.begin_I2C(0x68)) {
    // if (!icm.begin_SPI(ICM_CS)) {
    // if (!icm.begin_SPI(ICM_CS, ICM_SCK, ICM_MISO, ICM_MOSI)) {

    Serial.println("Failed to find ICM20948 chip");
    while (1) {
      delay(10);
      
    }
  }
  //Serial.println("ICM20948 Found!");
  icm.setAccelRange(ICM20948_ACCEL_RANGE_16_G);
  //Serial.print("Accelerometer range set to: ");
  switch (icm.getAccelRange()) {
  case ICM20948_ACCEL_RANGE_2_G:
    //Serial.println("+-2G");
    break;
  case ICM20948_ACCEL_RANGE_4_G:
    //Serial.println("+-4G");
    break;
  case ICM20948_ACCEL_RANGE_8_G:
    //Serial.println("+-8G");
    break;
  case ICM20948_ACCEL_RANGE_16_G:
    //Serial.println("+-16G");
    break;
[... 314 lines cut ...]
  delay(100);

  //  Serial.print(temp.temperature);
  //
  //  Serial.print(",");
  //
  //  Serial.print(accel.acceleration.x);
  //  Serial.print(","); Serial.print(accel.acceleration.y);
  //  Serial.print(","); Serial.print(accel.acceleration.z);
  //
  //  Serial.print(",");
  //  Serial.print(gyro.gyro.x);
  //  Serial.print(","); Serial.print(gyro.gyro.y);
  //  Serial.print(","); Serial.print(gyro.gyro.z);
  //
  //  Serial.print(",");
  //  Serial.print(mag.magnetic.x);
  //  Serial.print(","); Serial.print(mag.magnetic.y);
  //  Serial.print(","); Serial.print(mag.magnetic.z);

  //  Serial.println();
  //
  //  delayMicroseconds(measurement_delay_us);
}
*/

jonra1993, 2021-04-30:
> the board definition for esp32 by espressif

Hello, @limgm I am facing the same similar issue on esp32. What you mean with "remove and reinstall the board definition for esp32"

tvp14ss, 2021-08-03:
Hey, @jonra1993. Did you solve your problem? I am also having a similar issue on esp32.

Fabryz, 2022-11-24:
Same issue on Lilygo T-Display S3 (ESP32), I guess it will be just a cry in the void...

PaulZC, 2022-11-24:
Please give us a few more details... Are you using I2C or SPI? Which 20948 board are you using? If you are using I2C, it could be pull-up related. The ICM-20948 works really well with ESP32, but you do need pull-ups on the I2C bus. And I don't see any on the T-Display schematic...

Fabryz, 2022-11-24:
@PaulZC 

I have a ["SparkFun 9DoF IMU Breakout - ICM-20948"](https://www.sparkfun.com/products/15335) connected to the device via Qwiic on I2C, I uploaded [the basic example code](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example1_Basics/Example1_Basics.ino) on my device, but it keeps giving me errors:
```
22:49:09.008 -> ICM_20948::checkID: ICM_20948_check_id returned: Data Underflow
22:49:09.008 -> ICM_20948::startupDefault: checkID returned: Data Underflow
22:49:09.008 -> ICM_20948_I2C::begin: startupDefault returned: Data Underflow
22:49:09.008 -> Initialization of the sensor returned: Data Underflow
22:49:09.008 -> Trying again...
```

Some info about the T-Display S3:
* https://cdn.shopifycdn.net/s/files/1/0617/7190/7253/files/Lilygo-T-display_13_600x600.jpg?v=1657875106
* https://github.com/Xinyuan-LilyGO/T-Display-S3
* https://www.lilygo.cc/products/t-display-s3

PaulZC, 2022-11-24:
OK. Thanks.

This is probably an I2C pin definition issue.

The ESP32S3 module defines the I2C pins as 8 and 9:

https://github.com/espressif/arduino-esp32/blob/93a7f4e0db2c189836f85c0a75a7fcd8fad838cd/variants/esp32s3/pins_arduino.h#L30

But LilyGO seem to use 17 and 18:

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/876badcbff8e5c9a09d3bb7672334c930b833247/example/factory/pin_config.h#L47

You will - I think - need to define a custom Wire port using the correct pin numbers.

I hope this helps,
Paul

PaulZC, 2022-11-25:
Hi Fabrizio (@Fabryz ),

It might be as simple as changing line 43 from ```WIRE_PORT.begin();``` to ```WIRE_PORT.begin(18, 17, (uint32_t)100000);``` . Please let me know if this works.

Best wishes,
Paul

Fabryz, 2022-11-25:
Hi @PaulZC thanks for your help, I've just learned that I can find some useful information just by giving a look at the headers files inside the libraries, this will be helpful to debug stuff in the future.

I've changed that line with the initialization on SDA and SCL pins, but nothing changed, I still have the same error.

PaulZC, 2022-11-25:
Hi @Fabryz ,

OK. Thanks for the feedback.

Which board definition are you using? ESP32S3 Dev Module?

A logic analyzer would help you debug this. Just to make sure you see correct activity on the SDA and SCL pins.

Do you have a different I2C device you can try? Again that would help confirm you are using the correct port / pin definitions.

Best wishes,
Paul

Fabryz, 2022-11-25:
@PaulZC Yes, I'm using "ESP32S3 Dev Module" with all the settings suggested by the official docs for this device.

I don't have a logic analyzer, I'm just a beginner on this.

 I also have a "SparkFun Qwiic Haptic Driver - DA7280", again I've already used its [basic example](https://github.com/sparkfun/SparkFun_Qwiic_Haptic_Driver_DA7280_Arduino_Library/blob/main/examples/I2C_mode/I2C_mode.ino), I've just tried to set the pins for SDA and SCL, here actually I don't even get any output on the serial monitor.

I've created a thread on sparkfun's forums for both sensors, but so far there were no answers, I will update with your suggestions, just to keep track also there for the posterity
https://forum.sparkfun.com/viewtopic.php?f=83&t=58663&p=237235#p237235

Thanks for your time

Fabryz, 2022-11-26:
@PaulZC I've managed to find the solution!

Thanks to this guy https://github.com/VolosR/TemperatureTDisplay/blob/main/TemperatureTDisplay.ino#L43

The correct settings are:

```WIRE_PORT.begin(43, 44);```

But why? As we saw here: https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/7#issuecomment-1326882914
They're actually the TX and RX pins for the ESP32 :/

PaulZC, 2022-11-26:
Hi Fabrizio (@Fabryz ),

Excellent detective work!

This does all make sense... On the T-Display pinout diagram, they do show 43 and 44 next to the Qwiic connector:

[image]

And looking closely at the schematic, they have used U0RXD and U0TXD:

[image]

It's puzzling as I would have expected them to connect using 18 and 17 (I2C SDA and SCL) like they do on the breakout pins:

[image]

Anyway, problem solved! Glad that's working for you,
Paul