sparkfun/SparkFun_ICM-20948_ArduinoLibrary #8: Accelerometer full scale range issue

shridharkulkarni91, 2019-10-25:
Hi,

I tried to change the full scale settings of the accelerometer in the arduino Example2_Advanced code and it did not seem to work.
I am using ICM20948 with SPI communication.
 
myFSS.a = gpm8;         // (ICM_20948_ACCEL_CONFIG_FS_SEL_e)
                          // gpm2
                          // gpm4
                          // gpm8
                          // gpm16
This is all I need to do to change the settings to +-8G, is that correct?
(In the example code this was myFSS.a = gpm2);

The output raw values for the vertically pointed axis (z-axis in my case) shows +16384 value for all the 4 settings (gpm2, gpm4, gmp8 and gmp16). 

If the 4G, 8G, 16G settings have been configured successfully, then the expected output for z-axis raw values is:
approx 8192 for +-4G setting 
approx 4096 for +-8G setting
approx 2048 for +-16G setting

Have you observed the same problem or am I doing something wrong here?

oclyke, 2019-10-28:
@shridharkulkarni91 - I want to look into this but can't get to it very soon. Just as a sanity check does #6 have anything to do with it? Are you up-to-date on the version?

shridharkulkarni91, 2019-11-01:
Solved it. It was the older version issue. Sorry about that, I downloaded the library earlier and didn't update it for some reason.
It works fine with all the accelerometer and gyroscope ranges now.

oclyke, 2019-11-01:
Great! Thanks for reporting anyway :D

pietertry, 2021-12-07:
Hello, 

i am having the same issue of not being able to change the Full Scale value of the accelerometer and gyro. The z-axis acceleration is always around ~16500 despite dpm2, 4, 6, and 16 settings. I am using the advanced example code. 

// Set full scale ranges for both acc and gyr
  ICM_20948_fss_t myFSS; // This uses a "Full Scale Settings" structure that can contain values for all configurable sensors

  myFSS.a = gpm2; // (ICM_20948_ACCEL_CONFIG_FS_SEL_e)
                  // gpm2
                  // gpm4
                  // gpm8
                  // gpm16

  myFSS.g = dps250; // (ICM_20948_GYRO_CONFIG_1_FS_SEL_e)
                    // dps250
                    // dps500
                    // dps1000
                    // dps2000

I have the latest version and the solution from #6 is already implemented in my updated version. 

I have debug mode activated but there are no error messages. I am also able to change the sampling rate and the DLPF filter but the Full Scale value won't change regardless of setting. 

My setup uses a teensy4.1. 

I appreciate your help.

PaulZC, 2021-12-07:
Hello @pietertry ,

Can you please confirm that you are calling ```myICM.setFullScale((ICM_20948_Internal_Acc | ICM_20948_Internal_Gyr), myFSS);``` immediately after those lines?

Have you made any other modifications to the example code?

I do not have a Teensy 4.1 so I cannot replicate your setup, but I will try and help if I can.

Best wishes,
Paul

pietertry, 2021-12-08:
Hello PaulZC, 

thanks for your quick reply. 

Yes, I am calling the "setFullScale" functions. I'll post the setup and loop below. 

I only added a small code section in "printRawAGMT" of the "advanced" example sketch, that uses micros() to calculate the sampling rate. 

Further info to setup: I am using VSCode with PlatformIO as my IDE. 

Thanks for your help. 

```cpp

#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include "ICM_20948.h" // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

#define USE_SPI // Uncomment this to use SPI

#define SERIAL_PORT Serial

#define SPI_PORT SPI     // Your desired SPI port.       Used only when "USE_SPI" is defined
#define SPI_FREQ 4000000 // You can override the default SPI frequency
#define CS_PIN 10        // Which pin you connect CS to. Used only when "USE_SPI" is defined

#define WIRE_PORT Wire // Your desired Wire port.      Used when "USE_SPI" is not defined
#define AD0_VAL 1      // The value of the last bit of the I2C address.                \
                       // On the SparkFun 9DoF IMU breakout the default is 1, and when \
                       // the ADR jumper is closed the value becomes 0

#ifdef USE_SPI
ICM_20948_SPI myICM; // If using SPI create an ICM_20948_SPI object
#else
ICM_20948_I2C myICM; // Otherwise create an ICM_20948_I2C object
#endif
//##############################
//#######  MY VARIABLES  #######
//##############################

int Z = 0;
uint32_t lastValue = 0;
void printPaddedInt16b(int16_t val);
void printFormattedFloat(float val, uint8_t leading, uint8_t decimals);
void printScaledAGMT(ICM_20948_SPI *sensor);
void printRawAGMT(ICM_20948_AGMT_t agmt);

//#######################
//#######  SETUP  #######
//#######################

void setup()
{

    SERIAL_PORT.begin(4608000);
    while (!SERIAL_PORT)
    {
    };

#ifdef USE_SPI
    SPI_PORT.begin();
#else
    WIRE_PORT.begin();
    WIRE_PORT.setClock(400000);
#endif

    //myICM.enableDebugging(SERIAL_PORT); // Uncomment this line to enable helpful debug messages on Serial

    bool initialized = false;
[... 165 lines cut ...]
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.acc.axes.y);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.acc.axes.z);
    SERIAL_PORT.print(" ], Gyr [ ");
    SERIAL_PORT.print(agmt.fss.g);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.gyr.axes.x);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.gyr.axes.y);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.gyr.axes.z);
    SERIAL_PORT.print(" ], Mag [ ");
    printPaddedInt16b(agmt.mag.axes.x);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.mag.axes.y);
    SERIAL_PORT.print(", ");
    printPaddedInt16b(agmt.mag.axes.z);
    SERIAL_PORT.print(" ], Tmp [ ");
    printPaddedInt16b(agmt.tmp.val);
    SERIAL_PORT.print(" ]");
    SERIAL_PORT.println();
}

```

PaulZC, 2021-12-08:
Hello @pietertry ,

Wow. Fascinating. Great find! I see the same issue on Teensy 3.2 using SPI.

I think the root cause is that the Teensy is much faster than most devices...

I can correct the issue if I insert a ```delay(1);``` after the ```myICM.setSampleMode``` , before the ```myICM.setFullScale```. 

My guess is that the ICM20948 is still doing some internal processing / configuration after the setSampleMode and misses the first part of the setFullScale (which sets the accelerometer scale). The gyro scale - which is set immediately afterwards - was being set correctly without the delay.

Please let me know if this corrects your issue.

Best wishes,
Paul

pietertry, 2021-12-09:
Hello @PaulZC , 

your fix with the delay function worked for me! I added delay(1) after every set function, just to be sure. 

Thanks a lot for the quick help! 

best wishes 
Pieter

PaulZC, 2021-12-09:
Hi Pieter (@pietertry ),

Thank you for the update. I am glad that is working for you. I will add the delay into the library example.

Best wishes,
Paul

closed 2021-12-09 by PaulZC

LukeDeWaal, 2023-12-28:
I am having a similar issue using the Beaglebone Black board, but it seems no matter the delay, the full-scale register refuses to be set, even with multiple tries. I analyzed the I2C bus with a signal analyzer and the signals are all correct:

[image]

The register simply resets itself no matter what.