sparkfun/SparkFun_ICM-20948_ArduinoLibrary #12: FIFO enabled?

metanav, 2020-01-31:
I read the ICM 20948 code in the src directory. I can't see if FIFO is enabled by default. Also, I could not find any code that do so.

oclyke, 2020-01-31:
@metanav No currently the FIFO is unused (by default) and we've not provided any convenience functions to use it. 

We would welcome a PR of this functionality!

In the meantime you can always modify the device registers directly with 

``` c
ICM_20948_execute_r(ICM_20948_Device_t *pdev, uint8_t regaddr, uint8_t *pdata, uint32_t len);
ICM_20948_execute_w(ICM_20948_Device_t *pdev, uint8_t regaddr, uint8_t *pdata, uint32_t len);
```

That should allow you to enable / use the FIFO as desired

PaulZC, 2021-02-27:
Hi @metanav ,
[Version 1.2.0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0) now makes use of the FIFO to support the DMP. You can access DMP and raw sensor data through the new functions we've added to support the DMP. Please see the new examples for further details.
Closing...
Best wishes,
Paul

closed 2021-02-27 by PaulZC