sparkfun/SparkFun_ICM-20948_ArduinoLibrary #13: Quaternion outputs?

da40flyer, 2020-02-10:
I understand that this chip will output the game rotation vectors as quaternions.  Is it possible to extract them using this library?  I looked through the examples, but did not see any mention of quaternions.

oclyke, 2020-02-10:
This library does not support that directly - however you should be able to use this library to perform all necessary interactions with the chip. The issue here is that the DMP operation is outlined in source code from InvenSense that resides behind an NDA wall. We can't demonstrate how to access that functionality. Instead you will have to refer to the resources mentioned in #1

da40flyer, 2020-02-11:
Thank you for your reply and please pardon my ignorance.  I am relatively new to Arduino and coding.  

I am unsure I understand what you mean.  Are you saying that InvenSense does not want you showing your customers how to use their chip with Arduino, beyond reading basic gravity/gyro/mag?

oclyke, 2020-02-11:
Yes, basically. They have their own example code / projects that others have been able to decipher. 

[InvenSense Developer's Corner](https://www.invensense.com/developers/software-downloads/) - Here you could gain access to their examples for the ICM-20948 which include DMP (which includes quternion output)

Here's [Eric Alber's DMP Example that runs on ESP32 boards](https://github.com/ericalbers/ICM20948_DMP_Arduino/tree/master) - (maybe Arduino compatible?)

da40flyer, 2020-02-11:
Well that is certainly an odd business model...

I did look at the link you provided and at Eric's stuff, but I am just not proficient enough to be able to understand the code nor be able to pick out what is and is not needed.

Thank you again for the information.  I do appreciate it.

oclyke, 2020-02-13:
My pleasure - yeah it is a lot to process. I would prefer if their examples were more stripped-down. As it stands it would take a lot of effort and dedication to pull out just the parts you need. I personally haven't been able to justify that amount of work yet - and I think that most people wont. Eric is the exception and we definitely appreciate all that work!

closed 2020-02-13 by oclyke