sparkfun/SparkFun_ICM-20948_ArduinoLibrary #15: CMake support required

Sterpa, 2020-03-25:
Could you please add CMakeLists.txt to the library to create compatibility with СMake projects?
And also for compatibility with ([ESP-IDF4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) + [arduino-esp32-4.0](https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0) as a component) enviroment combination.

I tried the following CMakeLists.txt code and it performed very well:
```
idf_component_register(SRCS "src/ICM_20948.cpp" 
                            "src/util/ICM_20948_C.c"
                       INCLUDE_DIRS "src" 
                                    "src/util"
                       REQUIRES "arduino")

component_compile_options(-std=c++17)
```

PaulZC, 2021-02-08:
Hi @Sterpa ,
Apologies for the very slow response!
Is this still an issue for you?
Could you add the file, test it and send us a Pull Request?
Best wishes,
Paul

PaulZC, 2021-02-27:
Closing... Please feel free to submit a Pull Request if you would like us to merge the CMake file.
Please see [CONTRIBUTING.md](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/CONTRIBUTING.md) for details.
Best wishes,
Paul

closed 2021-02-27 by PaulZC