sparkfun/SparkFun_ICM-20948_ArduinoLibrary #16: License: MIT or GPL?

filipbudisa, 2020-04-29:
Hello. In the readme it is stated that the code is released under GPL v3, but in License.md it says that SparkFun code is released under the MIT license. Could you please clarify under which license the code is released?

Thanks for your time and great work :)

oclyke, 2020-05-05:
Excellent question - I will get in touch with someone who would know more than me about this topic and then I will report back. Thanks!

PaulZC, 2021-02-08:
Hi @filipbudisa ,
Apologies for the delay!
We've clarified the license details in [this commit](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/commit/98516dfafa795d8eb58b6aa8a1fcdc9b97f34881). It is MIT.
Thank you for your patience,
Paul

closed 2021-02-08 by PaulZC