sparkfun/SparkFun_ICM-20948_ArduinoLibrary #19: auxiliary I2C interface to external sensors

EmanueleFeola, 2020-09-03:
Hi, does anybody know how to get data from external sensors connected to the icm-20948?
because ICM-20948 supports an auxiliary I2C interface to external sensors
Thank you!

oclyke, 2020-09-04:
Try using these functions:
[I2C Aux Controller Setup Functions](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/dff1a740ec4a970117c9208a0c89541ad4ecb232/src/ICM_20948.h#L92-L99)

EmanueleFeola, 2020-09-04:
thank you for the advice, as soon as the icm-20948 arrives in the office I will give it a try and I'll write a feedback.
My goal is to make it work with bmp280 as auxiliary sensor

EmanueleFeola, 2020-09-24:
I have tried this function " i2cMasterPassthrough(true); " to enable the interface bypass multiplexer, since I read on the ICM documentation that:
"The ICM-20948 has an interface bypass multiplexer, which connects the system processor I2C bus pins 23 and 24 (SCL and SDA) directly to the auxiliary sensor I2C bus pins 7 and 21 (AUX_CL and AUX_DA)"
I'm not an expert, but once I have enabled this bypass can I read external sensor data directly from an ESP32?
If I execute a I2C scan on the I2C bus I still see only one address: 0x69.
It would be really appreciated if someone could give me an advice / code to try, thank you

PaulZC, 2021-02-27:
Hi @EmanueleFeola ,
It is really not clear to me why you would want to access the auxiliary I2C pins. You can connect your BMP280 to the main I2C bus and communicate with it there. Those pins are only really there to access an external magnetometer - and the ICM-20948 has that built-in.
I'm going to close this, but please feel free to reopen if you think we've missed something.
Best wishes,
Paul

closed 2021-02-27 by PaulZC

EmanueleFeola, 2021-02-28:
Hi, the reason why I asked that is because I thought that if I fed the digital motion processor with bpm280 data, then the dmp could use that data with other sensor's data (accelerometer, ...).
Anyway the project is no longer active, so after a while I stopped trying.
Thank you for your support!