sparkfun/SparkFun_ICM-20948_ArduinoLibrary #20: 通信失败

MapleHan, 2020-10-13:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/dff1a740ec4a970117c9208a0c89541ad4ecb232/src/ICM_20948.cpp#L994
先拉低CS，然后beginTransaction()会导致通信失败。
调整为先beginTransaction()然后拉低CS，通信成功。
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = arduino

PaulZC, 2021-02-08:
Hi @MapleHan ,
Thank you for raising this issue. Apologies for the very slow reply.
Google translates this as:
**First pull down CS, and then beginTransaction() will cause communication failure.
Adjust to beginTransaction() first and then pull down CS, the communication is successful.**
I will correct this - thank you!
Best wishes,
Paul

PaulZC, 2021-02-08:
Thank you @MapleHan ,
Corrected in Pull Request #29 
Best wishes,
Paul

closed 2021-02-08 by PaulZC