sparkfun/SparkFun_ICM-20948_ArduinoLibrary #21: myICM.lowPower() doesn't reduce current consumption

adamgarbo, 2020-11-26:
### Subject of the issue
Hey @oclyke, I'm trying to place the ICM-20948 into its full-chip sleep mode (8 μA draw), but the `lowPower` function doesn't appear to have any effect on current consumption. 

### Your workbench
* What platform are you using?
  * SparkFun Qwiic Micro & ICM-20948
* How is the device wired to your platform?
  * Qwiic connector
* How is everything being powered?
  * USB. Current measured between 3.3 V 
* Are there any additional details that may help us help you?
  * LED and AUX PU jumpers cut

### Steps to reproduce
* I modified the "Example1_Basics" code to add the `myICM.sleep(true)` and `myICM.lowPower(true)` commands.
* I can observe the `myICM.sleep(true)` command reducing the ICM-20948 current consumption from 7.4 mA to 3.4 mA. 
* However, `myICM.lowPower(true)` has no effect on current draw, either on its own, or when paired with the sleep command.

```
void loop() {

  myICM.sleep(false);
  //myICM.lowPower(false);
  if ( myICM.dataReady() ) {
    myICM.getAGMT();                // The values are only updated when you call 'getAGMT'
    //printRawAGMT( myICM.agmt );     // Uncomment this to see the raw values, taken directly from the agmt structure
    printScaledAGMT( myICM.agmt);   // This function takes into account the sclae settings from when the measurement was made to calculate the values with units
    delay(2000);
  } else {
    Serial.println("Waiting for data");
    delay(500);
  }
  myICM.sleep(true);
  //myICM.lowPower(true);
  delay(2000);
}
```
### Expected behaviour
* I realize there's going to be some overhead from the level shifting circuity, but I would expect that current consumption should be able to drop significantly lower than 3.4 mA when placed into full-chip sleep/low power mode.

### Actual behaviour
* Current consumption remains excessively high.

Curious if you have any thoughts! In the meantime, I'll delve into the registers to see what could be up.

Cheers,
Adam

PaulZC, 2021-02-09:
Hi Adam (@adamgarbo ),

From the datasheet, it looks like sleep trumps lowPower:

[image]

I suspect you'd see a change if you compare lowPower to lowPower + sleep. And to get the best performance, it looks like you also need to call ```setSampleMode``` to set each sensor into duty-cycled mode.

Have you got time to give this a try?
Thanks,
Paul

adamgarbo, 2021-02-09:
Hi @PaulZC,

Thanks for the suggestion. I just tested this but it looks like no dice with setting the sample mode to duty cycle. For some reason, it just doesn't seem to want to fall below 3.5 mA draw. As far as I can tell, low power mode has no impact on current draw.

This isn't a huge priority for me any longer (I've moved on to the BNO080), but I would be curious to know what the cause of the high draw is. Could it be from all the level shifting circuity?

Cheers,
Adam

```
void loop()
{
  myICM.sleep(false);
  myICM.lowPower(false);

  myICM.setSampleMode((ICM_20948_Internal_Acc | ICM_20948_Internal_Gyr), ICM_20948_Sample_Mode_Continuous);
  Serial.print(F("Set sample mode of the sensor returned: "));
  Serial.println( myICM.statusString() );

  if (myICM.dataReady())
  {
    myICM.getAGMT();             // Values are only updated when 'getAGMT' is called
    printScaledAGMT(myICM.agmt); // Take into account the scale settings from when the measurement was made to calculate the values with units
    delay(30);
  }
  else
  {
    Serial.println("Waiting for data");
    delay(500);
  }

  myICM.setSampleMode((ICM_20948_Internal_Acc | ICM_20948_Internal_Gyr), ICM_20948_Sample_Mode_Cycled);
  Serial.print(F("Set sample mode of the sensor returned: "));
  Serial.println( myICM.statusString() );

  myICM.lowPower(true);
  myICM.sleep(true);

  delay(2000);
}
```

PaulZC, 2021-02-15:
Hi Adam,

I guess it must be the level-shifting circuitry... But I can't really tell without getting the soldering iron out. (And I don't have time to do that at the moment.)

The library is certainly doing everything it is supposed to be doing in terms of setting and clearing the SLEEP and LP_EN bits in the PWR_MGMT_1 register. And this is the same chip we use on OpenLog Artemis - where we can get the sleep current down below 100uA. So, level shifting and 1.8V regulation is pretty much all that is left.

If it's OK, I'd like to close this issue. You're welcome to raise an issue on the [hardware repo](https://github.com/sparkfun/SparkFun_Qwiic_9DoF_IMU_Breakout) if you'd still like us to dig into this - when time permits! ;-)

Cheers,
Paul

adamgarbo, 2021-02-15:
Hi @PaulZC,

I'm happy for this issue to be closed as unresolved. 

Cheers,
Adam

closed 2021-02-15 by adamgarbo