sparkfun/SparkFun_ICM-20948_ArduinoLibrary #24: What exactly does the "Wrong ID" error mean?

PBFInc, 2021-01-25:
Hello ICM-20948 Team,

I am attempting to use the aforementioned sensor in a project I'm working on. I am attempting to run the "Example1_Basics" code provided with the library to get a feel for how the sensor reacts to certain movements. However, every time I upload the code onto the Arduino UNO I've connected to the sensor, the initialization returns "Wrong ID" and then loops infinitely trying to correctly initialize. 

I've hooked up the sensor to the UNO using the SPI breakout pins as recommended by the SparkFun Hookup guide (as seen here: https://learn.sparkfun.com/tutorials/sparkfun-9dof-imu-icm-20948-breakout-hookup-guide?_ga=2.24541626.1092070436.1611006899-1783455205.1549928219 ). The only change I have made to the code is adjusting the baud rate to be the same as other devices used o the project. The reason I am using the SPI method is that the UNO I'm using currently has another sensor attached to it that is using I2C. From what I've read on the subject this shouldn't cause an issue with an SPI device (especially since the I2C sensor shouldn't even be transmitting anything when I upload the "Example1_Basics code onto the Arduino). The Arduino IDE I'm using is version 1.8.13 and the ICM-20948 library I'm using is 1.1.2. The sensor is being powered directly from the 3.3V pin and GND pins on the UNO, and everything is connected using hook-up wires and headers (which I have triple checked all the connections are secure.

I've tried to solve the issue myself, but every instance of the "Wrong ID" error found in the code I looked through in the depository does not explain what it means. Even in instances where there are comments explaining certain error codes, no comment exists for the "Wrong ID" error. Hopefully you can provide me with an explanation of what it means so we can solve this problem as quickly as possible.

PaulZC, 2021-02-09:
Hi @PBFInc ,
I've added a whole bunch of debug features in PR #30 .
Please add ```myIMU.enableDebugging();``` to your code _**before**_ the loop containing ```myIMU.begin();``` to see the debug messages on Serial.
You can use (e.g.) SerialUSB too. Call ```myIMU.enableDebugging(SerialUSB);``` instead.
I've added more features to help .begin start more smoothly so you should see the "Wrong ID" error a lot less than previously.
Please download the new code and give it a try. You may need to download it manually from the front page of this repo as the Arduino Library Manager may not pick up the new version until it is released. Look for the green Code link.
Let me know what you find.
Best wishes,
Paul

PaulZC, 2021-02-27:
Resolved in [Version 1.2.0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0)
Closing...

closed 2021-02-27 by PaulZC