sparkfun/SparkFun_ICM-20948_ArduinoLibrary #25: Unable to compile for Arduino Nano 33 BLE

ablankenship10, 2021-01-27:
### Subject of the issue
Unable to compile for Arduino Nano 33 BLE

### Your workbench
* What platform are you using? Windows 10, Arduino IDE 1.8.13 (Windows store 1.8.42.0)
* What version of the device are you using? Is there a firmware version? Arduino Nano 33 BLE
* How is the device wired to your platform? 3.3v, GND, SDA, SCL
* How is everything being powered? USB from PC
* Are there any additional details that may help us help you?

### Steps to reproduce
Use an Arduino Nano 33 BLE and attempt to compile and upload the Basic example sketch provided with this library.

### Expected behaviour
Should successfully compile the example sketch and upload it to the device

### Actual behaviour
```
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Nano 33 BLE"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Aaron\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Aaron\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Aaron\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=2341_805A -ide-version=10813 -build-path C:\Users\Aaron\AppData\Local\Temp\arduino_build_589287 -warnings=none -build-cache C:\Users\Aaron\AppData\Local\Temp\arduino_cache_73247 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino13 -prefs=runtime.tools.openocd-0.10.0-arduino13.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino13 -prefs=runtime.tools.dfu-util.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\dfu-util\0.9.0-arduino2 -prefs=runtime.tools.dfu-util-0.9.0-arduino2.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\dfu-util\0.9.0-arduino2 -prefs=runtime.tools.bossac.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\bossac\1.9.1-arduino2 -verbose C:\Users\Aaron\Documents\Arduino\libraries\SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library\examples\Arduino\Example1_Basics\Example1_Basics.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Aaron\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Aaron\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Aaron\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=2341_805A -ide-version=10813 -build-path C:\Users\Aaron\AppData\Local\Temp\arduino_build_589287 -warnings=none -build-cache C:\Users\Aaron\AppData\Local\Temp\arduino_cache_73247 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.openocd.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino13 -prefs=runtime.tools.openocd-0.10.0-arduino13.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\openocd\0.10.0-arduino13 -prefs=runtime.tools.dfu-util.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\dfu-util\0.9.0-arduino2 -prefs=runtime.tools.dfu-util-0.9.0-arduino2.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\dfu-util\0.9.0-arduino2 -prefs=runtime.tools.bossac.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Aaron\Documents\ArduinoData\packages\arduino\tools\bossac\1.9.1-arduino2 -verbose C:\Users\Aaron\Documents\Arduino\libraries\SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library\examples\Arduino\Example1_Basics\Example1_Basics.ino

Using board 'nano33ble' from platform in folder: C:\Users\Aaron\Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.1

Using core 'arduino' from platform in folder: C:\Users\Aaron\Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.1

Detecting libraries used...

"C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino" "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino" "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_589287\\sketch\\Example1_Basics.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for ICM_20948.h: [SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library@1.1.2]

ResolveLibrary(ICM_20948.h)

  -> candidates: [SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library@1.1.2]

"C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino" "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Aaron\\Documents\\Arduino\\libraries\\SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library\\src" "-IC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\cores\\arduino" "@C:\\Users\\Aaron\\Documents\\ArduinoData\\packages\\arduino\\hardware\\mbed\\1.3.1\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_589287\\sketch\\Example1_Basics.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for Wire.h: [Wire]

ResolveLibrary(Wire.h)

  -> candidates: [Wire]

[... cut ...]
Using library SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library at version 1.1.2 in folder: C:\Users\Aaron\Documents\Arduino\libraries\SparkFun_9DoF_IMU_Breakout_-_ICM_20948_-_Arduino_Library 

Using library Wire in folder: C:\Users\Aaron\Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.1\libraries\Wire (legacy)

Using library SPI in folder: C:\Users\Aaron\Documents\ArduinoData\packages\arduino\hardware\mbed\1.3.1\libraries\SPI (legacy)

exit status 1

Error compiling for board Arduino Nano 33 BLE.

```

PaulZC, 2021-02-08:
Hi Aaron (@ablankenship10),
Thank you for your patience - and apologies for the slow response.
I've corrected the code in [this commit](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/pull/28/commits/4806b519ec350fc320d297eff11d9d812aa42040). It now compiles cleanly on the NANO 33 BLE and every other board I have installed.
Can you please download the latest code and give it a try?
You may have to download the code manually - look for the green Code icon on the [GitHub repo](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary). The Arduino Library Manager may not pick up the changes right away.
Best wishes,
Paul

PaulZC, 2021-02-27:
Resolved in [Version 1.2.0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0)
Closing...

closed 2021-02-27 by PaulZC