sparkfun/SparkFun_ICM-20948_ArduinoLibrary #26: Connecting multiple SPI sensors to Arduino Nano

d-blaise, 2021-02-02:
### Subject of the issue
I cannot receive SPI data from multiple ICM-20948 sensors. I have four ICM-20948 sensors soldered to a custom PCB which connects to an Arduino Nano. The PCB layout is nearly identical to the [Adafruit TDK InvenSense ICM-20948 9-DoF IMU](https://learn.adafruit.com/adafruit-tdk-invensense-icm-20948-9-dof-imu/arduino) board. Each sensor has its own CS line.

### Your workbench
* Windows 10
* Arduino IDE 1.8.13
* Arduino Nano
* Custom PCB board, very similar to Adafruit TDK InvenSense ICM-20948 9-DoF IMU
* Power from USB, dropped to 1.8 V on custom board using LDO regulator and logic level converter
* Each ICM-20948 has its own CS line

### Steps to reproduce
I am using the Example1_Basics.ino code, but I include three additional ICM_20948 objects and CS lines:

`#include "ICM_20948.h"  // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

#define USE_SPI       
#define SERIAL_PORT Serial
#define SPI_PORT SPI            // Your desired SPI port.       Used only when "USE_SPI" is defined

#define CS_PIN 10               // Which pin you connect CS to. Used only when "USE_SPI" is defined
#define CS_PIN_orange 9
#define CS_PIN_white  8
#define CS_PIN_gray   7

#define WIRE_PORT Wire  // Your desired Wire port.      Used when "USE_SPI" is not defined
#define AD0_VAL   1     // The value of the last bit of the I2C address. 
                        // On the SparkFun 9DoF IMU breakout the default is 1, and when 
                        // the ADR jumper is closed the value becomes 0

#ifdef USE_SPI
  ICM_20948_SPI myICM;  // If using SPI create an ICM_20948_SPI object
  ICM_20948_SPI orange;
  ICM_20948_SPI white;
  ICM_20948_SPI gray;
#else
  ICM_20948_I2C myICM;  // Otherwise create an ICM_20948_I2C object
#endif

void setup() {

  SERIAL_PORT.begin(115200);
  while(!SERIAL_PORT){};

  pinMode (CS_PIN, OUTPUT);
  pinMode (CS_PIN_orange, OUTPUT);
  pinMode (CS_PIN_white, OUTPUT);
  pinMode (CS_PIN_gray, OUTPUT);

#ifdef USE_SPI
    SPI_PORT.begin();
#else
    WIRE_PORT.begin();
    WIRE_PORT.setClock(400000);
#endif
  
  bool initialized = false;
  while( !initialized ){

#ifdef USE_SPI
    myICM.begin( CS_PIN, SPI_PORT ); 
    orange.begin( CS_PIN_orange, SPI_PORT );
    white.begin( CS_PIN_white, SPI_PORT );
    gray.begin( CS_PIN_gray, SPI_PORT );

#else
    myICM.begin( WIRE_PORT, AD0_VAL );
#endif

      initialized = true;
  }
}

void loop() {

    Serial.println("1st Sensor:");
    myICM.getAGMT();                // The values are only updated when you call 'getAGMT'
    printRawAGMT( myICM.agmt );     // Uncomment this to see the raw values, taken directly from the agmt structure

    Serial.println("2nd Sensor:");
    orange.getAGMT();                // The values are only updated when you call 'getAGMT'
    printRawAGMT( orange.agmt );     // Uncomment this to see the raw values, taken directly from the agmt structure

    Serial.println("3rd Sensor:");
    white.getAGMT();                // The values are only updated when you call 'getAGMT'
    printRawAGMT( white.agmt );     // Uncomment this to see the raw values, taken directly from the agmt structure

    Serial.println("4th Sensor:");
    gray.getAGMT();                // The values are only updated when you call 'getAGMT'
    printRawAGMT( gray.agmt );     // Uncomment this to see the raw values, taken directly from the agmt structure
    
    delay(30);
}`

### Expected behaviour
I should expect to receive nonzero values from my serial monitor from each ICM-20948 sensor.

### Actual behaviour
Only the first ICM-20948 sensor produces a nonzero reading. My serial monitor data look like this:

1st Sensor:
RAW. Acc [  02264, -07008,  15064 ], Gyr [  00167,  00075,  00021 ], Mag [  00072, -00211,  00299 ], Tmp [  03232 ]
2nd Sensor:
RAW. Acc [ -00000, -00000, -00000 ], Gyr [ -00000, -00000, -00000 ], Mag [ -00000, -00000, -00000 ], Tmp [ -00000 ]
3rd Sensor:
RAW. Acc [ -00000, -00000, -00000 ], Gyr [ -00000, -00000, -00000 ], Mag [ -00000, -00000, -00000 ], Tmp [ -00000 ]
4th Sensor:
RAW. Acc [ -00000, -00000, -00000 ], Gyr [ -00000, -00000, -00000 ], Mag [ -00000, -00000, -00000 ], Tmp [ -00000 ]

PaulZC, 2021-02-09:
Hi @d-blaise ,
Thank you for raising this issue. I will investigate it for you as soon as I have the extra boards I need to duplicate your set-up.
Best wishes,
Paul

PaulZC, 2021-02-15:
Hi @d-blaise,

Thank you for your patience.

I've created a new example which communicates with two IMUs over SPI and I get valid data from both:

[image]

Can you please try the new example on your hardware and let me know if it works for you?

The example is [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example6_DualSPITest/Example6_DualSPITest.ino).

I have made other improvements to the library and the other examples too, so please do **download the complete library** - not just the new example. Look for the green Code link on the [front page of the repo](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary). The Arduino Library Manager won't pick up the new version until it is released.

Best wishes,
Paul

d-blaise, 2021-02-16:
Paul,

Thanks for your help! I will give this a shot today.

Best,
Dominic

d-blaise, 2021-02-16:
Hi Paul,

Thanks for your code, but I am still running into issues.

I am attaching a brief report on my attempt to run and debug this. I used
the same setup that you used (Arduino Nano and two ICM-20948 9-DoF IMUs),
the updated library you sent me, and your code. The result was that the
first sensor could not be initialized.

In the report, I scoped out all of my signals. The only signal that looked
a little sus was the SDO signal because the rise times look a little
excessive.

Please have a look. I feel like I am very close to resolving this.

Thank you for your time.

Best,
Dominic

d-blaise, 2021-02-16:
Hi Paul,

Please disregard that last message, I got it working. It must have been a
faulty connection on my breadboard.

[image: image.png]

Thank you!

Best,
Dominic

PaulZC, 2021-02-17:
HI Dominic (@d-blaise ),
Glad it is working for you. Thanks for letting me know.
Closing...
Best wishes,
Paul

closed 2021-02-17 by PaulZC