sparkfun/SparkFun_ICM-20948_ArduinoLibrary #34: Using DMP to receive multiple sensor data values -

n2d7, 2021-02-21:
Many thanks for adding the DMP support for this library. 
I followed the recent example _Example5_DMP_Quat9_Orientation_ and was successful getting the quaternion values from the sensor as the example shows. However, I would also want to get multiple data values at the same time, including quaternions and raw accel,gyro,mag data. 
To test it out, I added the following lines in the setup;
```
success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_ACCELEROMETER) == ICM_20948_Stat_Ok);
success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Accel, 200) == ICM_20948_Stat_Ok);

```
and in the loop
else if ((data.header & DMP_header_bitmap_Accel) > 0) // We have asked for orientation data so we should receive Quat9
```
{

	float acc_x = ((float)data.Raw_Accel.Data.X); 
	float acc_y = ((float)data.Raw_Accel.Data.Y); 
	float acc_z = ((float)data.Raw_Accel.Data.Z); 

	char sensor_string_buff[128];
	sprintf(sensor_string_buff, "{\"accel_x\":%f, \"accel_y\":%f, \"accel_z\":%f}", acc_x,acc_y,acc_z);
	Serial.println(sensor_string_buff);
}
```
However, I am not getting any data (both quaternion and accel values), and it seems like the fifo is empty. 
When reading the data `if ((myICM.status == ICM_20948_Stat_Ok) || (myICM.status == ICM_20948_Stat_FIFOMoreDataAvail))`  is always false. 
Appreciate any help to fix this.

PaulZC, 2021-02-22:
Hi @n2d7 ,

As you can probably tell, we're still gaining experience with the DMP and so we're not 100% confident on how to use it. Also, InvenSense admit that the documentation is not complete (which is why they don't want to release it openly).
Let me give you some suggestions:

I've updated [the comments in the DMP example](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example5_DMP_Quat9_Orientation/Example5_DMP_Quat9_Orientation.ino#L210-L225) to give more information on which combination of sensors are enabled for each mode.

```
success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_ACCELEROMETER) == ICM_20948_Stat_Ok);
success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Accel, 200) == ICM_20948_Stat_Ok);
```
This _should_ work fine and should enable the raw accelerometer output (via the FIFO). You may need to try lower rates than 200. Try 12 or 50 and see if that works. Try INV_ICM20948_SENSOR_RAW_ACCELEROMETER too - but it seems to load the same value into DATA_OUT_CTL1 so I really don't understand the difference. 

Your loop code looks good. Yes, I do think you need to check for:
```
else if ((data.header & DMP_header_bitmap_Accel) > 0)
```
and, yes, the accel data should be returned in:
```
data.Raw_Accel.Data.X
```
etc.. (Great detective work!)

So, I'm not quite sure what is going wrong...

What I plan to do next is to run @ericalbers [port of the InvenSense example](https://github.com/ericalbers/ICM20948_DMP_Arduino/tree/master), capture the SPI data so I can see exactly which registers are being set and with what values, and then make sure that my code is doing the same thing. It is very possible that we have missed a configuration step. (The InvenSense example code is very opaque and, like I said, InvenSense admit the documentation isn't complete.)

Please let me know what you find!
Best wishes,
Paul

n2d7, 2021-02-22:
Hi PaulZC,
Thank you very much for your prompt answer. And I really appreciate the effort that you put in to this work. 
I will try it with lower rates to see if it works. 

I also tested this with 2 IMUs in the same SPI bus. It seems to be working well after checking the incoming quaternion values with a gui application. 

Also, is it possible to read the quaternion values directly from the DMP without using the fifo?

PaulZC, 2021-02-22:
Hi @n2d7 ,
I _think_ the DMP can _only_ write to the FIFO. Or that‘s my understanding at the moment.
It will be a few days before I can try capturing the SPI (or I2C) data from @ericalbers code and do a write-by-write comparison. If you have the time and the curiosity to try it, please do. I’d really appreciate it.
Best wishes,
Paul

PaulZC, 2021-02-24:
Hi @n2d7 ,
Just for interest, here's a summary of the I2C traffic from @ericalbers's port of the InvenSense example. There are a few surprises in there and it has highlighted a few gremlins in my version of the code. I'll go through this in a day or two and will update my code to match.
Enjoy!
Paul

```
/* 
 *  From @ericalbers port of the InvenSense example (https://github.com/ericalbers/ICM20948_DMP_Arduino)
 *  The eric + InvenSense code configures the ICM-20948 (via I2C) as follows:
 *  
 *  Select and read register 0x00 (WhoAmI)
 *  ICM returns 0xEA
 *  Set register 0x06 (Power Management 1) to 0x01 (Auto clock)
 *  Select and read register 0x03 (User Control)
 *  ICM returns 0xC0
 *  Set register 0x07 (Power Management 2) to 0x47 (Reserved bit 6 set. Disable all Gyro axes)
 *  Select and read register 0x00 (WhoAmI)
 *  ICM returns 0xEA
 *  Set register 0x05 (Low Power Configuration) to 0x70 (I2C Master, Accel and Gyro in duty cycled mode)
 *  Set register 0x03 (User Control) to 0x00 (Disable DMP, disable FIFO, disable I2C Master)
 *  Set register 0x7C (Memory Start Address) to 0x90
 *  .....
 *  Write DMP firmware data to register 0x7D (Memory Read/Write) in blocks of 16 bytes
 *  During write, increment register 0x7E (Memory Bank Select) every 256 bytes
 *  .....
 *  Verify DMP firmware by reading it all back again
 *  .....
 *  Select and read register 0x7F (Register Bank Select) 
 *  ICM returns 0x00
 *  Set register 0x7F (Register Bank Select) to 0x20 == Bank 2
 *  Write 0x1000 to register 0x50 (Program Start Address High) (Magic Number)
 *  Set register 0x7F (Register Bank Select) to 0x00 == Bank 0
 *  Set register 0x7E (Memory Bank Select) to 0x00
 *  Set register 0x7C (Memory Start Address) to 0x40
 *  Write 0x0000 to memory (Data Out Control 1) (Disable all DMP features)
 *  Set register 0x7C (Memory Start Address) to 0x42
 *  Write 0x0000 to memory (Data Out Control 2) (Disable all "header2" features)
 *  Set register 0x7C (Memory Start Address) to 0x4C
 *  Write 0x0000 to memory (Data Interrupt Control) (Disable DMP interrupts)
 *  Set register 0x7C (Memory Start Address) to 0x4E
 *  Write 0x0000 to memory (FIFO Watermark) (Set watermark to zero)
 *  Set register 0x7C (Memory Start Address) to 0x8A
 *  Write 0x0000 to memory (Data Ready Status) (Disable all data ready indicators)
 *  Set register 0x7E (Memory Bank Select) to 0x01
 *  Set register 0x7C (Memory Start Address) to 0xFE
 *  Write 0x0320 to memory (31 * 16 + 14 = FIFO Watermark) (Set FIFO watermark to 800)
 *  Set register 0x10 (Reg Interrupt Enable) to 0x02 (Enable DMP interrupt to interrupt pin)
 *  Set register 0x12 (Reg Interrupt Enable 2) to 0x01 (Enable interrupt for FIFO overflow - 'channel' 0 only)
 *  Set register 0x26 (Single FIFO Priority Select) to 0xE4 (Worth investigating?)
 *  Select and read register 0x75 (Hardware Fix Disable) 
 *  ICM returns 0x48
 *  Set register 0x75 (Hardware Fix Disable) to 0x48 (Worth investigating?)
 *  Select and read register 0x7F (Register Bank Select)
 *  ICM returns 0x00
 *  Set register 0x7F (Register Bank Select) to 0x20 == Bank 2
 *  Set register 0x00 (Gyro Sample Rate Divider) to 0x13
 *  Write 0x0013 to 0x10 (Accelerometer Sample Rate Divider 1&2)
 *  Select and read register 0x7F (Register Bank Select)
 *  ICM returns 0x20
 *  Set register 0x7F (Register Bank Select) to 0x00 == Bank 0
 *  Set register 0x7E (Memory Bank Select) to 0x03
 *  Set register 0x7C (Memory Start Address) to 0x0A
 *  Write 0x0000 to memory (48 * 16 + 10 = BAC Rate)
 *  Set register 0x7C (Memory Start Address) to 0x08
 *  Write 0x0000 to memory (48 * 16 + 8 = B2S Rate)
 *  Set register 0x76 (FIFO Config) to 0x00
 *  Set register 0x68 (FIFO Reset) to 0x1F
 *  Set register 0x68 (FIFO Reset) to 0x1E
 *  Set register 0x66 (FIFO Enable 1) to 0x00
 *  Set register 0x67 (FIFO Enable 2) to 0x00
[... 411 lines cut ...]
 *  Accel:   0x0134 0xFDAC 0x1FC4
 *  Gyro:    0xFFF9 0x0013 0x0001
 *  Gyro_Bias:      0x0000 0x0000 0x0000
 *  Footer:  0x70DD
 *  .....
 *  Select and read register 0x19 (Interrupt Status)
 *  ICM returns 0x02
 *  Select and read register 0x18 (DMP Interrupt Status)
 *  ICM returns 0x01
 *  Select and read register 0x70 (FIFO Count Hi&Lo)
 *  ICM returns 0x0022
 *  Select and read register 0x72 (FIFO Read Write)
 *  Data Stream is: 
 *  Header:  0xC840 : Accel & Gyro & Quat6 & Gyro_Calibr
 *  Accel:   0x0188 0xFDC4 0x1FA4
 *  Gyro:    0xFFF1 0x0019 0xFFFF
 *  Gyro_Bias:      0x0000 0x0000 0x0000
 *  Quat6:   0xFFC7A057 0x0030BD92 0x000A6646
 *  No footer!
*/
```

The raw I2C data is in the attached file:

[ericalbers1.txt](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/files/6039240/ericalbers1.txt)

n2d7, 2021-02-25:
Hi Paul,
Really appreciate you looking in to this. Sadly, I don't have much experience at low level code :/  (I am a student).  But I will try to see how the above differs from the current code.

PaulZC, 2021-02-26:
Hi @n2d7 ,
Success!
I've made improvements to the DMP configuration and "read FIFO" code.
The critical thing (which I had missed) is that the Data Ready Status register needs to be set when requesting accelerometer and/or gyro values from the DMP. Oddly, it doesn't need to be set when requesting only quaternions...
I've added a [new example](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example8_DMP_RawAccel/Example8_DMP_RawAccel.ino) which - I think - does exactly what you need. Please give it a try and let me know what you find.
You will need to download the complete library, not just the new example.
And please don't forget to edit ICM_20948_C.h and uncomment line 29, otherwise the DMP won't work.
Best wishes,
Paul

PaulZC, 2021-02-27:
Resolved in [Version 1.2.0](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.0)
Closing...

n2d7, 2021-02-28:
Hi Paul, 
I started checking the code with `Example6_DMP_Quat9_Orientation`. I checked with 3 IMUs connected with SPI (one at a time). While I receive the incoming quaternion values, I think they are wrong. For 2 IMUs I checked, when I visualized the values, 90 degree turn of the real IMU was visualized as 2 full turns. Additionally it was not the visualization was not flat it was supposed to (I have used the same visualization setup before). For the other one, q1,q2,q3 values were either increasing or decreasing constantly). So I think their may be a something missing when configuring the quaternion values. 

Similarly, for ` Example7_DMP_Quat6_EulerAngles.ino `yaw, pitch, roll values were constantly increasing or decreasing for all 3 IMUs. 
```
Initialization of the sensor returned: All is well.
Device connected!
DMP enabled!
Roll:0.0 Pitch:0.1 Yaw:0.3
Roll:0.0 Pitch:0.2 Yaw:0.6
Roll:0.1 Pitch:0.4 Yaw:0.9
Roll:0.1 Pitch:0.5 Yaw:1.1
Roll:0.2 Pitch:0.6 Yaw:1.4
Roll:0.2 Pitch:0.7 Yaw:1.7
Roll:0.2 Pitch:0.8 Yaw:2.0
Roll:0.2 Pitch:0.9 Yaw:2.3
Roll:0.3 Pitch:1.0 Yaw:2.6
Roll:0.3 Pitch:1.1 Yaw:2.8
Roll:0.3 Pitch:1.2 Yaw:3.1
Roll:0.4 Pitch:1.3 Yaw:3.4
Roll:0.4 Pitch:1.4 Yaw:3.7
Roll:0.4 Pitch:1.4 Yaw:4.0
Roll:0.5 Pitch:1.5 Yaw:4.2
Roll:0.5 Pitch:1.6 Yaw:4.5
Roll:0.5 Pitch:1.7 Yaw:4.8
Roll:0.5 Pitch:1.7 Yaw:5.0
Roll:0.5 Pitch:1.8 Yaw:5.3
Roll:0.5 Pitch:1.8 Yaw:5.6
Roll:0.5 Pitch:1.9 Yaw:5.8
Roll:0.6 Pitch:2.0 Yaw:6.1
Roll:0.6 Pitch:2.0 Yaw:6.4
Roll:0.6 Pitch:2.1 Yaw:6.7
Roll:0.6 Pitch:2.1 Yaw:6.9
Roll:0.6 Pitch:2.2 Yaw:7.2
Roll:0.6 Pitch:2.3 Yaw:7.5
Roll:0.7 Pitch:2.3 Yaw:7.8
Roll:0.7 Pitch:2.3 Yaw:8.0
Roll:0.7 Pitch:2.4 Yaw:8.3
Roll:0.7 Pitch:2.4 Yaw:8.6
Roll:0.7 Pitch:2.4 Yaw:8.9
Roll:0.7 Pitch:2.5 Yaw:9.1
Roll:0.7 Pitch:2.5 Yaw:9.4
Roll:0.7 Pitch:2.6 Yaw:9.7
Roll:0.8 Pitch:2.6 Yaw:9.9
Roll:0.8 Pitch:2.6 Yaw:10.2
Roll:0.8 Pitch:2.6 Yaw:10.5
Roll:0.8 Pitch:2.6 Yaw:10.8
Roll:0.8 Pitch:2.7 Yaw:11.1
Roll:0.8 Pitch:2.7 Yaw:11.4
Roll:0.8 Pitch:2.7 Yaw:11.6
Roll:0.9 Pitch:2.7 Yaw:11.9
Roll:0.9 Pitch:2.8 Yaw:12.2
Roll:0.9 Pitch:2.8 Yaw:12.4
Roll:0.9 Pitch:2.8 Yaw:12.7
Roll:0.9 Pitch:2.8 Yaw:12.9
```
I also check with the `Example8_DMP_RawAccel.ino`. For all 3 IMUs, I did a plot of the acceleration values and tested them with simple movements. Their values seems to be accurate (by dividing the values by 2^16, it shows closed to 1g in the correct direction, also I plotted the values and moved the IMUs and checked if the movement matched the plot)

PaulZC, 2021-02-28:
Hi @n2d7,
Can you please check that you have installed version 1.2.0 of the library? There was a mistake in the earlier code that meant the magnetometer was not being configured correctly. I was seeing the same "2 full turn" behavior before I corrected the code.
Also, can you try turning the sensor through a complete revolution in all three axes - to allow the DMP to calibrate.
Please let me know what you find.
Best wishes,
Paul

PaulZC, 2021-02-28:
Hi @n2d7 ,
OK. I see the same behavior with the Quat9 example - a slow drift in Q3. The Quat6 data (Example7) seems much better behaved.
InvenSense did give me some advice:
"I would start with the Quat6 first and see if you are seeing this drift. The reason is that Quat9 integrate the mag and until the mag data is calibrated it could cause probably this strange yaw movements. With the Quat6 you will only get the quaternions from the accel and gyro which would be more stable.
You might see a drift due to the initial offsets. But the drift should be smaller and less erratic. Then after the DMP calibrates the gyro it should be much more stable."
So, I suspect this is a calibration issue.
I will investigate further!
Best wishes,
Paul

PaulZC, 2021-02-28:
Hi @n2d7 ,
I've made some improvements to the way the DMP is configured in [Version 1.2.1](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.1).
Please give the updated examples a try. I still see drift on the Quat9 Q3 initially, but after rolling the sensor around all three axes, it seems to stabilize OK.
Best wishes,
Paul

n2d7, 2021-03-02:
Hi Paul,
Sorry, I was not able to check it till now. I drift of the values are not there anymore. But I am still getting wrong orientations when I plot the quaternion values. I took a small [video](https://drive.google.com/file/d/1KnMTI6iS33EM4RUNWid6ar1XH80amuyD/view?usp=sharing) while I plot the orientation 

I calculate q0 as ;` float q0 = sqrt(1-q0*q0-q1*q1-q2*q2-q3*q3);`. Not sure if I am doing anything wrong when calculating q0.

PaulZC, 2021-03-02:
Hi @n2d7 ,
Thanks for the video. That is much easier to view than columns of numbers! Can you please share your 3D orientation plotting code so I can try the same thing here? I made some small improvements to the library yesterday and it would be good to test them in the same way.
Thank you,
Paul

n2d7, 2021-03-02:
Hi Paul,
It is not my own work. I use Zanel's code to visualize the quaternion values. [ICM library ](https://github.com/ZaneL/Teensy-ICM-20948) and [visualization ](https://github.com/ZaneL/quaternion_sensor_3d_nodejs). I only change the `serial.println()` in your code to match the format of the [QuaternionAnimation](https://github.com/ZaneL/Teensy-ICM-20948/tree/master/examples/QuaternionAnimation) example of Zanel's. Hope this helps. 
The library is very useful itself, the issue with that library  i that I couldn't get it to work with multiple IMUs (It uses invensense example code).

PaulZC, 2021-03-03:
Hi @n2d7 ,
No problem - thanks for the links!
Please be aware that ZaneL's example uses the Game Rotation Vector for the Quaternions. GRV is Quat6, not Quat9.
https://github.com/ZaneL/Teensy-ICM-20948/blob/master/src/Teensy-ICM-20948.cpp#L455
https://github.com/ZaneL/Teensy-ICM-20948/blob/master/src/Teensy-ICM-20948.cpp#L305-L315
I will keep looking at this. Our library does not current support multiple DMP sensors/features. You currently cannot select GRV+Accel+Gyro; you can only select GRV or Accel or Gyro. I am going to add some extra code to support multiple sensors.
Best wishes,
Paul

PaulZC, 2021-03-03:
Hi @n2d7 ,
Please give v1.2.2 a try. I think we might be getting closer! Example6 (9-axis Quaternion) and Example7 (6-axis Quaternion) have been updated. You can now pipe data straight to ZaneL's Node.js Quaternion animation tool. Much nicer than columns of numbers! Just uncomment the ```#define QUAT_ANIMATION```. You can now also enable multiple sensors/features simultaneously - see Example9.
Best wishes,
Paul

n2d7, 2021-03-05:
Hi Paul,
I checked the examples and they are working perfectly! Thank you very much for all the hard work. :)

I have another small question. 
` // myICM.setSampleRate( (ICM_20948_Internal_Acc | ICM_20948_Internal_Gyr), mySmplrt ); // ** Note: comment this line to leave the sample rates at the maximum **`
Here what is the maximum sample rate? Is it 225Hz(In Zanel's library, I read that the maximum output rates are 225Hz)?

PaulZC, 2021-03-05:
Hi @n2d7,
I'm glad that is working for you!
Yes, I believe 225 is the maximum. You will need to change several things:
In these lines, change 19 to 4:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L155-L156
Change 19 to 4 again here:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L229
Use the values for 225Hz in these three lines:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L240
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L245
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L250
Set the ODR Interval to zero:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L297
I'm going to close this issue now (again), but please let me know if 225Hz is successful.
Best wishes,
Paul

Niv1900, 2021-03-08:
Hello @PaulZC, 

First, thank you very much for all this work!

I had the same problem as @n2d7, getting the info from the ICM at a high frequency (200Hz) and having in parallel the info from the DMP. 

By testing example 9 with the suggested modifications, I only get the classic ICM information, I don't really no where I failed here... 

[image]

additional info:
- ESP WROOM 32 Devboard;
- SPI connection;
- VsCode / PlatformIO;
- 1.2.2 Version of this lib (not using PIO lib_deps feature, lib manually placed in the lib folder to be sure to use the last version);
- code used : https://gist.github.com/Niv1900/b787691bc7be84c95ea41117a673a751

n2d7, 2021-03-09:
Hi @PaulZC  and @Niv1900;
I have the same issue, the quaternion values are not printing. I checked with the original example for Example9_DMP_MultipleSensors sensors and its the same (I thought it was working well earlier, but I might have missed checking for quaternion values).  I am using SPI with Teensy 3.6. 

Thanks

n2d7, 2021-03-09:
Hi,
I figured out the issue is;
seems like the line `if ( (data.header & DMP_header_bitmap_Quat9) > 0 ) // Check for orientation data (Quat9)`
correspond to `sensor_orientation `data than `game_rotation_vector`
So I changed the line
`success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_GAME_ROTATION_VECTOR) == ICM_20948_Stat_Ok);`
to
 `success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_ORIENTATION) == ICM_20948_Stat_Ok);
`
After following Paul's instruction I was able to get data at 225Hz. (But I think the maximum speeds for mag and quat data or 70Hz or something).

@PaulZC 
My next step is to get linear acceleration from the dmp with `INV_ICM20948_SENSOR_LINEAR_ACCELERATION`. However, in `DMP_Header_Bitmap` its not clear which one to use as the header.  Would you be able to help out on reading linear acceleration data? 

Thanks!

Niv1900, 2021-03-09:
Thanks for that @n2d7!

what about retrieving data this way @PaulZC? 

```
if (myICM.dataReady())
{
        myICM.getAGMT();
        float aX = myICM.accX();
        float aY = myICM.accY();
        float aZ = myICM.accZ();

        float gX = myICM.gyrX();
        float gY = myICM.gyrY();
        float gZ = myICM.gyrZ();

        float mX = myICM.magX();
        float mY = myICM.magY();
        float mZ = myICM.magZ();    
  }
```

is this an outdated way of doing things?

PaulZC, 2021-03-09:
Hi @Niv1900 / @n2d7 ,
@n2d7 : thanks for the correction! Updated examples will follow shortly...
@Niv1900 : getAGMT pulls data from the standard ICM registers. It doesn't use the DMP data from the FIFO. It _may_ be possible to use both, but I haven't tried it!
Best wishes,
Paul

closed 2021-03-09 by PaulZC

n2d7, 2021-03-12:
Hi @PaulZC 
My next step is to get linear acceleration from the dmp with `INV_ICM20948_SENSOR_LINEAR_ACCELERATION`. However, in `DMP_Header_Bitmap` its not clear which one to use as the header.  Would you be able to help out on reading linear acceleration data? It's not very much clear to me how to map the sensor_type to the header.

PaulZC, 2021-03-12:
Hi @n2d7,

Sorry, I missed your question earlier...

```INV_ICM20948_SENSOR_LINEAR_ACCELERATION``` is defined [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L327)

In the code, the enum gets converted into an "Android Sensor" number: ```ANDROID_SENSOR_LINEAR_ACCELERATION; // 10``` [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_C.c#L2346)

That then gets converted into a bit pattern for the DMP Data Out Control Register 1 [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_C.c#L72)

So, three 'sensors' or 'features' are being enabled: 16-bit accel, 32-bit 6-axis quaternion, and Header2

```inv_icm20948_enable_dmp_sensor``` updates DATA_OUT_CTL1 [here](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_C.c#L1701). It also updates DATA_OUT_CTL2 with any 'extra' features that need to be enabled. In your case, the accel accuracy will be enabled too.

So, when you check Header, you should see:
[DMP_header_bitmap_Accel](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L474)
[DMP_header_bitmap_Quat6](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L470)
[DMP_header_bitmap_Header2](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L462)

When you check Header2, you should see:
[DMP_header2_bitmap_Accel_Accuracy](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L485)

In the ```icm_20948_DMP_data_t```, you can extract the data from:
[Raw_Accel](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L540-L549)
[Quat6](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L583-L592)
[Accel_Accuracy](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/src/util/ICM_20948_DMP.h#L647)

I hope that helps!
Paul

PaulZC, 2021-05-10:
Hi @QuaternionNightmare ,

You will only get the Header2 accuracy data if they are requested when enabling the sensor. The relevant code is this:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/ffca57bba17eaf5e20af321bcd2a98fb9020e556/src/util/ICM_20948_C.c#L1706-L1719

With Quat9, only the compass accuracy is requested.

I'm sure I based this directly on the InvenSense code, but, as usual, it will take me a while to find the exact lines in their code.

Please poke at this a bit more and let me know if things are behaving as they should be.

Thanks and best wishes,
Paul

QuaternionNightmare, 2021-05-10:
Hi @PaulZC,

Many thanks for your reply. I will poke at this more and let you know what I did to get it to work.

Kind regards.

PaulZC, 2021-05-10:
Just a random thought... I wonder if all of the DMP misbehavior could be linked to the low power mode?
If you look through the code, there are many places where I make sure the chip is not in low power mode. Again this is based on what InvenSense do in their example code. Here's an example:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/ffca57bba17eaf5e20af321bcd2a98fb9020e556/src/util/ICM_20948_C.c#L1849-L1853

But, I commented out the code which puts it back into the low power state afterwards and, right now, I can't remember why I did that... (But I was working my way through many issues back then.) I wonder if making sure the chip goes back into the low power state actually helps the DMP, or perhaps is even essential?!

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/ffca57bba17eaf5e20af321bcd2a98fb9020e556/src/util/ICM_20948_C.c#L1864-L1866

I will give this a try as soon as I get some spare time - but I have other things I need to work on today. If you want to give it a try, please be my guest - I appreciate the help.

Best wishes,
Paul

QuaternionNightmare, 2021-05-11:
Hi @PaulZC,

I gave this a bash last night found something interesting. For the Quat9 example, the very first line of the buffer outputted to serial shows header2 has 0x1000 which is compass accuracy, the rest afterwards are all 0x0000 however despite the actual data.Quat9.Data.Accuracy values still being printed and updated to serial.

I did try uncommenting the power mode lines you suggested in ICM_20948_C.c but that didn't seem to change anything for me.
I'll keep chipping away at this and let you know if I have any luck.

Kind regards.

PaulZC, 2021-05-11:
Hi @QuaternionNightmare ,
Thank you for digging in to this... I really appreciate the help!
Best wishes,
Paul

Hamdi8, 2021-05-24:
Hi
I'm finding an error in this ligne ! 
 success &= (myICM.setGyroSF(19, 3) == ICM_20948_Stat_Ok); // 19 = 55Hz (see above), 3 = 2000dps (see above)
I work with openlog Artemis and i use the IMU_DMP QUAt6 , 
who can help meplease ?
Hamdi,

PaulZC, 2021-05-24:
Hi Hamdi (@Hamdi8 ),
What kind of error? If you see errors on the Serial Monitor of the compiler output, please copy and paste them here.
Best Wishes,
Paul

Hamdi8, 2021-05-24:
Hi Paul , 
This is the error .
[image]
Kind regards.,
Hamdi

PaulZC, 2021-05-24:
Hi Hamdi (@Hamdi8 ),
[Example10](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example10_DMP_FastMultipleSensors/Example10_DMP_FastMultipleSensors.ino) calls setGyroSF and it compiles correctly for me. Please make sure you are using version 1.2.5 of the ICM20948 library. You may need to update through the Library Manager.
There are some updated examples for the OpenLog Artemis [here](https://github.com/sparkfun/OpenLog_Artemis/tree/release_candidate/Firmware/Test%20Sketches) which you may find useful. (They have not been released yet - they are in a separate branch of the OpenLog Artemis repository.)
Best wishes,
Paul

Hamdi8, 2021-05-24:
Hi Paul , 
Ok thanks for your help. 
Actually i'm using the same example to find out 'yaw' , 'pitch' and 'roll' , but the problem is that the values display in the serial monitor very slowly , have you an idea how to raise the rate . 
Best wishes,
Hamdi

PaulZC, 2021-05-27:
Hi @QuaternionNightmare ,

Please try version 1.2.6 of the library. I have made a small but important change to allow the DMP to work correctly. Please see issue #59 for more details.

I hope this fixes your issue!

Best wishes,
Paul

QuaternionNightmare, 2021-05-27:
Hi @PaulZC,
Wow! What a difference. Yep, all looks good on header2 now with the Quat9 example. Thanks for copying in the other thread too for reference, I will take a look at the exact cause in detail.
Many thanks!

Varocol, 2022-11-13:
thanks for this issue, cause i found the name of the unknow 9 axis sensor right now,many thanks