sparkfun/SparkFun_ICM-20948_ArduinoLibrary #37: Several vars defined in header (ICM_20948_DMP.h)

nikitos1550, 2021-02-28:
### Subject of the issue
Several vars defined in header (ICM_20948_DMP.h).

### Your workbench
I am using x86_64 with CJMCU-20948 board connected to PC via i2c_tiny_usb adapter (cheap chinese USBASP v2 reflashed with another firmware to get /dev/i2c-X device in Linux env).

### Steps to reproduce
cc util/ICM_20948_C.c ./main.c ./i2c.c -o ./i2c-test
where main.c includes ICM_20948_C.h

### Expected behaviour
Compilation should work

### Actual behaviour
As several vars define in header (ICM_20948_DMP.h), I have following errors:
```
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0x0): multiple definition of `inv_androidSensor_to_control_bits'; /tmp/cc5DSmAB.o:(.rodata+0x0): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0x60): multiple definition of `DMP_Quat9_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0x60): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0xa0): multiple definition of `DMP_Quat6_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0xa0): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0xd0): multiple definition of `DMP_PQuat6_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0xd0): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0x100): multiple definition of `DMP_Raw_Gyro_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0x100): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0x130): multiple definition of `DMP_Activity_Recognition_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0x130): first defined here
/usr/bin/ld: /tmp/ccVjw7df.o:(.rodata+0x148): multiple definition of `DMP_Secondary_On_Off_Byte_Ordering'; /tmp/cc5DSmAB.o:(.rodata+0x148): first defined here
```

I made them static, like this:
```
// ICM-20948 data is big-endian. We need to make it little-endian when writing into icm_20948_DMP_data_t
const static int DMP_Quat9_Byte_Ordering[icm_20948_DMP_Quat9_Bytes] =
{
        3,2,1,0,7,6,5,4,11,10,9,8,13,12 // Also used for Geomag
};
const static int DMP_Quat6_Byte_Ordering[icm_20948_DMP_Quat6_Bytes] =
{
        3,2,1,0,7,6,5,4,11,10,9,8 // Also used for Gyro_Calibr, Compass_Calibr
};
const static int DMP_PQuat6_Byte_Ordering[icm_20948_DMP_PQuat6_Bytes] =
{
        1,0,3,2,5,4 // Also used for Raw_Accel, Compass
};
const static int DMP_Raw_Gyro_Byte_Ordering[icm_20948_DMP_Raw_Gyro_Bytes + icm_20948_DMP_Gyro_Bias_Bytes] =
{
        1,0,3,2,5,4,7,6,9,8,11,10
};
const static int DMP_Activity_Recognition_Byte_Ordering[icm_20948_DMP_Activity_Recognition_Bytes] =
{
        0,1,5,4,3,2
};
const static int DMP_Secondary_On_Off_Byte_Ordering[icm_20948_DMP_Secondary_On_Off_Bytes] =
{
        1,0
};
```
It is no problem to patch lib this way, but maybe it is possible to redeclare this vars in source (.c) file, as now several copies of these vars will appear in final binary.

PaulZC, 2021-02-28:
Hi Nikita (@nikitos1550 ),
Can you please submit this as a Pull Request?
Please see [this link](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/CONTRIBUTING.md) for details.
Many thanks,
Paul

nikitos1550, 2021-03-01:
I will try!

nikitos1550, 2021-03-01:
@PaulZC Besides issue described above, I am thinking is it possible to avoid C++ in this lib, or make lib with bith C and C++ interfaces.
For mine raw experiments I did something like this https://pastebin.com/QGQm4Qqx

I am looking for some way making lib available for C and C++ cases, do you have any ideas?

PaulZC, 2021-03-01:
Hi Nikita (@nikitos1550 ),
It might be worth trying the InvenSense eMD examples. They are written in C. But they are opaque and very difficult to follow.
Most of the functionality in the SparkFun library is C, but we provide the C++ wrapper for Arduino.
Please feel free to develop your own fork. Start with ICM_20948_C.h and ICM_20948_C.c. They will do most of what you need. You will need to figure out how to do your I2C/SPI transactions using C.
Best wishes,
Paul

closed 2021-03-09 by nikitos1550