sparkfun/SparkFun_ICM-20948_ArduinoLibrary #38: DMP examples not architecture agnostic

adamgarbo, 2021-02-28:
Hi @PaulZC,

Great work on the ICM-20948 library!

A quick note to let you know that the DMP examples make use of `Serial.printf` for displaying outputs, which is not supported on all platforms (e.g. SAMD21).

Cheers,
Adam

PaulZC, 2021-02-28:
Thanks Adam,
If you've got a truly agnostic solution for this, I'm all ears...
Cheers,
Paul

adamgarbo, 2021-02-28:
Hi Paul,

I just replaced the `printf` commands with normal print statements. For example:

```
// Existing
SERIAL_PORT.printf("Q1:%.3f Q2:%.3f Q3:%.3f\r\n", q1, q2, q3);

// New
SERIAL_PORT.print("Q1: "); SERIAL_PORT.print(q1, 3);
SERIAL_PORT.print(" Q2: "); SERIAL_PORT.print(q2, 3);
SERIAL_PORT.print(" Q3: "); SERIAL_PORT.println(q3, 3);
```

I tested the above code with a SparkFun Qwiic Micro and it works well. I haven't been able to get the DMP examples to work with v2.0.5 of the Apollo3 Core yet, but the print statements should work just as well.

Cheers,
Adam

PaulZC, 2021-02-28:
Okie dokie... Pull Request please? (I think you'll need to update all three DMP examples)
Thanks!
Paul

PaulZC, 2021-02-28:
I had to update the examples for Version 1.2.1 - so I changed the Serial Prints too...
Closing...
Cheers!
Paul

closed 2021-02-28 by PaulZC

adamgarbo, 2021-02-28:
Thanks, Paul! I'll keep my fork going for more testing and troubleshooting.