sparkfun/SparkFun_ICM-20948_ArduinoLibrary #42: Question: DMP and FIFO, case when you need to grab data rarely.

nikitos1550, 2021-03-06:
In all DMP examples there is following thing:
```
  if ( myICM.status != ICM_20948_Stat_FIFOMoreDataAvail ) // If more data is available then we should read it right away - and not delay
  {
    delay(10);
  }
```
As I understand DMP is working following way:
All time it is getting raw values for some rate, perform calculations, finally produce for example Quat and put it to FIFO.
According example code, it is supposed to read FIFO fast enough.

But what if I don`t want to grab data always, In my case I need sometimes get actual data(actual current orientation).
How can I reach this? 

How deep is FIFO? Will DMP still push new data to FIFO, when it will be full (deleting oldest data)? 
Even yes, I will always get some old data (depending on FIFIO size).

Is it possible to disable FIFO or adjust it`s size?

PaulZC, 2021-03-07:
Hi Nikita (@nikitos1550 ),

The DMP can only deliver its data through the FIFO, so, no, you cannot disable it.
And the size is fixed - at 512 bytes. It is big enough to hold several packets of data - depending on which sensors or features are enabled.

The FIFO can be configured to use "Stream" or "Snapshot" mode via the FIFO_MODE register and the ```setFIFOmode``` command. "Stream" mode overwrite old data when the FIFO becomes full. "Snapshot" causes the FIFO to stall when full - data is not overwritten. The default is "Stream".

You can adjust the rate of the DMP data. Please see [Example9](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L292-L302). 1Hz is possible.

You can also use interrupts. The ICM can generate an interrupt when new DMP data is ready (so you do not need to keep polling the FIFO). [The interrupt is disabled by default](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L265). But you can enable it. You can also tell the DMP to generate interrupts for each sensor/feature using the ```enableDMPSensorInt``` but I don't have an example for that - sorry!

Please close this issue if this answers your question.

Good luck!

Best wishes,
Paul

nikitos1550, 2021-03-07:
@PaulZC Seems "Stream" mode can help, could you point me to something related to changing FIFO mode?

PaulZC, 2021-03-07:
The oldest data is removed:
[image]

```setFIFOmode(true)``` will enable snapshot mode if you want to try it.
Paul

closed 2021-03-08 by nikitos1550