sparkfun/SparkFun_ICM-20948_ArduinoLibrary #52: In which way are the agmt values set?

JinjaFormbi, 2021-03-24:
### Subject of the issue
I can't find/understand the way, the value of the agmt are set.

Currently I am starting the process of re purposing your library for a Raspberry 4. I am already able to get readings from accY(); and the others, now in testing the readouts they are either nonsensical or unchanging. So i am trying to understand where the values are actually "created".

Thank you for your patience.

PS.: I am only aware of your library, you don't know randomly of a raspberry library for the SparkFun_ICM-20948?

mfide, 2021-03-25:
I use ICM_20948_get_agmt() function to read all values and seems they are working for me.

PaulZC, 2021-03-27:
Hi @JinjaFormbi ,
We have a Python library for the ICM20948 which will run on Raspberry Pi:
https://github.com/sparkfun/Qwiic_9DoF_IMU_ICM20948_Py
Best wishes,
Paul

closed 2021-03-27 by PaulZC