sparkfun/SparkFun_ICM-20948_ArduinoLibrary #53: Example7_DMP_Quat6_EulerAngles pitch and roll error

BaptisteLucas, 2021-03-25:
Hi,
I'm testing the IMU with the Example7_DMP_Quat6_EulerAngles example.
I tried to turn around the yaw axis with IMU put on the table, so that "normaly" the roll and pitch axis needs to be constant during the rotation.
The result is that the yaw axis works perfectly, but I can see angular bias on both pitch and roll axis during the rotation (of about 20° max), this bias seems repeatable in function of the yaw angle.

Am I alone with this problem ?

PaulZC, 2021-03-27:
HI @BaptisteLucas ,
We are still gaining experience with the DMP and sadly we do not have all the answers...
What happens if you use quaternions? Do you still see the issue if you try [Zane L's quaternion animation tool](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example7_DMP_Quat6_EulerAngles/Example7_DMP_Quat6_EulerAngles.ino#L27)?
Does the [Quat9 example](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example6_DMP_Quat9_Orientation/Example6_DMP_Quat9_Orientation.ino) improve the bias?
Have you tried calibrating the DMP by rotating the sensor around all three axes, and have had it stationary in all six orientations?
Best wishes,
Paul

BaptisteLucas, 2021-03-29:
I can't install the Zane L's animation tool so I can't try it.

For the Quat 9 example, the result is the same.

How can I calibrate the DMP ? 

I also seen an angle bias (for the yaw angle) of some degrees between 0° and +/-90° (+/-180° is correctly aligned with 0°).

Is there a way to force the DMP calibration ? Because the bias seems to be reproductibles.

PaulZC, 2021-03-29:
Hi @BaptisteLucas ,
Like I said, we are still gaining experience with the DMP and sadly we do not have all the answers...
As I understand it, you calibrate the DMP by rotating the sensor around all three axes, and hold it stationary in all six orientations (X up/down, Y up/down, Z up/down).
Best wishes,
Paul

PaulZC, 2021-04-26:
Hi @BaptisteLucas ,
Please test v1.2.5. I think it will resolve your issue.
Please remember that the Quat6 data does not use the magnetometer so you may still see some drift there.
Quat9 includes the (corrected) magnetometer readings.
I am going to close this issue as I believe it is resolved, but please reopen if you see any more errors.
Best wishes,
Paul

closed 2021-04-26 by PaulZC