sparkfun/SparkFun_ICM-20948_ArduinoLibrary #55: Quat9 Issues with Q0 Calculations

Zed-svg, 2021-04-22:
For the Example6_DMP_Quat9_Orientation sketch, when calculating Q0 you do:

q0 = sqrt( 1.0 - ((q1 * q1) + (q2 * q2) + (q3 * q3)));

Isn't this incorrect because q0 can be positive or negative? The equation above always makes q0 positive, which is incorrect.

I am thinking of how to solve this issue but my understand of quaternions is entry level in terms of finding Q0 from the DMP or with XYZ.

PaulZC, 2021-04-23:
Hi @Zed-svg ,
Good point. [The comments in the code](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/master/examples/Arduino/Example7_DMP_Quat6_EulerAngles/Example7_DMP_Quat6_EulerAngles.ino#L366-L368) come straight from the InvenSense documentation. But I agree, that gives you the magnitude of Q0, not the sign.
If you have a solution, I'm all ears!
It may be possible to calculate the sign from the rotation angle in [these equations](https://en.wikipedia.org/wiki/Rotation_formalisms_in_three_dimensions#Quaternions). There is more useful information [here](http://danceswithcode.net/engineeringnotes/quaternions/quaternions.html#:~:text=A%20rotation%20quaternion%20is%20similar).
Very best wishes,
Paul

higherstateofawkwardness, 2021-05-14:
Guys,

First I'd like to thank @PaulZC for all the work that he has been doing on this - really appreciate it. I've been having a play with the code to stream out both Quat6 and acceleration values simultaneously to the serial port of an Openlog Artemis. 

I've been having a look over the maths and can't see an obvious way of determining whether a positive or negative value is required, so I tried the next best obvious thing. I used the Quat6 to calculate a rotation matrix to correct the accelerometer readings and then positioned the sensor in various orientations. Using gravity as a reference vector when the sensor is still, I can confirm that regardless of whether one, two, or all three of the q1-q3 values are negative, a positive q0 is required to give the right rotation matrix so that gravity always points "down" (or at least, in the same direction). 

I therefore suspect, but cannot confirm, that the q1-q3 values are corrected in the DMP so that a positive q0 always gives the right rotation matrix for the sensor xyz axis. In the papers that @PaulZC provides, they highlight that keeping the same values for q1-q3 but flipping the sign of q0 results in the reverse rotation matrix (undoes the original rotation). It might make sense that the designers introduced the positive convention to distinguish between the forward and reverse rotation operations.

As a separate aside, I'm looking to use the IMU and OpenLog Artemis to log details about my run training as a cheap alternative to the Stryd power meter on the market. I'm pretty handy with coding and willing to contribute to the GitHub projects but wouldn't know exactly where to start!

Thanks,
Jon

PaulZC, 2021-05-17:
Hi Jon (@higherstateofawkwardness ),

This is _really_ helpful - thank you. So, helpful that I think we can safely close this issue.

@Zed-svg: please check if this clears up the Q0 issue for you too - and please feel free to reopen if you need more help or advice.

Best wishes,
Paul

closed 2021-05-17 by PaulZC

tosa-no-onchan, 2022-01-13:
I tried the following.
      //double q0 = sqrt(1.0 - ((q1 * q1) + (q2 * q2) + (q3 * q3))); 
      double q0=1.0 - ((q1 * q1) + (q2 * q2) + (q3 * q3));
      if(q0 >= 0.0) q0 = sqrt(q0);
      else q0 = sqrt(q0 * -1.0) * -1.0;

But,the Example6_DMP_Quat9_Orientation dose not work correctly.
Is DMP_Quat9_Orientation fixed all bug!.