sparkfun/SparkFun_ICM-20948_ArduinoLibrary #63: Arduino Uno and the DMP

define-private-public, 2021-05-15:
I'm trying to work on a small visualization application.  Using this library along with Madgwick AHRS, I was able to get my Arduino Uno sketch to output it's orientation on the serial communication line (e.g. Yaw, Pitch, Roll in degrees).  On my PC, I have a python script setup to rotate a 3D model to match that orientation.

While I was initially successful, there was an unfortunate amount of drift, so say every second the computed orientation would drift about 5 seconds.  This isn't acceptable.

I'd really like to use the DMP on the chip.  I read the docs and it looks like there are features such as doing the sensor fusion for me, as well as calibrating the chip (in hopes of preventing drift).  When I tried to enable the DMP in my Arduino Uno Sketch code, it said that there wasn't enough space on the chip.

Does the DMP require a more powerful micro controller than that of the Arduino Uno?

PaulZC, 2021-05-16:
Hi Benjamin (@define-private-public ),

The issue here is that the DMP occupies 14301 bytes of memory - and the Uno's memory is very limited. By default the compiler tries to store the DMP firmware in RAM:

```
text section exceeds available space in board   Sketch uses 34774 bytes (107%) of program storage space. Maximum is 32256 bytes.
Global variables use 15377 bytes (750%) of dynamic memory, leaving -13329 bytes for local variables. Maximum is 2048 bytes.
```

You can force the compiler to store the DMP firmware in program memory (PROGMEM) by: including avr/pgmspace.h; and adding PROGMEM to the dmp3_image definition. Lines 11 and 12 of ICM_20948_C.c become:

```
#include <avr/pgmspace.h>
const uint8_t dmp3_image[] PROGMEM = {
```

But even then, it still exceeds the available memory:

```
Sketch uses 34774 bytes (107%) of program storage space. Maximum is 32256 bytes.text section exceeds available space in board
Global variables use 1075 bytes (52%) of dynamic memory, leaving 973 bytes for local variables. Maximum is 2048 bytes.
```

It might be possible to strip the library down and get it to fit into the Uno memory, but you really won't have much room left for anything else.

Best wishes,
Paul

define-private-public, 2021-05-17:
Would a MEGA work?  I also have a (FRDM-K32L2B)[https://www.nxp.com/design/development-boards/freedom-development-boards/mcu-boards/nxp-freedom-development-platform-for-k32-l2b-mcus:FRDM-K32L2B3] which is way more beefier.  But I'm not much of a hardware microcontroller guy, which is why I mostly stick to Arduino.

I don't think I could simply just copy n' paste the C/C++ files from this over to the FRDM.  What sort of work do you think would need to be done to get that working there?

PaulZC, 2021-05-17:
ATmega2560 should work as you have 256KB of program memory to play with, instead of 32KB.
You will still need to tell the compiler to use PROGMEM to store the DMP firmware though.
Sorry, I have no experience with the FRDM.
Best wishes,
Paul

define-private-public, 2021-05-19:
Hi,

I got an ATmega2560 and it failed to compile as well.  This because there wasn't enough RAM:

```
Arduino: 1.8.13 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Sketch uses 34868 bytes (13%) of program storage space. Maximum is 253952 bytes.
Global variables use 15367 bytes (187%) of dynamic memory, leaving -7175 bytes for local variables. Maximum is 8192 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
data section exceeds available space in board
Error compiling for board Arduino Mega or Mega 2560.
```

This was from trying to compile example no. 6, and turning on ` #define ICM_20948_USE_DMP`.

Is there a good way to put this library on a "diet" to use less memory?

PaulZC, 2021-05-19:
Hi Benjamin (@define-private-public ),

Like I said above, you will need to tell the compiler to use PROGMEM to store the DMP firmware.

Best wishes,
Paul

define-private-public, 2021-05-21:
How exactly do I use PROGMEM with this?   I've never used it before.  Checking the Arduino docs, it looks like I need to specify it on my variables.  Where in your code is the DMP firmware?

PaulZC, 2021-05-21:
Hi Benjamin (@define-private-public ),

Like I said above, you need to edit ICM_20948_C.c and replace this line:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/ffca57bba17eaf5e20af321bcd2a98fb9020e556/src/util/ICM_20948_C.c#L10

with:

```
#include <avr/pgmspace.h>
const uint8_t dmp3_image[] PROGMEM = {
```

You will find ICM_20948_C.c in Arduino\libraries\SparkFun_ICM-20948_ArduinoLibrary\src\util

You also need to edit ICM_20948_C.h and uncomment this line:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/ffca57bba17eaf5e20af321bcd2a98fb9020e556/src/util/ICM_20948_C.h#L29

Best wishes,
Paul

HyperKokichi, 2021-05-28:
Hi Paul and Benjamin,

I've ran into a similar situation as Benjamin(@define-private-public). I'm using an Arduino Mega with the ICM 20948 (SEN-15335).

I edited both ICM_20948_C.c and ICM_20948_C.h just like you said but I still couldn't run example6. 
After some digging I found out that the firmware wasn't being written correcly in the ICM.

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d0b72df898289a047ac9fc4665c68d94f367640e/src/util/ICM_20948_C.c#L1219
This function (_inv_icm20948_firmware_load_) was returning **ICM_20948_Stat_DMPVerifyFail** in line:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d0b72df898289a047ac9fc4665c68d94f367640e/src/util/ICM_20948_C.c#L1292-L1293

The function _inv_icm20948_firmware_load_ copies the firmware 16 bytes at a time to the ICM.
After some more digging I realized the issue was that when you put the firmware in PROGMEM you can't access it in a traditional way. As far as I know, there is no function that copies memory directly from PROGMEM to ICM 
I was able to fix this issue by creating a temporary variable that served as an intermediary between the PROGMEM and the ICM.
I created the temporary variable `unsigned char data_not_pg[INV_MAX_SERIAL_READ];` and replaced line: 
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d0b72df898289a047ac9fc4665c68d94f367640e/src/util/ICM_20948_C.c#L1264
with:
`memcpy_P(data_not_pg, data, write_size);`
`result = inv_icm20948_write_mems(pdev, memaddr, write_size, (unsigned char *)data_not_pg);`

And replaced line:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d0b72df898289a047ac9fc4665c68d94f367640e/src/util/ICM_20948_C.c#L1290-L1293
with:
`memcpy_P(data_not_pg, data, write_size);`
`if (result != ICM_20948_Stat_Ok) flag++;`
`if (memcmp(data_cmp, data_not_pg, write_size)) return ICM_20948_Stat_DMPVerifyFail;`

I'm writing this here for noobs at hardware systems like myself so hopefully you won't have to spend as much time as I did sorting this out.

PaulZC, 2021-05-28:
Hi @HyperKokichi ,

Great detective work! Thank you for submitting this. I will try to add your code - or something similar to it - in the next release.

Best wishes,
Paul

PaulZC, 2021-05-29:
Hi @HyperKokichi / Benjamin (@define-private-public ),

Thanks again for the suggestion. I've added AVR PROGMEM support in v1.2.7. Please see the [release notes](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/releases/tag/v1.2.7) for more info.

Closing... But please re-open if you think I have missed something.

Best wishes,
Paul

closed 2021-05-29 by PaulZC

define-private-public, 2021-05-29:
Thanks for the work on this you two.  I'll be sure to take a look soon.