sparkfun/SparkFun_ICM-20948_ArduinoLibrary #64: Calculating ACC_SCALE & ACC_SCALE2 for the DMP

hubertGir, 2021-05-16:
Hi ! 
First of all, thanks for the tremendous amount of work you've made here ! 

I have a question about the way you calculate the value that will be store in the DMP in order to scale the raw accelerometer data. By default it's set for 4g as it is written in the ICM_20948.ccp file 

Line 1220 :   
```
ICM_20948_fss_t myFSS; // This uses a "Full Scale Settings" structure that can contain values for all configurable sensors
  myFSS.a = gpm4; 
```
and later Line 1287 :

```
  // Configure Accel scaling to DMP
  // The DMP scales accel raw data internally to align 1g as 2^25
  // In order to align internal accel raw data 2^25 = 1g write 0x04000000 when FSR is 4g
  const unsigned char accScale[4] = {0x04, 0x00, 0x00, 0x00};
  result = writeDMPmems(ACC_SCALE, 4, &accScale[0]); if (result > worstResult) worstResult = result; // Write accScale to ACC_SCALE DMP register
  // In order to output hardware unit data as configured FSR write 0x00040000 when FSR is 4g
  const unsigned char accScale2[4] = {0x00, 0x04, 0x00, 0x00};
  result = writeDMPmems(ACC_SCALE2, 4, &accScale2[0]); if (result > worstResult) worstResult = result; // Write accScale2 to ACC_SCALE2 DMP register
```

So, I want to set the full scale setting at +-16g and after I changed the Line 1220 `myFSS.a = gpm4; ` to `myFSS.a = gpm16; `, I wonder now how to calculate the new values of accScale & accScale2 that will be passed to the DMP.

 It states that 1g=2^25 but, for +-4g the value is  0x04000000=2* 2^25 and not 4* 2^25... Same issue for accScale2, I couldn't figure from where 0x00040000 come from and how I can calculate the correct value for my 16g setting.

If you have the way to figured these two values out it would be awesome !

Have a nice day !
Regards

PaulZC, 2021-05-17:
Hi @hubertGir ,

The comments in our library come directly from the InvenSense Application Note:

```
Configure Accel scaling to DMP

The DMP scales accel raw data internally to align 1g as 2^25.

To do this and output hardware unit again as configured FSR, write 0x4000000 to ACC_SCALE DMP register, and write 0x40000 to ACC_SCALE2 DMP register.

ACC_SCALE
Type: Read/Write Register Name Bits[31:0] ACC_SCALE Write accel scaling value for internal use
Description:
In order to align internal accel raw data 2^25 = 1g write 0x4000000 when FSR is 4g.

ACC_SCALE2
Type: Read/Write Register Name Bits[31:0] ACC_SCALE2 Write accel scaling down value
Description:
In order to output hardware unit data as configured FSR write 0x40000 when FSR is 4g.
```

There are more comments in the InvenSense example code (ZaneL's Teensy-ICM-20948 library is a wrapper for the InvenSense code):

https://github.com/ZaneL/Teensy-ICM-20948/blob/0d24a64e57c27b0003f3114197ddf8d53dd6afba/src/Icm20948Dmp3Driver.c#L1005-L1042

https://github.com/ZaneL/Teensy-ICM-20948/blob/0d24a64e57c27b0003f3114197ddf8d53dd6afba/src/Icm20948Dmp3Driver.c#L1053-L1084

Hopefully these should give you what you need?

Best wishes,
Paul

closed 2021-05-17 by PaulZC

Trey-Raventor, 2022-08-31:
Hey @hubertGir,
Can you share your experience on converting the Teensy scales? I am trying to test out some different configurations and am having a bit of trouble converting the scale values.

I have made this so far to take scale values from Zane's switch cases:
typedef unsigned char     uint8_t;
#include <stdio.h>
	unsigned char big8[4];
	int result;
	long x;
	int i;
int main()
{
    x =  536870912L;
	big8[0] = (uint8_t)((x >> 24) & 0xff);
	big8[1] = (uint8_t)((x >> 16) & 0xff);
	big8[2] = (uint8_t)((x >> 8) & 0xff);
	big8[3] = (uint8_t)(x & 0xff);
	//printf("%s\n", big8);
for (i = 0; i < 4; i++) {
  printf("%d ", big8[i]);
}
}