sparkfun/SparkFun_ICM-20948_ArduinoLibrary #68: All values are very inaccurate in Example2_Advanced

Bascy, 2021-05-24:
### Subject of the issue
When the advanced example is run (including fix needed, see PR https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/pull/67) on a stationary board, especially Accellero are far from stationary, even after rotating device on all axis to calibrate.

Magneto values seem pretty stable,
AccX values run from 1 to 31, AccY from 83 to 110, AccZ values from 1041 to 1070
Gyro all directions jump from 0 to 3 degrees per second

Running a  espressif32@~3.2.0 with the Arduino framework, using address $68

`Scaled. Acc (mg) [ -00011.72, -00083.98,  01050.78 ], Gyr (DPS) [ -00001.27, -00001.08,  00001.19 ], Mag (uT) [  00013.50, -00006.00,  00083.25 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00001.46, -00096.19,  01068.85 ], Gyr (DPS) [ -00001.02,  00001.76,  00001.42 ], Mag (uT) [  00013.65, -00005.55,  00084.15 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00014.65, -00088.87,  01050.29 ], Gyr (DPS) [ -00002.46, -00000.62, -00000.85 ], Mag (uT) [  00014.55, -00006.75,  00083.25 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00031.25, -00084.96,  01056.64 ], Gyr (DPS) [ -00000.56, -00001.31, -00002.40 ], Mag (uT) [  00014.85, -00005.40,  00084.30 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00026.86, -00084.96,  01057.13 ], Gyr (DPS) [ -00001.39,  00000.26,  00001.07 ], Mag (uT) [  00014.70, -00005.55,  00083.40 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00008.79, -00100.10,  01067.87 ], Gyr (DPS) [ -00000.09, -00000.11,  00001.72 ], Mag (uT) [  00014.40, -00005.10,  00084.90 ], Tmp (C) [  00028.32 ]
Scaled. Acc (mg) [ -00016.60, -00097.66,  01065.92 ], Gyr (DPS) [  00000.03, -00001.97,  00001.38 ], Mag (uT) [  00014.85, -00006.75,  00084.75 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00030.27, -00090.82,  01060.55 ], Gyr (DPS) [ -00000.85, -00000.98, -00003.08 ], Mag (uT) [  00014.25, -00005.70,  00085.35 ], Tmp (C) [  00028.60 ]
Scaled. Acc (mg) [ -00012.70, -00091.31,  01051.27 ], Gyr (DPS) [ -00000.93,  00000.76, -00000.29 ], Mag (uT) [  00014.25, -00005.40,  00083.55 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00017.09, -00089.84,  01058.11 ], Gyr (DPS) [ -00001.10, -00000.85, -00000.27 ], Mag (uT) [  00014.40, -00005.85,  00084.15 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00023.93, -00093.75,  01055.18 ], Gyr (DPS) [ -00002.36,  00001.12, -00000.37 ], Mag (uT) [  00013.35, -00005.55,  00084.90 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00011.72, -00082.52,  01056.64 ], Gyr (DPS) [ -00001.34, -00000.69, -00002.05 ], Mag (uT) [  00013.65, -00005.85,  00083.40 ], Tmp (C) [  00028.60 ]
Scaled. Acc (mg) [ -00018.07, -00082.52,  01064.45 ], Gyr (DPS) [ -00001.95, -00000.92, -00002.27 ], Mag (uT) [  00014.85, -00006.45,  00084.00 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [  00001.95, -00076.66,  01049.32 ], Gyr (DPS) [ -00003.15,  00000.73,  00000.76 ], Mag (uT) [  00013.80, -00007.20,  00083.55 ], Tmp (C) [  00028.22 ]
Scaled. Acc (mg) [ -00021.48, -00083.01,  01050.78 ], Gyr (DPS) [ -00001.37, -00000.73,  00000.25 ], Mag (uT) [  00015.45, -00007.05,  00084.15 ], Tmp (C) [  00027.98 ]
Scaled. Acc (mg) [ -00009.77, -00087.89,  01062.50 ], Gyr (DPS) [ -00004.31,  00002.25, -00001.50 ], Mag (uT) [  00014.55, -00007.50,  00085.05 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00013.18, -00096.68,  01060.55 ], Gyr (DPS) [ -00000.85,  00000.46,  00001.74 ], Mag (uT) [  00015.45, -00007.35,  00083.85 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00024.90, -00096.19,  01058.11 ], Gyr (DPS) [ -00002.50,  00000.95,  00001.44 ], Mag (uT) [  00014.55, -00005.40,  00085.05 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00018.07, -00100.10,  01061.52 ], Gyr (DPS) [ -00002.05,  00000.30, -00000.54 ], Mag (uT) [  00014.85, -00005.40,  00084.75 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00033.20, -00095.70,  01040.53 ], Gyr (DPS) [ -00000.53,  00000.99,  00001.15 ], Mag (uT) [  00015.60, -00006.75,  00084.30 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00017.58, -00094.73,  01059.08 ], Gyr (DPS) [ -00002.02,  00002.08, -00000.01 ], Mag (uT) [  00013.65, -00005.85,  00084.90 ], Tmp (C) [  00028.17 ]
Scaled. Acc (mg) [ -00014.16, -00091.31,  01061.04 ], Gyr (DPS) [ -00001.69,  00001.98, -00001.17 ], Mag (uT) [  00015.45, -00006.30,  00084.45 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00016.60, -00091.80,  01062.01 ], Gyr (DPS) [ -00001.46,  00001.24, -00001.16 ], Mag (uT) [  00014.40, -00006.30,  00084.90 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00031.25, -00110.35,  01069.34 ], Gyr (DPS) [ -00001.66,  00000.60, -00003.24 ], Mag (uT) [  00013.50, -00006.75,  00084.00 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00014.65, -00087.89,  01054.69 ], Gyr (DPS) [ -00002.19,  00000.49, -00000.58 ], Mag (uT) [  00012.90, -00006.60,  00083.85 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00024.90, -00090.82,  01044.92 ], Gyr (DPS) [ -00000.14, -00000.18,  00001.03 ], Mag (uT) [  00015.15, -00006.60,  00084.45 ], Tmp (C) [  00028.60 ]
Scaled. Acc (mg) [ -00023.44, -00090.82,  01053.22 ], Gyr (DPS) [ -00002.63,  00000.31, -00002.95 ], Mag (uT) [  00014.70, -00007.35,  00084.15 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00028.32, -00091.31,  01064.94 ], Gyr (DPS) [ -00001.31,  00001.27, -00002.93 ], Mag (uT) [  00013.65, -00006.60,  00084.45 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00026.86, -00098.14,  01066.41 ], Gyr (DPS) [ -00002.60,  00000.74,  00001.02 ], Mag (uT) [  00013.35, -00007.05,  00083.85 ], Tmp (C) [  00028.17 ]
Scaled. Acc (mg) [ -00010.74, -00098.14,  01050.78 ], Gyr (DPS) [ -00001.30, -00000.18, -00002.82 ], Mag (uT) [  00013.80, -00006.75,  00085.05 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00025.39, -00100.59,  01045.90 ], Gyr (DPS) [ -00002.27,  00001.02, -00000.79 ], Mag (uT) [  00013.80, -00004.95,  00085.80 ], Tmp (C) [  00028.60 ]
Scaled. Acc (mg) [ -00005.37, -00086.91,  01048.34 ], Gyr (DPS) [ -00001.23, -00000.21,  00000.24 ], Mag (uT) [  00014.70, -00006.60,  00084.90 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00027.34, -00088.38,  01057.13 ], Gyr (DPS) [ -00001.63,  00001.21, -00002.69 ], Mag (uT) [  00015.30, -00005.70,  00084.30 ], Tmp (C) [  00028.60 ]
Scaled. Acc (mg) [ -00021.48, -00089.84,  01053.71 ], Gyr (DPS) [ -00002.62,  00001.20, -00000.58 ], Mag (uT) [  00013.95, -00005.85,  00085.20 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00024.90, -00099.12,  01058.59 ], Gyr (DPS) [ -00001.95,  00000.44, -00001.21 ], Mag (uT) [  00012.75, -00005.70,  00084.00 ], Tmp (C) [  00028.56 ]
Scaled. Acc (mg) [ -00013.18, -00091.80,  01039.55 ], Gyr (DPS) [ -00001.76,  00001.09, -00001.30 ], Mag (uT) [  00015.15, -00004.80,  00083.85 ], Tmp (C) [  00028.08 ]
Scaled. Acc (mg) [ -00022.95, -00087.89,  01051.27 ], Gyr (DPS) [ -00001.89,  00000.86,  00000.03 ], Mag (uT) [  00014.70, -00006.60,  00084.45 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00017.58, -00088.87,  01050.29 ], Gyr (DPS) [ -00000.75, -00000.11, -00000.33 ], Mag (uT) [  00014.25, -00005.70,  00084.30 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00014.65, -00093.26,  01057.13 ], Gyr (DPS) [ -00001.30,  00002.66,  00001.25 ], Mag (uT) [  00013.80, -00006.75,  00085.05 ], Tmp (C) [  00028.37 ]
Scaled. Acc (mg) [ -00011.72, -00103.52,  01066.89 ], Gyr (DPS) [ -00000.27,  00002.05,  00000.61 ], Mag (uT) [  00013.80, -00006.00,  00084.30 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00024.41, -00087.89,  01053.71 ], Gyr (DPS) [ -00000.12,  00000.09, -00000.20 ], Mag (uT) [  00014.55, -00006.45,  00084.30 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00016.11, -00087.40,  01057.62 ], Gyr (DPS) [ -00003.13,  00000.49, -00001.82 ], Mag (uT) [  00014.85, -00005.70,  00083.55 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00010.74, -00093.26,  01056.64 ], Gyr (DPS) [ -00001.17,  00001.82,  00000.02 ], Mag (uT) [  00014.55, -00006.75,  00084.75 ], Tmp (C) [  00028.32 ]
Scaled. Acc (mg) [ -00026.37, -00097.66,  01055.66 ], Gyr (DPS) [ -00003.73,  00000.74, -00002.21 ], Mag (uT) [  00014.55, -00006.00,  00084.75 ], Tmp (C) [  00028.22 ]
Scaled. Acc (mg) [ -00030.27, -00095.21,  01059.57 ], Gyr (DPS) [ -00001.27,  00000.78,  00000.74 ], Mag (uT) [  00014.55, -00005.70,  00084.75 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00023.44, -00092.77,  01058.59 ], Gyr (DPS) [ -00001.22,  00000.73,  00002.84 ], Mag (uT) [  00013.95, -00005.10,  00084.45 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00018.55, -00083.50,  01041.02 ], Gyr (DPS) [ -00002.71, -00000.78, -00000.98 ], Mag (uT) [  00015.15, -00006.30,  00083.85 ], Tmp (C) [  00028.51 ]
Scaled. Acc (mg) [ -00020.02, -00083.50,  01043.46 ], Gyr (DPS) [ -00002.50,  00000.82, -00001.18 ], Mag (uT) [  00013.65, -00006.30,  00085.20 ], Tmp (C) [  00028.41 ]
Scaled. Acc (mg) [ -00024.41, -00086.43,  01070.80 ], Gyr (DPS) [ -00000.73,  00000.52, -00003.96 ], Mag (uT) [  00014.55, -00004.95,  00084.00 ], Tmp (C) [  00028.51 ]`

PaulZC, 2021-05-24:
Hi @Bascy ,
What happens if you change the DLP configuration? Please try a lower bandwidth and report what happens.
Best wishes,
Paul

Bascy, 2021-05-24:
Hmm .. I think I missread the readings. As they are in mg they it is not as bad as I orginally thought ... But still, when converting the accl values to tilt angels there is more than 1 degree 
.

I've tried DLP setting acc_d5bw7_n8bw3 and this does increase the stability of the readings a bit

PaulZC, 2021-05-24:
OK. Thanks for letting me know. Closing - as this is appears not to be an issue with the library,
Paul

closed 2021-05-24 by PaulZC

Bascy, 2021-05-24:
Well, the tilt angle is computed directly from the AccX, Y and Z values .. so it is still their instability that is the main cause of the problem.
Is there any other way I can try to make it more stable?