sparkfun/SparkFun_ICM-20948_ArduinoLibrary #71: Bank change to 0 after reset is omitted

peterleif, 2021-06-07:
### Subject of the issue
This library implements a mechanism to reduce the amount of I2C traffic by caching the previously used register bank and skip the bank change operation when it is not required.

Currently, no initialization method for `ICM_20948_Device_t` is provided and therefore all members including `_last_bank` is initialized as `0` (if declared as a global or static variable).

It means that if the first register access uses register bank 0 no bank change operation is performed.

This behavior can be problematic if the MCU is reset while the power to ICM-20948 is kept on.
Suppose the MCU is performing register access using bank 2, for example, and then the MCU resets, and the first register access after the reset is intended to use register bank 0.
Since the register bank change operation is skipped, the register access is recognized as access to the register bank 2.

To prevent this problem, I would suggest implementing an initialization method for `ICM_20948_Device_t` that sets an invalid value to `_last_bank` so the first register bank change operation is performed 100%.

### Your workbench
* What platform are you using?: Custom board
* What version of the device are you using? Is there a firmware version?: STM32L4 MCU, FreeRTOS (I only use the underlying C layer of the library)
* How is the device wired to your platform?: Custom PCB
* How is everything being powered?: Battery powered, ICM-20948 is always on
* Are there any additional details that may help us help you?

### Steps to reproduce
- Use the C layer of the library and declare `ICM_20948_Device_t` as a global variable
- Keep ICM-20948 always powered on
- Program MCU so it accesses the bank 0 and then changes to another bank (e.g. bank 2)
- Reset MCU after it accesses a register bank other than bank 0

### Expected behavior
The MCU should access bank 0 after the reset.

### Actual behavior
The MCU accesses bank 2 since no bank change operation is performed after the reset.

PaulZC, 2021-06-07:
Hi Etsushi (@peterleif ),
Great suggestion - thank you. Would you like to send a Pull Request with the changes?
Best wishes,
Paul

peterleif, 2021-06-07:
Hi Paul (@PaulZC ),
I would like to send a PR.
I don't have an Arduino test environment, so please forgive me for not being able to do adequate testing.

Best wishes,
Etsushi

PaulZC, 2021-11-15:
Merged in v1.2.8. Closing...

PR #72 (2021-06-09): ICM_20948_Device_t with _last_bank set to an invalid number
Resolves #71 
- Implement `ICM_20948_init_struct` function
- Initialize `_last_bank` to `4` (invalid bank number) 
- Update C++ constructor to use `ICM_20948_init_struct`
- Update the Example999
closed 2021-11-15 by PaulZC