sparkfun/SparkFun_ICM-20948_ArduinoLibrary #74: 9DoF Quaternion Values switch signs

A6Deg, 2021-06-17:
Hey Guys,

I have been working on a quaternion project for quite some time and just recently saw the updated library (1.2.7) and have to say great work!!! 👏🏼👏🏼👏🏼

I do have an issue ide like to discuss here:

Using:
SparkFun Pro nrf52840 Mini
ICM - 20948 connected I2C via qwiik
Bootloader - s140 6.1.1 r0
ICM Library - 1.2.7
Used the example 6 - 9DoF Quat. 

I have 2 devices located on my left and right leg that transmit the quaternions to a host device via BLE. 

I have noticed that the right leg is working perfectly as it is supposed to while kicking forward and sideways, but the left leg is acting weird especially when kicking to the side (to the right side).

When a kick is noticed the Quaternions (Q1 Q2 Q3) switch signs 😲

This Image is the RIGHT leg. (The deltas we see are due to a lag in data coming in.. please ignore). This is what the data should look like.

[image]

This is the LEFT leg.  As you can see the value of Q3 goes from (-0.893) to (0.893) when kicking to the right.

[image]

This is the Data

[image]

This is when I take my leg down after the kick it goes back to normal

[image]

The code on the Devices is the same:
Initiating the DMP exactly the same as in the Example, and grabbing the DMP data exactly the same as the example.

Any help is appreciated!

Thanks
Almog

PaulZC, 2021-06-17:
Hi Almog (@A6Deg ),
I have seen the same thing, a pair of Quats suddenly inverting, even with the ICM sitting stationary on my desk.
All I can say is that that is the data that is being returned by the DMP... Why does it do it? I don't know...
My knowledge of quaternion math is rather basic, but, I would ask: does it matter if a _pair_ of quaternions invert? Does that get you back where you started from?
Best wishes,
Paul

A6Deg, 2021-06-17:
Hey Paul,
Unfortunately, it does matter because the quaternions I'm sending are being interpreted as rotations in a 3D game. When Q1, Q2 & Q3 suddenly Invert the rotation in the game gets thrown off completely. 
Only when my leg goes down a bit do the quaternions reinvert and the game rotation goes back to normal.
It may be hardware-related, I'm going to switch between my left and right bands and see if the problem repeats itself.
Will update.
Almog

PaulZC, 2021-06-17:
Please also try Quat6 (Game Rotation Vector). Do you see the same thing there? I really can't promise to fix this, I have no idea what is causing it, but it would be good to understand if it is limited to Quat9.
All the best,
Paul

A6Deg, 2021-07-04:
Hey Paul,
Sorry for the long delay between replies.
We have just seen the same thing happen with the Quat6 Example :/
Almog

PaulZC, 2021-07-04:
Hi Almog (@A6Deg ),
(No worries about the delay!)
I was worried you might say that... I really don't know what to suggest. I suspect this is a feature of the DMP that we have no control over.
One thing you could try is running [this wrapper for the InvenSense code](https://github.com/isouriadakis/Arduino_ICM20948_DMP_Full-Function). I normally recommend ZaneL's wrapper, but that only supports SPI. If you still see the sign switch then you can be confident that this is a feature of the DMP that we can probably do nothing about. If you don't see it then maybe, just maybe, it is a small error in the way we have the DMP configured...
Best of luck - and thanks again for the useful feedback,
Paul

closed 2023-03-19 by A6Deg