sparkfun/SparkFun_ICM-20948_ArduinoLibrary #75: (DMP) Step Counter / Detector appear to not work

xavasxxv, 2021-06-21:
### Subject of the issue
Enabling either step counter or step detection features on the DMP appear non funcional

### Your workbench
* What platform are you using?
Custom board w/ nRF52832
* What version of the device are you using? Is there a firmware version?
N/A
* How is the device wired to your platform?
SPI @ 7MHz using a level shifter
* How is everything being powered?
Li-Ion Battery, w/ 3.3V & 1.8V switch-mode regulators
* Are there any additional details that may help us help you?
I may just be being a dunce enabling the features :P... But quaternions, accelerometer, gyro and magnetometer are working as intended.... Also, i'm using interrupts instead of polling the data from the FIFO

### Steps to reproduce
Using example 10, enable the step counter or step detection feature as below
```c
   // Enable the DMP Game Rotation Vector sensor (Quat6)
  // success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_GAME_ROTATION_VECTOR) == ICM_20948_Stat_Ok);

  // Enable additional sensors / features
  //success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_RAW_GYROSCOPE) == ICM_20948_Stat_Ok);
  // success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_RAW_ACCELEROMETER) == ICM_20948_Stat_Ok);
  //success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_MAGNETIC_FIELD_UNCALIBRATED) == ICM_20948_Stat_Ok);
  success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_STEP_DETECTOR) == ICM_20948_Stat_Ok);
  success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_STEP_COUNTER) == ICM_20948_Stat_Ok);
  success &= (myICM.enableDMPSensorInt(INV_ICM20948_SENSOR_STEP_DETECTOR) == ICM_20948_Stat_Ok);

  // Configuring DMP to output data at multiple ODRs:
  // DMP is capable of outputting multiple sensor data at different rates to FIFO.
  // Setting value can be calculated as follows:
  // Value = (DMP running rate / ODR ) - 1
  // E.g. For a 225Hz ODR rate when DMP is running at 225Hz, value = (225/225) - 1 = 0.
  // success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Quat6, 0) == ICM_20948_Stat_Ok); // Set to 225Hz
  // success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Accel, 0) == ICM_20948_Stat_Ok); // Set to 225Hz
  //success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Gyro, 0) == ICM_20948_Stat_Ok); // Set to 225Hz
  //success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Cpass, 0) == ICM_20948_Stat_Ok); // Set to 225Hz
```

How i'm processing the data on my loop
```c
    // works
    if ((dmp_data.header & DMP_header_bitmap_Compass_Calibr) > 0) { // Check for Compass
      float scale_div = 1.0f / (1 << 16);
      pout->mag.axes.z = dmp_data.Compass_Calibr.Data.Z * scale_div;
      pout->mag.axes.x = dmp_data.Compass_Calibr.Data.X * scale_div;
      pout->mag.axes.y = dmp_data.Compass_Calibr.Data.Y * scale_div;
    }
    // doesn't work
    if ((dmp_data.header & (DMP_header_bitmap_Step_Detector | DMP_header_bitmap_PQuat6)) > 0) { // Check for Step Detector
      static uint64_t steps = 0;
      steps++;
      printf("step %lld %lu\n", steps, dmp_data.Pedometer_Timestamp);
      nrf_gpio_pin_toggle(LED_BT);
    }
```

### Expected behaviour
I should be able to extract step data ou have an interrupt generated from the IC

### Actual behaviour
Very sporadically, i get garbage on the data read from the FIFO relative to the steps, but i can't pinpoint times or movements to trigger it

I know that these features aren't guaranteed to be working as said on the [DMP.md](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/DMP.md), but either a few pointer to something i might be doing that's dumb or a bit of help would be much appreciated...

Also, idk if it's of interest, but, i found [this git](https://github.com/isouriadakis/Arduino_ICM20948_DMP_Full-Function) that claims it has support for the step counting feature, but unfortunately i can't test it cause i don't have any breakout boards with an ICM-20948...

Best regards.
Xavier

PaulZC, 2021-06-21:
Hi Xavier (@xavasxxv ),
If I recall correctly, the DMP code is missing some critical configuration steps which are required for step counting. I was so relieved when I finally got the ‘standard’ DMP modes working that I never went back to revisit the step counter.
Let’s leave this issue open but unfortunately I won’t be able to revisit this any time soon. It will be a few weeks at least. Sorry!
Best wishes,
Paul

xavasxxv, 2021-06-21:
Hey,
1st of all thanks for the quick reply and thanks for your work keeping this library...

Figures that there would be missing something if the code base is as convoluted as you say it is xD
No harm no fowl, I unfortunately can't help much it since the project that this depends is already behind schedule enough... but I'll take a peek every now and then and see if there's any progress...

Best of luck,
Xavier