sparkfun/SparkFun_ICM-20948_ArduinoLibrary #77: Problem reading accurate motorbike tilt angle

fralbo, 2021-07-13:
Hello,

I try to reading motorbike tilt angle and I experiment a problem I also have with MPU-6050 in fact.
These chip seem to work well on desktop but using it on vehicle seem to be a challenge.
As ICM-20948 seems to be usable on drone, I though it was dedicated to give accurate tilt angle but it seems to be more complicated than it appears.

First, I tested it on my desktop, applying the same movements to my Samsung S10+ (embedding a LSM6DSO) and the ICM-20948 breakout.
The result is this one [image]
as you can see, the waveforms perfectly match.

Here, I did the same mounting both devices on my motorbike. [image]
Between 547 and 807, I really was in stable turn, with the same tilt angle as given by my S10+. But the ICM-20948 was seriously drifting to 0 and the offset accumulated at 807 seems to appear at 911.
Between 1639 and 1925, I was accelerating as the turn decreases. S10+ gives an understandable value but the ICM-20948 dramatically drift to 0, probably while the z accel increases.

I guess the problem could come from the pitch drift correction made with the g acceleration. And as on motorbike, as well as on drone, a turn cause the g acceleration to increase, maybe this algorithm causes the pitch reading to be false.

I hope I was clear enough, do you have an idea about the problem and how to solve it?

PaulZC, 2021-07-15:
Hi @fralbo ,

By default, the DMP code in the ICM-20948 library puts the accelerometers into 4g mode (and the gyros into 2000dps mode):

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/486fdbfdf73d8d790aef1903f2d5ceca22a35f46/src/ICM_20948.cpp#L1228-L1241

Maybe more g would help? Please take a look at issue #64.

Best wishes,
Paul

PaulZC, 2021-07-15:
There is some helpful info in #51 too:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/51#issuecomment-859713598

fralbo, 2021-07-15:
Your idea is to decrease the effect of g correction to x/y orientation ?
Do you think changing the scale can change it ? I'm going to try.
Do you know if it's possible to deactivate the automatic correction? Maybe to only enable it when we are just between +/- 2 degrees or so?

fralbo, 2021-07-15:
So, I made the test but that doesn't change the problem.
I guess the correction doesn't  rely on the scaled g value.

PaulZC, 2021-07-15:
Hi @fralbo ,
OK - thank you for trying that.
I do not know what else to suggest - we have no control over the inner workings of the DMP.
All I can suggest is that you try the actual InvenSense code and see if that works for you. Please see #74 for details.
Best wishes,
Paul

fralbo, 2021-07-16:
I tried to use an implementation not using the DMP but it's really really not good!
https://forums.adafruit.com/viewtopic.php?f=19&t=181180&p=881568#p881568
I'm stil searching...

fralbo, 2021-11-21:
Ok so, according to my tests, tilt measurement on motorbike and certainly drones, as they turn banking like a motorcycle, we do not have to use acceleration but gyro only.
So, in `Setup()` function, I use acceleration to measure the initial tilt angle, because the motorbike could be on its lateral crutch, but while driving, I only read the gyro values.
And what about gyro drift ? Yes, I obviously have this problem, of course ! So, I use the tilt angle derivative value. I mean, if I detect that tilt angle does not change for a while, I certainly drive in straight line so, I apply an integral correction to reset the angle to 0.
And that works fine!

closed 2021-11-21 by fralbo

PaulZC, 2021-11-21:
Hi @fralbo ,

Thank you for the update. I am glad that is working for you!

Best wishes,
Paul