sparkfun/SparkFun_ICM-20948_ArduinoLibrary #78: avr/pgmspace.h incorrectly included

spacecheese, 2021-07-14:
### Subject of the issue
I'm using the C backend of this library on a raspberry pi. When compiling using arm-linux-gnueabihf-gcc the __arm__ macro is defined causing ICM_20948_C.c to attempt to include the avr/pgmspace.h header. I've corrected this in my local copy by changing line 11 from:
`#if defined(__AVR__) || defined(__arm__) || defined(__ARDUINO_ARC__)`
to
`#if defined(__AVR__) || defined(__ARDUINO_ARC__)`
I would submit a pull request but I'm not sure whether there was some reason for checking the __arm__ macro?

PaulZC, 2021-07-15:
Hi Space Cheese (@spacecheese ),
Cool handle! ;-)
This is primarily an Arduino library. The three ```if defined``` cover the most common Arduino platforms which need the pgmspace header. If we remove the __arm__ case it is likely to cause the code to fail on some Arduino boards. If you have a _**tested**_ solution which works on both Arduino and the Pi, I'd be happy to take a look.
Best wishes,
Paul

PaulZC, 2021-07-15:
Maybe there is a Pi-specific identifier which we could and into the mix?
```#if (defined(__AVR__) || defined(__arm__) || defined(__ARDUINO_ARC__)) && !defined(__i_am_a_Pi__)```

igor-markovic, 2021-11-15:
Should this issue be closed now that the pull request has been merged?

PaulZC, 2021-11-15:
Hi @igor-markovic ,

Thanks for the reminder!

I merged the changes into the _release_candidate_ branch, but I have not released the changes in the _main_ branch. I left this issue open as a reminder to do the release. I think I was waiting to include #73 first.

I will release the changes soon.

Best wishes,
Paul

PaulZC, 2021-11-15:
Merged in v1.2.8. Closing...

PR #79 (2021-07-15): Don't include pgmspace header on linux targets
Fixes #78 for the Raspberry Pi (and other linux targets).
`#elif defined(__AVR__) || defined(ARDUINO)`
Also works but the ARDUINO macro is seemingly undocumented.
closed 2021-11-15 by PaulZC