sparkfun/SparkFun_ICM-20948_ArduinoLibrary #84: Error verifying image in initialiseDMP

CestLaGalere, 2021-11-25:
### Subject of the issue
Error in the image section of inv_icm20948_firmware_load

### Your workbench
Windows 11,
VS Code (1.62.3)
PlatformIO
with env:
platform = espressif32
board = node32s
framework = arduino
lib_deps = 
	adafruit/Adafruit BusIO@^1.9.3
	sparkfun/SparkFun 9DoF IMU Breakout - ICM 20948 - Arduino Library@^1.2.8

ESP32 ICM20948 on a dev board (many sellers on amazon selling same board)
ESP powered through USB
short I2C cable to ICM20948 - breadboard with same test setup as I've had with older MPU6050 which works fine.

running Example1_Basics.ino

#define ICM_20948_USE_DMP 
#define DMP_CODE_SIZE 14301 

### Steps to reproduce
added debugging code to evaluate every step of inv_icm20948_firmware_load
including a pointer in ICM_20948_Device_t to hold more error info

line  1313
  // Verify DMP memory

  pdev->ThisStep = "Verify DMP firmware";
  data = data_start;
  size = size_start;
  memaddr = load_addr;
  while (size > 0)
  {
    //write_size = min(size, INV_MAX_SERIAL_READ); // Read in chunks of INV_MAX_SERIAL_READ
    if (size <= INV_MAX_SERIAL_READ) // Read in chunks of INV_MAX_SERIAL_READ
      write_size = size;
    else
      write_size = INV_MAX_SERIAL_READ;
    if ((memaddr & 0xff) + write_size > 0x100)
    {
      // Moved across a bank
      write_size = (memaddr & 0xff) + write_size - 0x100;
    }
    result = inv_icm20948_read_mems(pdev, memaddr, write_size, data_cmp);
    if (result != ICM_20948_Stat_Ok)
      flag++; 
#ifdef ICM_20948_USE_PROGMEM_FOR_DMP
    memcpy_P(data_not_pg, data, write_size);  // Suggested by @HyperKokichi in Issue #63
    if (memcmp(data_cmp, data_not_pg, write_size))
#else
    if (memcmp(data_cmp, data, write_size)) // Compare the data
#endif
      {
        char *err = malloc(100);
        snprintf(err, 100, "start %d, size %d, ws %d", size_start, size, write_size);
        pdev->ThisStep  = err;
        return ICM_20948_Stat_DMPVerifyFail;
      }

### Expected behaviour
should verify the DMP image by reading it back

### Actual behaviour
occasionally the verification does run OK and it runs
more often I get an error somewhere through the read / memcmp calls
it is always in teh same place, the err string contains
start 14301, size 14205, ws 16

when it does work, it goes through OK, and runs, though will run for a bit then come up with the message Waiting for data

I don't know if this is an I2C on the limit (I have tried with external 1k pullups as well as using the internal 10k)
the wires to the ESP32 are only 3cm long. It seems strange though that if it fails it is in the same memoery place when verifying the image...

PaulZC, 2021-11-25:
Hi @CestLaGalere ,

Are you using v2.0.1 of the ESP32 core?
Please try reverting to v2.0.0 of the core. v2.0.1 has an 'issue' with I2C clock-stretching. That might explain what you are seeing?

Best wishes,
Paul

CestLaGalere, 2021-11-25:
Thanks, I'll look up at how to do this as I'm unsure how with platformIO, I'll come back when I've tried.

CestLaGalere, 2021-11-26:
Closed - unrelated to library

closed 2021-11-26 by CestLaGalere