sparkfun/SparkFun_ICM-20948_ArduinoLibrary #86: Read Compass unnormally by using SPI protocol

JacksonLtt, 2021-12-16:
### Subject of the issue
I used SPI protocol to read 9-axis from ICM20948. Though reading of gyro and accel is good, reading of compass is always zero. 

### Your workbench
* What platform are you using?
* Arduino IDE to program code
* What version of the device are you using? Is there a firmware version?
* I used NRF52832 feather, and adafruit ICM20948 as IMU
* How is the device wired to your platform?
* SCK pin#12
* MOSI pin#13
* MISO pin#14
* CS pin#27
* How is everything being powered?
* I used USB to power the device
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.
 
### Expected behaviour
Tell us what should happen
Reading of compass should NOT be always zero.

### Actual behaviour
Tell us what happens instead
Reading of compass IS always zero.

PaulZC, 2021-12-16:
Hello Jackson (@JacksonLtt ),

The library works perfectly over SPI. We just need to work out what is different with your setup.

First, please check you have not connected anything to the AUX_SCL or AUX_SDA pins. Those are the internal data connection between the controller (accelerometer and gyro) and the magnetometer. If you connect anything to those pins, you will stop the magnetometer from communicating.

Next please run [Example1_Basics](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/main/examples/Arduino/Example1_Basics/Example1_Basics.ino) but uncomment line 47 first to enable extra debug messages:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/examples/Arduino/Example1_Basics/Example1_Basics.ino#L47

You should see the output shown below. Please copy and paste your output here - that will give us more clues about what is going wrong.

Best wishes,
Paul

```
ICM_20948::startupMagnetometer: successful magWhoIAm after 1 try
Initialization of the sensor returned: All is well.
Waiting for data
Scaled. Acc (mg) [  00013.67,  00014.65,  00995.12 ], Gyr (DPS) [ -00000.87,  00000.61,  00000.83 ], Mag (uT) [ -00016.50,  00034.80,  00028.80 ], Tmp (C) [  00021.70 ]
Scaled. Acc (mg) [  00008.79, -00006.84,  00990.72 ], Gyr (DPS) [ -00001.87,  00000.22,  00001.94 ], Mag (uT) [ -00017.85,  00035.40,  00029.40 ], Tmp (C) [  00022.09 ]
Scaled. Acc (mg) [  00022.95,  00005.86,  00994.63 ], Gyr (DPS) [ -00003.71,  00002.30, -00000.12 ], Mag (uT) [ -00016.35,  00034.95,  00028.65 ], Tmp (C) [  00021.75 ]
Scaled. Acc (mg) [  00010.74,  00011.72,  01001.95 ], Gyr (DPS) [  00001.10, -00000.76,  00003.82 ], Mag (uT) [ -00015.45,  00036.30,  00030.30 ], Tmp (C) [  00021.70 ]
Scaled. Acc (mg) [  00014.16, -00000.49,  01005.37 ], Gyr (DPS) [ -00002.19,  00001.77, -00003.94 ], Mag (uT) [ -00015.90,  00036.30,  00028.05 ], Tmp (C) [  00022.04 ]
Scaled. Acc (mg) [  00017.09,  00001.46,  00996.58 ], Gyr (DPS) [ -00002.57,  00002.56,  00000.06 ], Mag (uT) [ -00016.20,  00034.80,  00028.05 ], Tmp (C) [  00022.18 ]
```

JacksonLtt, 2021-12-16:
Hello Paul,

Thank you for your reply. 
I follow your instruction. I run Example1_Basics and uncomment line 47.
Here is the output that I got.
[image]

Though I could get reading from accel and gyro, reading of magnetometer is still zero. 
However, surprisingly when I run Example_9_DMP_MultipleSensors, the reading of magnetometer is non-zero.  
Here is the output that I got:
[image]

Here is how i connect wire
[image]

Please let me know what I should do.

JacksonLtt, 2021-12-16:
Hello Paul,

I found a strange thing. If i compile Example1_Basics using my MAC book (MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)), reading of magnetometer is not zero.

The output is shown below:
[image]

If I complie Example1_Basics using my Windows desktop, the reading of the magnetometer is always zero.

PaulZC, 2021-12-17:
Hello Jackson (@JacksonLtt ),

I understand what is happening.

The magnetometer needs to be configured differently when you use the DMP. When you have edited ICM_20948_C.h and uncommented [this line](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/src/util/ICM_20948_C.h#L29) to enable support for the DMP, ```.begin``` and ```startupDefault``` then perform a "[minimal startup](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/src/ICM_20948.cpp#L861-L875)". It checks that the magnetometer is connected, but does not configure it. The configuration is performed later by ```initializeDMP```.

If you try to run Example1 with support for the DMP enabled, then, yes, you will see no output from the magnetometer.

I suspect that on your MacBook, you have not yet edited ICM_20948_C.h to enable support for the DMP?

If you want Example1 to run correctly, you can ask the code to perform a full startup by calling ```myICM.startupDefault(false);``` _after_ the ```myICM.begin()```. (But be aware, if you do that, the DMP probably will not work correctly?)

```
#ifdef USE_SPI
    myICM.begin(CS_PIN, SPI_PORT);
#else
    myICM.begin(WIRE_PORT, AD0_VAL);
#endif
    myICM.startupDefault(false); // Force a full - not minimal - startup
```

I am going to close this issue as I believe we have understood and resolved it. But please reopen if you see more problems.

Best wishes,
Paul

closed 2021-12-17 by PaulZC

JacksonLtt, 2021-12-17:
Thanks. it works well!