sparkfun/SparkFun_ICM-20948_ArduinoLibrary #87: How to change SPI speed of

JacksonLtt, 2021-12-17:
How could i change the speed of SPI? I want to increase speed of spi to 7 MHz. How could I achieve it?

PaulZC, 2021-12-18:
I see you found the answer to this question:

https://github.com/JacksonLtt/sign_language/blob/7ebb86e165bc563886061e093ff36393b5663adf/Arduino/lib/SPI/six_imu_spi_dmp/six_imu_spi_dmp.ino#L88

Closing...

closed 2021-12-18 by PaulZC

JacksonLtt, 2021-12-18:
> <sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issue/87/issue_event/5787832751
> @github.com>

Hello Paul,

This is my code. I tried it but it didn’t work.

Best regards,
Taiting

PaulZC, 2021-12-18:
Hello Taiting (@JacksonLtt ),

You can pass the SPI Speed as the third parameter when you call ```.begin``` :

```
myICM1.begin(CS_PIN_1, SPI_PORT, 7000000);
```

Best wishes,
Paul

JacksonLtt, 2021-12-18:
Hello Paul,

I tried this code.
`myICM1.begin(CS_PIN_1, SPI_PORT, 7000000);`
However, I got wrong feedback. Here is my output:
[image]