sparkfun/SparkFun_ICM-20948_ArduinoLibrary #91: Unable to read ICM20948 Using NRF52840 via SPI

JacksonLtt, 2022-04-07:
### Subject of the issue
I tried to use NRF52840 to read data from ICM20948 through SPI protocol, but I found that NRF52840 could not read the data from ICM20948 using SPI.

### Your workbench
* What platform are you using?
I am using Arduino IDE. I used the Sparkfun ICM20948 library.

* What version of the device are you using? Is there a firmware version?
I used NRF52840 Adafruit Feather Express.

* How is the device wired to your platform?
[image]
NRF52840 could not recognize the ICM20948 device.

* How is everything being powered?
I used usb to power the device.

* Are there any additional details that may help us help you?
I has successfully used NRF52840 to read data from ICM20948 using I2C. Additionally, I could used NRF52832 to read data from ICM20948 using I2C and SPI.

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.
I used this firmware to program NRF52840.
https://github.com/JacksonLtt/sign_language/blob/main/Arduino/lib/SPI/nrf_one_imu/nrf_one_imu.ino

### Expected behaviour
The NRF52840 should be able to read data from ICM20948.

### Actual behaviour
[image]
NRF52840 could not recognize the ICM20948 device.

PaulZC, 2022-04-07:
Hello Taiting (@JacksonLtt ),

This sounds the same as issue #88 ?

My answer is the same. The library works well on other platforms using SPI. There must be something unusual about this nRF board which is preventing the code from working.

Best wishes,
Paul

closed 2022-04-07 by PaulZC

JacksonLtt, 2022-04-07:
> <sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issue/91/issue_event/6391888201
> @github.com>

Hello Paul,

I think the #91 and #88 are different questions. For #91, I could use
nrf52832 to read data from one icm20948 data using SPI but I could not use
nrf52840 to read data from one icm20948 using SPI.

makin-stuff, 2022-04-07:
@JacksonLtt Does "Example1_Basics.ino" work in SPI mode using your hardware setup?

Also, what boards package do you have installed to program with the nRF52840? I think @PaulZC  is right that this is likely an issue with the Adafruit nRF board and/or board package you are using.

JacksonLtt, 2022-04-07:
Hello,

Thank you for response:
The Example 1_Basics.ino does not work. 
Here is screen shoot of the serial monitor:
[image]

I am using the Adafruit Feather nRF52840 Express from adafruit:
Here is setting:
[image]

PaulZC, 2022-04-08:
Hello Taiting (@JacksonLtt ),

Can you please confirm if you are using this exact Adafruit board?

https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/pinouts

We need to know exactly which board you are using, because the pin numbering is very different from board to board.

I have Example1 working on a [SparkFun Pro nRF52840 Mini](https://www.sparkfun.com/products/15025) on SPI. But I had to change the pin numbering (in variant.h) to make it work with the Adafruit Feather nRF52840 Express board package.

Best wishes,
Paul

PaulZC, 2022-04-08:
If you are using that exact Adafruit board, then your connections should be:

| Adafruit Feather nRF52840 Express | SparkFun ICM20948 |
| --- | --- |
| MI | MISO |
| MO | MOSI |
| A4 | CS |
| SCK | SCLK |
| GND | GND |
| 3V | 1V8-5V5 |

On the Adafruit board, analog pin A4 is digital pin 18. So you need to change the start of Example1 to:

```
#include "ICM_20948.h" // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

#define USE_SPI       // Uncomment this to use SPI

#define SERIAL_PORT Serial

#define SPI_PORT SPI // Your desired SPI port.       Used only when "USE_SPI" is defined
#define CS_PIN 18    // Which pin you connect CS to. Used only when "USE_SPI" is defined
```

Please let us know if this solves your issue.

Best wishes,
Paul

JacksonLtt, 2022-04-08:
Hello Paul,

Thank you for your response.
The problem is solved. I did not setup correct pin of CS_PIN.

Best regards,
Taiting