sparkfun/SparkFun_ICM-20948_ArduinoLibrary #97: Wrong ID message with "SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)" and "Arduino Uno Mini"

vgarciavazquez, 2022-06-27:
Hello Sparkfun ICM-20948 team

I am trying to obtain data from "SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)" using Arduino Uno Mini and SPI, but I obtain the message "Wrong ID".

My settings are as follows:
- "Arduino IDE" version 1.8.19 with "Arduino AVR Boards" 1.8.5 and "Sparkfun 9DoF IMU Breakout - ICM 20948" version 1.2.9
- Board selected in Arduino IDE: Arduino Uno Mini
- Upload the program Example1_Basics.ino to Arduino Uno Mini after uncommenting "#define USE_SPI" and replacing "#define CS_PIN 2" to "#define CS_PIN 10"
- The connections between Arduino Uno Mini and Sparkfun 9DoF IMU Breakout as follows:

ARDUINO UNO Mini ------ SparkFun 9DoF IMU Breakout - ICM-20948
          D10 (SS)          ------          SPI – CS
          D12 (CIPO)      ------          SPI – MISO
          D13 (SCK)        ------          SPI – SCLK
          D11 (COPI)      ------          SPI – MOSI
          3V3                  ------          SPI – 1V8-5V5
          GND                ------          SPI – GND

The expected behaviour would be to obtain the IMU data but what I obtain is in the Arduino IDE Serial Monitor Tool:

"Initialization of the sensor returned: Wrong ID
Trying again...
Initialization of the sensor returned: Wrong ID
Trying again.⸮⸮'K⸮⸮⸮⸮⸮⸮⸮⸮'K⸮⸮⸮΁⸮⸮⸮S1Нh⸮	⸮⸮hp⸮⸮⸮⸮⸮&"

If I uncomment myICM.enableDebugging(); the output in the Arduino IDE Serial Monitor Tool:

ICM_20948::checkID: ICM_20948_check_id returned: Wrong ID
ICM_20948::startupDefault: checkID returned: Wrong ID
ICM_20948_SPI:⸮⸮⸮8⸮⸮⸮h⸮2⸮⸮⸮⸮d⸮⸮Y⸮~]@⸮?Æ⸮⸮<K0⸮⸮⸮V?Þ6⸮

I have read the issue: https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/24 but no idea what I am doing wrong.

Why I receive a wrong ID? 
How can I test that “SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)” is working fine and is not damaged?

Thank you so much!

PaulZC, 2022-06-27:
Hello @vgarciavazquez ,

The Arduino UNO Mini is a 5V board. It has 5V I/O pins. The ICM20948 is a 3.3V board. You are trying to interface 3.3V SPI signals to 5V pins. That will cause all kinds of problems.

You need a 4-channel level shifter to make this work.

https://www.sparkfun.com/products/19626

https://www.sparkfun.com/products/12009

Best wishes,
Paul

PaulZC, 2022-06-27:
Ah. Sorry. My mistake. The ICM20948 breakout has level-shifters on-board. The 20948 is actually a 1.8V device.

Please try connecting the ICM20948 VIN pin to 5V - not 3.3V. It should work?

Best wishes,
Paul

vgarciavazquez, 2022-06-27:
Thank you so much, Paul, for your suggestion.

I have connected Vin to Arduino Mini 5V and I have the same message: _Wrong ID_ :-(

PaulZC, 2022-06-27:
Please check the ADR jumper:

“When open (default) the address of the ICM is 0x69 and it is possible to use SPI communication. When soldered closed the address changes to 0x68. Closing the jumper prevents you from using SPI.“

Check the “jumpers” section in the hook-up guide:

https://learn.sparkfun.com/tutorials/sparkfun-9dof-imu-icm-20948-breakout-hookup-guide

PaulZC, 2022-06-27:
Or try:

#define CS_PIN SS

vgarciavazquez, 2022-06-28:
Thanks a lot for your suggestions Paul.

I have checked the ARD jumper and it is open so it would be possible to use SPI communication.

I have also replaced _#define CS_PIN 10_ with _#define CS_PIN SS_ but I have the same error :-(.

Why do I have the _Wrong ID message_? How can I check that the _SparkFun 9DoF IMU Breakout - ICM-20948_ is not damaged?

PaulZC, 2022-06-28:
Hi @vgarciavazquez ,

The “wrong ID” message indicates a communication error. The code is not reading the correct identifier number from the chip. This happens when the SPI connections or pin definitions are not correct.

Please check the CIPO and COPI connections. Perhaps you have them swapped?

You are the first person I have talked to who is using the Arduino Mini. Perhaps there is something different with its configuration? I will test it - on Saturday. I do not have hardware with me at the moment.

We do not have any stock of the ICM-20948. It is not available at the moment. If it is a hardware fault, we can only offer you a refund, not a replacement. Sorry!

Best wishes,
Paul

vgarciavazquez, 2022-06-28:
Paul, I have also swapped CIPO and COPI connections and the same message... I have rechecked all connections and I have as follows.

ARDUINO UNO Mini ------ SparkFun 9DoF IMU Breakout - ICM-20948
D10 (SS) ------ SPI – CS
D12 (CIPO) ------ SPI – MISO
D13 (SCK) ------ SPI – SCLK
D11 (COPI) ------ SPI – MOSI
5V ------ SPI – 1V8-5V5
GND ------ SPI – GND

The only thing that I can see is that SparkFun 9DoF IMU Breakout is power on (LED ON).

Thanks a lot for testing this Saturday.

Regarding a hardware fault, I bought _SparkFun Qwiic SHIM Kit for Raspberry Pi_ that is available (https://www.sparkfun.com/products/16987) that includes a _9DoF IMU Breakout - ICM-20948_. Could I have a replacement of the whole kit just in case it is a hardware fault?

All the best

PaulZC, 2022-06-28:
Can you test the ICM20948 with the Qwiic Shim? It uses the I2C interface instead of SPI, but will prove if the chip is functioning. Please try it if you can.

vgarciavazquez, 2022-06-28:
Paul, can I connect the Qwiic Shim to Arduino Uno Mini? I have seen that it is for Raspberry PI. 

Another option is to use I2C interface if I do the following connections and upload the original _Example1_Basics.ino_ to Arduino Uno Mini:

ARDUINO UNO Mini ------     SparkFun 9DoF IMU Breakout - ICM-20948
D18 (SDA)                  ------     I2C – SDA
D19 (SCL)                   ------     I2C – SCL
3.3V                            ------     I2C – VIN (https://www.sparkfun.com/products/15335 specifies 1.95 V to 3.6 V supply voltage)
GND                           ------     I2C – GND

Is it right?

PaulZC, 2022-06-28:
Hi @vgarciavazquez ,

OK. That’s good. It proves the ICM-20948 is working.

Yes, your connections are OK. And, yes, it is OK to connect VIN to 5V. (The 3.6V must be a typo. Both the schematic and PCB say the maximum is 5.5V.)

But the ‘garbage’ for the gyro and magnetometer data is very confusing. I have not seen that before.

I do not know what to suggest. It looks like the ICM is OK, but that there is something strange about your Arduino Mini?

Is the ICM the only thing you have connected? Have you also connected the LCD from your kit? You must not connect the LCD. It is 3.3V only and your Arduino Mini is 5V. Only the ICM breakout is compatible with 5V.

Paul

vgarciavazquez, 2022-06-28:
Hi Paul,

Thanks a lot for your support!

Regarding the Arduino Uno Mini, I am just connecting the ICM-20948 to the Arduino, nothing else (not the LCD). The results are displayed on _Arduino IDE - Serial Monitor Tool_. 

What board have you used to test _SparkFun 9DoF IMU Breakout - ICM-20948_ other times? I selected Arduino Uno Mini since I need that the board is small for my application and this board has SPI and I2C communication (https://docs.arduino.cc/hardware/uno-mini-le).

All the best,

Veronica

PaulZC, 2022-07-02:
Hi Veronica (@vgarciavazquez ),

I have access to my hardware again...

I've tried to replicate your issue - but I've failed so far.

I'm using Example1 from the library with USE_SPI and with CS_PIN set to 10:
```
#define USE_SPI       // Uncomment this to use SPI

#define SERIAL_PORT Serial

#define SPI_PORT SPI // Your desired SPI port.       Used only when "USE_SPI" is defined
#define CS_PIN 10     // Which pin you connect CS to. Used only when "USE_SPI" is defined
```

I'm using an old 5V RedBoard which has the same voltages and ATmega328P processor as the Arduino Uno Mini.

I'm using the connections:

RedBoard ------ SparkFun 9DoF IMU Breakout - ICM-20948
D10 (SS) ------ SPI – CS
D12 (CIPO) ------ SPI – MISO
D13 (SCK) ------ SPI – SCLK
D11 (COPI) ------ SPI – MOSI
5V ------ SPI – VIN - 1V8-5V5
GND ------ SPI – GND

I've tried the SparkFun AVR Board definitions, and the Arduino AVR Board definitions (for both Uno and Uno Mini) and I'm getting clean data in all cases:

[image]

I really don't know what to suggest. There must be something 'unusual' about your Uno Mini but I really cannot work out why your Serial prints are being scrambled.

Are you sure you are running Example1 with only those two changes? (Maybe you have added extra code which is causing it to run out of memory? Maybe you are trying to use SoftwareSerial?)

The Teensy 4.1 is a good choice - it's a real powerhouse. Possibly overkill for your project? The Teensy LC is smaller and should work perfectly for your project. Both are 3.3V boards - you will need to connect the ICM-20948 VIN to 3.3V - not 5V.

Apologies again. I really don't know what else to suggest...

Best wishes,
Paul

vgarciavazquez, 2022-07-04:
Hi Paul,

Thanks a million for your tests last Saturday. 

I am running Example1_Basics with just those two changes when I use SPI and without any change when I use I2C. I am not using SoftwareSerial.

I have bought Teensy LC (https://store.arduino.cc/products/teensy-lc-usb-development-board?queryID=undefined). I hope I can test it by Wednesday.

All the best,

Verónica

vgarciavazquez, 2022-07-06:
Hi Paul, 

Example1_Basics works with Teensy LC using SPI:

[image]

I have also tried Example6_DMP_Quat9_Orientation (also using SPI):

[image]

So the problem was the Arduino UNO Mini (maybe mine). I have a weird behaviour. When I connect the _Arduino UNO Mini_ to _SparkFun 9DoF IMU Breakout - ICM-20948_ via a protoboard I had the _Wrong ID_ message. On the other hand, if I connect directly the _Arduino UNO Mini_ to _SparkFun 9DoF IMU Breakout - ICM-20948_ I have 

_Initialization of the sensor returned: All is well. Waiting for data Scaled. Acc (mg) [ 00637.21, 00754.88, 00161.62 ], Gyr00001.08, -00002.69 ], Ma⸮⸮⸮B&(⸮M#� �`$� Z(⸮�Rb ��⸮<�0_

No idea what it is going on with the Arduinoi board (I have check that the connections are ok using a multimeter).

Thanks a million, Paul! I will use Teensy LC for my project now!

All the best,

Veronica

PaulZC, 2022-07-06:
Hi Veronica,

I'm glad the Teensy LC is working for you!

Very best wishes,
Paul

closed 2022-07-06 by PaulZC