sparkfun/SparkFun_ICM-20948_ArduinoLibrary #99: Expected yaw values when using DMP?

vgarciavazquez, 2022-08-03:
Hi Paul,

Maybe this is a very basic question about IMUs... What would be the expected yaw values when using Example6_DMP_Quat9_Orientation each time that _SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)_ is connected to the computer? 

I have marked a region on the floor to place _SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)_ at the same orientation. On the same day, each time that I connect the sensor to the computer I obtain different values  (for example, 24.5º, 57.6º, 40º...). I wait around 30 seconds to get the yaw value since during these first 30 seconds the data is not stable (I do not move the sensor during and after these 30 seconds). 

My settings are:
- _SparkFun_ICM-20948_ArduinoLibrary_, version 1.2.9.
-  _SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)_ connected to _Teensy LC_ using DMP and SPI (therefore I uncomment line 29 of ICM_20948_C.h and line 31 of Example6_DMP_Quat9_Orientation.ino, respectively).
- I read the quaternion data via the serial port and convert the quaternion to angles using Python.
- Sensor placed at the same position (thus same orientation) each time, connect the sensor to the computer, not moving the sensor during the experiment and waiting around 30 seconds to obtain the yaw angle.

I assume that in Example6_DMP_Quat9_Orientation.ino, the Digital Motion Processor (DMP) is using the data of the gyroscopes, accelerometers, and magnetometers to obtain the quaternion. If I use Example7_DMP_Quat6_EulerAngles.ino (I assume the DMP is using the data of the gyroscopes and accelerometers to obtain the angles) and wait around 30 seconds, each time that I connect the sensor to the computer yaw is around 1º.

Should I do anything with the sensor each time that I connect it to the computer during the first 30 seconds? Or maybe just to wait these seconds with the sensor fixed, get the yaw value ("initial" value) and then for the next measurements to obtain the relative orientation to that "initial" yaw value?

Thanks!

Veronica

PaulZC, 2022-08-04:
Hi Veronica,

The IMU contains nine sensors: three-axis accelerometer combined with three-axis gyros; and a separate three-axis magnetometer. These nine sensors all have different gains and offsets and the DMP - internally - needs to calibrate itself for those different values. It can only do that if you rotate the IMU around all three axes, so the DMP can measure the full range of data for each axis. Ideally, it would be possible to 'save' the calibration information, and 'load' or 'restore' it the next time you power the sensor on. In theory that is possible, but we have not worked out how to do it successfully. There are still many things about the DMP we do not understand. I would recommend doing the three axis rotation each time you power on.

The Quat6 example only uses the accelerometer and gyros. I believe the Euler angles are all relative to the orientation of the IMU at power on.

The Quat9 example uses the magnetometer and the Euler angles only become valid after performing the rotations. If you do not move the IMU, the magnetometer data does not change and so the IMU cannot know which way is North. If you power on, rotate the IMU around all three axes, and then put it down on your reference position (with the same orientation) then the angles should be consistent.

I hope this helps! If this answers your questions, please close this issue.
Paul

vgarciavazquez, 2022-08-05:
Thanks a lot Paul! I will try it and let you know.

vgarciavazquez, 2022-08-18:
Hi Paul,

I have used Example6_DMP_Quat9_Orientation for my tests. During around 35 seconds (around 2000 measurements), I have rotated the IMU around each axis each time that I power it on. Is it enough time for the IMU to calibrate itself? How long do you do it?

Then I fixed the sensor on the floor at the same position and now I obtain the same Yaw value (in my case, around 52º). I have placed a compass at that position and its value is 50º so the IMU is doing well. 

However, if I placed the IMU at that initial orientation but I just change the height (IMU parallel to the floor), yaw value changes to 33º. I do not understand why yaw changes... 

If I place the IMU at that initial orientation on the floor and I rotate the IMU 45º each time (so just changing yaw 8 times to do a loop of 360º, I have a millimetre paper on the floor with some marks so I can place the IMU at the right orientation), pitch and roll are around 0º (that's ok) but the difference between the new orientation and the previous orientation (yaw) goes from 36º to 61º when it should be around 45º each time.  

My last test was to rotate the IMU 90º in pitch (not changing yaw or roll angle) and the IMU gaves me a difference between the previous position and the new position of 89º in pitch (that's ok) but 9º in yaw and 30º in roll. I do not why roll is changing so much.

Paul, do you have experience these problems when you have tested this IMU? I assume that using Example7_DMP_Quat6_EulerAngles.ino would give me worse results since this code does not use the magnetometers to obtain the orientation. Maybe the errors that I see are usual errors for this low-cost IMU... 

By the way, looking at DMP sensor options (ICM_20948_DMP.h), Example6_DMP_Quat9_Orientation and Example9_DMP_MultipleSensors.ino, I think that we can get 9-axis quaternion + raw accelerometer measurements (we would like to also calculate the relative position doing double integration of accelerometer measurements) doing as follows. Am I right?

```
success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_ORIENTATION) == ICM_20948_Stat_Ok);
success &= (myICM.enableDMPSensor(INV_ICM20948_SENSOR_ACCELEROMETER) == ICM_20948_Stat_Ok);
...
success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Quat9, 0) == ICM_20948_Stat_Ok); // Set to the maximum
success &= (myICM.setDMPODRrate(DMP_ODR_Reg_Accel, 0) == ICM_20948_Stat_Ok); // Set to the maximum
...
if ((data.header & DMP_header_bitmap_Quat9) > 0)
{
      double q1 = ((double)data.Quat9.Data.Q1) / 1073741824.0; // Convert to double. Divide by 2^30
      double q2 = ((double)data.Quat9.Data.Q2) / 1073741824.0; // Convert to double. Divide by 2^30
      double q3 = ((double)data.Quat9.Data.Q3) / 1073741824.0; // Convert to double. Divide by 2^30
      SERIAL_PORT.print(F("Q1:"));
      SERIAL_PORT.print(q1, 3);
      SERIAL_PORT.print(F(" Q2:"));
      SERIAL_PORT.print(q2, 3);
      SERIAL_PORT.print(F(" Q3:"));
      SERIAL_PORT.print(q3, 3); 
}
if ((data.header & DMP_header_bitmap_Accel) > 0)
{
      float acc_x = (float)data.Raw_Accel.Data.X; // Extract the raw accelerometer data
      float acc_y = (float)data.Raw_Accel.Data.Y;
      float acc_z = (float)data.Raw_Accel.Data.Z;

      SERIAL_PORT.print(F("Accel: X:"));
      SERIAL_PORT.print(acc_x);
      SERIAL_PORT.print(F(" Y:"));
      SERIAL_PORT.print(acc_y);
      SERIAL_PORT.print(F(" Z:"));
      SERIAL_PORT.println(acc_z);
} 
```

Thanks a lot Paul!

Best

PaulZC, 2022-08-19:
Hi Veronica,

I am working on lots of different projects at the moment, but I will try and give you a little help.

Could there be something metallic (ferrous) in your floor? Screws, nails, steel beams, rebar? That would explain why the magnetometer reading is irregular or non-linear.

Changing the Pitch to 90 degrees produces a condition called "gimbal lock". The IMU can no longer measure roll. You may need to work directly in quaternions if you want to record the IMU orientation in all possible orientations.

Double-integrating the accelerometer data should be possible, but I have not tried that before.

Good luck!
Paul

vgarciavazquez, 2022-08-29:
Thanks a lot for your help, Paul.

Regarding something ferrous in my floor, I assume that that's not the problem since I remove everything close to the IMU and the tests with the compass and the IMU were done in the same conditions. I did not see that weird behaviour with the compass.

Right now, I have not the IMU but as soon as I have it I will repeat the test doing the rotation difference in quaternions. I will let you know my results,

Thanks a lot!

Verónica

HRiaz1, 2024-06-05:
Hi @vgarciavazquez, 

I am working with the Sparkfun OLA that uses ICM-20948. I have faced the same issue of my yaw value changing if i change the elevation of the IMU even after proper calibration. Were you able to find out the cause of this by any chance?

Regards

vgarciavazquez, 2024-06-06:
No, sorry :-(