sparkfun/SparkFun_ICM-20948_ArduinoLibrary #100: Extracting accuracy for accelerometer, gyroscope and magnetometer

tomwcourtney, 2022-08-23:
I am unable to obtain the accuracy of the raw sensors despite having them enabled. `data->header` never has the header2 bit set and hence the values of the  I have tried rolling back to the library version around when these answers were posted that list how to access these values but the result was the same. [Forum Post.](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/40#issuecomment-790877432)

Has anyone been able to get non-zero values for 

`data->Accel_Accuracy`
`data->Gyro_Accuracy`
`data->Compass_Accuracy`

using this library.

SamuelHawkins, 2023-02-01:
I can only speak with any confidence about the gyro values but I suspect this carries across to the acc and compass values.

The Gyro_accuracy is not the number which to scale the raw gyro values by, but is an expression of accuracy, between 0~3. The lowest accuracy is 0 and 3 is the highest. This is probably done to save space in data registers on the chip.

This code implements the gyro scaling:

`float ICM_20948::getGyrDPS(int16_t axis_val)
    {
      switch (agmt.fss.g)
      {
      case 0:
        return (((float)axis_val) / 131);
        break;
      case 1:
        return (((float)axis_val) / 65.5);
        break;
      case 2:
        return (((float)axis_val) / 32.8);
        break;
      case 3:
        return (((float)axis_val) / 16.4);
        break;
      default:
        return 0;
        break;
      }
    }`
In this code, agmt.fss.g is accessing the same data as your data->Gyro_Accuracy.
A Gyro_Accuracy value of 0 therefore corresponds to actual accuracy of 131, 1 to 65.5, 2 to 32.8, and 3 to 16.4. I assume 0 is the default scaling setting, hence your code returning 0 values.

Getting a value of 0 for the gyro accuracy is expected, though this isn't sufficient to say your implementation is working properly; that might require seeing the Gyro_Accuracy value change between 0 and 3 when loaded with other accuracy settings.

Hope this helps a little.

SamuelHawkins, 2023-03-13:
Hello again!

I've found, oddly, that sometimes on startup (and very occasionally during operation), the data.Gyro_Accuracy term will begin to report a different value (in my case, erroneously reporting "0" when the true value is "3"), which does not correspond to an actual change in the resolution of the reported gyro values. This results in the gyro values suddenly changing in scale!

I do not know how much this behavior is caused by my implementation of the gyro scaling vs. caused by the library.

This has been a frustrating problem and my solve for now is to hardcode the gyro_accuracy number within my gyro scaling code,
HOWEVER I think in light of this bug my previous answer is at best incomplete and at worst misinformation.
Use at your own risk and please remember this problem I faced!

tomwcourtney, 2023-04-03:
Thanks for your comments @SamuelHawkins. You wouldn't happen to know what the critical configuration is that has the DMP produce accuracy data for **icm_20948_DMP_data_t** data packets.

The IMU is working however whenever I look in **data.Quat9.Data.Accuracy** the accuracy is 0. I should not that I've ported the Cpp code to C for an STM32. 

**Example6_DMP_Quat9_Orientation.ino** produces an accuracy reading in the 0 - 800 range however I get only 0.

tomwcourtney, 2023-04-03:
Hi Sam, yes I am referencing the line you've linked and yes as you say it's different to the `data.Gyro_Accuracy` you mention. I am at this stage hoping to get the accuracy/certainty of the DMP rather than the individual sensors so that I might filter out data from when the IMU becomes 'uncertain'.

In my ported STM version of Example 6 I manage to access and convert the quarternions to Euler's no problem but unfortunately accuracy is always 0. No doubt I've made a mistake in my porting from cpp to c  but I am unsure where exactly haha. The configurations for the DMP firmware are fairly tedious. I am able to run an unsullied Example 6 on an Arduino with the IMU attached without issue.

texasfunambule, 2023-04-14:
Same issue, I've been wondering about the meaning of the accuracy number (data.Quat9.Data.Accuracy). the type is int16_t, so 16 bits. Did not have time to check but my guess is that bits 0 and 1 are for one modality (say accel), then bits 4 and 5 for the next (say gyro) and 8 and 9 for the last (say compass)? Not sure about the position of the bits used (seems reasonnable to space every 4 bits). One way to test would be to list all the observed values.