sparkfun/SparkFun_ICM-20948_ArduinoLibrary #101: How to read Quat6 data without using FIFO in ICM20948

LakkojiAshokKumar, 2022-09-16:
I tested ICM20948 with an example of DMP_Quat6_Euler angles with a delay 80ms after calculating roll, pitch and yaw data. For example sensor rotated to 90 degrees but the output yaw is slowly increased to 90 degrees instead of fast response. 

This is due to reading full FIFO data instead of latest one. How we achieve recent data without using FIFO.  

### My workbench
* Platform : Arduino IDE 
* Microcontroller: Arduino Nano BLE
* Sensor: ICM20948
* Wiring: 
                Arduino Nano              ICM20948
                       3.3V                            VIN
                       GND                          GND
                         A4                            SDA
                         A5                            SCL

* This setup is powered up by USB

PaulZC, 2022-09-16:
Hello @LakkojiAshokKumar ,

The DMP data can only be read through the FIFO. There is no way to bypass it - as far as I am aware. My suggestion is to read the data faster...

This piece of code reads the first value from the FIFO:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/examples/Arduino/Example6_DMP_Quat9_Orientation/Example6_DMP_Quat9_Orientation.ino#L185-L193

And this piece of code tells you if more data is available:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/examples/Arduino/Example6_DMP_Quat9_Orientation/Example6_DMP_Quat9_Orientation.ino#L241-L244

Example9 shows how to set different data rates:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/d5ae1eba1ecbf808fca9bff0b0b6dc4e571e947c/examples/Arduino/Example9_DMP_MultipleSensors/Example9_DMP_MultipleSensors.ino#L129-L139

Best wishes,
Paul

closed 2022-09-16 by PaulZC

LakkojiAshokKumar, 2022-09-19:
Hello @PaulZC,

Thanks for answering the query. 
In our application the microcontroller is busy with processing other data and it will take around 80ms to read data from IMU sensor. By considering the above constrain we added a delay of 80ms in the example of quat6 euler angle. 

We moved IMU sensor from 172 degrees to 123 degrees. As you can see attached video link below on rotating the IMU sensor the change in values form 172 to 123 is taking 2 seconds of delay.   

How to avoid this delay in getting data or how can we get most recent data from IMU by using FIFO.

We will try setting  different data rates and inform you the result. 

https://user-images.githubusercontent.com/103926084/190967892-9f403f9d-fc38-4f70-a861-fd761e69a832.mp4