sparkfun/SparkFun_ICM-20948_ArduinoLibrary #102: True compass heading

RossMelbourne, 2022-09-18:
I am using an ICM 20948 IMU and I am trying to get a magnetic compass heading (0 to 360 degrees) from the device.  Using the SparkFun example 'Example7_DMP_Quat6_EulerAngles.ino' I am able to get a good relative heading meaning it tracks heading well as you move in different directions.  How do I modify this example or one of the other examples to give a true compass heading?

PaulZC, 2022-09-20:
Hi Ross (@RossMelbourne ),

"Quat6" only uses the data from the accelerometers and gyros (6 degrees of freedom). It provides _relative_ orientation compared to the starting orientation. And will drift over time...

"Quat9" includes the magnetometer (9 degrees of freedom).

Quat9 is your friend...

Best wishes,
Paul

closed 2022-09-20 by PaulZC

RossMelbourne, 2022-09-22:
Thanks Paul.  I tried the sample Example6_DMP_Quat9_Orientation.ino which uses "Quat9".  I modified the code to convert from quat angles to euler angles using the calculations from the Example7_DMP_Quat6_EulerAngles.ino. I am getting very strange results from the yaw axis - it is definitely not acting like a compass or an IMU.  Here is the code (see 'Start of Changes''):

/****************************************************************
 * Example6_DMP_Quat9_Orientation.ino
 * ICM 20948 Arduino Library Demo
 * Initialize the DMP based on the TDK InvenSense ICM20948_eMD_nucleo_1.0 example-icm20948
 * Paul Clark, April 25th, 2021
 * Based on original code by:
 * Owen Lyke @ SparkFun Electronics
 * Original Creation Date: April 17 2019
 * 
 * ** This example is based on InvenSense's _confidential_ Application Note "Programming Sequence for DMP Hardware Functions".
 * ** We are grateful to InvenSense for sharing this with us.
 * 
 * ** Important note: by default the DMP functionality is disabled in the library
 * ** as the DMP firmware takes up 14301 Bytes of program memory.
 * ** To use the DMP, you will need to:
 * ** Edit ICM_20948_C.h
 * ** Uncomment line 29: #define ICM_20948_USE_DMP
 * ** Save changes
 * ** If you are using Windows, you can find ICM_20948_C.h in:
 * ** Documents\Arduino\libraries\SparkFun_ICM-20948_ArduinoLibrary\src\util
 *
 * Please see License.md for the license information.
 *
 * Distributed as-is; no warranty is given.
 ***************************************************************/

//#define QUAT_ANIMATION // Uncomment this line to output data in the correct format for ZaneL's Node.js Quaternion animation tool: https://github.com/ZaneL/quaternion_sensor_3d_nodejs

#include "ICM_20948.h" // Click here to get the library: http://librarymanager/All#SparkFun_ICM_20948_IMU

//#define USE_SPI       // Uncomment this to use SPI

#define SERIAL_PORT Serial

#define SPI_PORT SPI // Your desired SPI port.       Used only when "USE_SPI" is defined
#define CS_PIN 2     // Which pin you connect CS to. Used only when "USE_SPI" is defined

#define WIRE_PORT Wire // Your desired Wire port.      Used when "USE_SPI" is not defined
#define AD0_VAL 1      // The value of the last bit of the I2C address.                \
                       // On the SparkFun 9DoF IMU breakout the default is 1, and when \
                       // the ADR jumper is closed the value becomes 0

#ifdef USE_SPI
ICM_20948_SPI myICM; // If using SPI create an ICM_20948_SPI object
#else
ICM_20948_I2C myICM; // Otherwise create an ICM_20948_I2C object
#endif

void setup()
{

  SERIAL_PORT.begin(115200); // Start the serial console
#ifndef QUAT_ANIMATION
  SERIAL_PORT.println(F("ICM-20948 Example"));
#endif

  delay(100);

#ifndef QUAT_ANIMATION
  while (SERIAL_PORT.available()) // Make sure the serial RX buffer is empty
    SERIAL_PORT.read();

  SERIAL_PORT.println(F("Press any key to continue..."));

  while (!SERIAL_PORT.available()) // Wait for the user to press a key (send any serial character)
    ;
#endif

[... 176 lines cut ...]
      SERIAL_PORT.print(F(" Pitch:"));
      SERIAL_PORT.print(pitch, 1);
      SERIAL_PORT.print(F(" Yaw:"));
      SERIAL_PORT.println(yaw, 1);
#else
      // Output the Quaternion data in the format expected by ZaneL's Node.js Quaternion animation tool
      SERIAL_PORT.print(F("{\"quat_w\":"));
      SERIAL_PORT.print(q0, 3);
      SERIAL_PORT.print(F(", \"quat_x\":"));
      SERIAL_PORT.print(q1, 3);
      SERIAL_PORT.print(F(", \"quat_y\":"));
      SERIAL_PORT.print(q2, 3);
      SERIAL_PORT.print(F(", \"quat_z\":"));
      SERIAL_PORT.print(q3, 3);
      SERIAL_PORT.println(F("}"));
#endif
    // End of changes
    }
  }

  if (myICM.status != ICM_20948_Stat_FIFOMoreDataAvail) // If more data is available then we should read it right away - and not delay
  {
    delay(10);
  }
}

PaulZC, 2022-09-26:
Hi Ross (@RossMelbourne ),

I suspect this is probably a calibration issue? Please see this forum issue for more info.:

https://forum.sparkfun.com/viewtopic.php?p=235880#p235880

Best wishes,
Paul

RossMelbourne, 2022-09-26:
Hi Paul,
Thank you for the advice.  With a roving robot the IMU calibration would need to be saved off and reloaded upon power up.  I read one of your comments which said something to the effect that sample code or instructions for save and load of configurations did not exist?