sparkfun/SparkFun_ICM-20948_ArduinoLibrary #104: Yaw value delayed around 5 seconds by using quat6 euler angles example.

LakkojiAshokKumar, 2022-09-26:
When IMU sensor is moved to around z axis to 90 degrees output yaw value is delayed around 5 seconds when we introduced dynamic delay range of 50milli seconds to 2 seconds by using quat6 euler angles example.  How can we achieve the recent data without getting delay.

Platform : Arduino IDE
Microcontroller: Arduino Nano BLE
Sensor: ICM20948
Wiring:
Arduino Nano -------------> ICM20948
3.3V  ------------------------> VIN
GND ------------------------> GND
A4 --------------------------->SDA
A5 ---------------------------> SCL

This setup is powered up by USB

Changes we made in quat6 example is data rate set to 11Hz  and added dynamic delay in the range of 50ms - 2000ms.

`double t3 = +2.0 * (q0 * q3 + q1 * q2);
 double t4 = +1.0 - 2.0 * (q2sqr + q3 * q3);
 double yaw = atan2(t3, t4) * 180.0 / PI;
  yaw+=180;

  randomNumber=random(50,2000);
  SERIAL_PORT.print(F(" Yaw-> "));
  SERIAL_PORT.println(yaw, 2);
  delay(randomNumber);`

In the below picture we can see the behavior of the sensor. we rotated  the IMU around z-axis 90 degrees at 13:12:33 (timing) and desired yaw value got around 13:12:39. This is producing around 6 seconds of delay.  
[image]

Thanks.

PaulZC, 2022-09-26:
Hello @LakkojiAshokKumar ,

You are thinking about this the wrong way...

The DMP produces data at a constant rate. If you read the data (from the FIFO) at a slower rate, then data will form a "queue" (or "backlog") in the FIFO.

You need to make sure you are reading data from the FIFO at the same rate as the data is produced.

Best wishes,
Paul

closed 2022-09-26 by PaulZC

PaulZC, 2022-09-26:
Please also see my answer to your first question:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/101#issuecomment-1249458531

To empty the FIFO, keep reading data until ```(myICM.status != ICM_20948_Stat_FIFOMoreDataAvail)```

LakkojiAshokKumar, 2022-09-26:
Hello @PaulZC ,

Thanks for your response. 
In my application, we are using single core processor which is busy with multiple tasks which is taking different processing time. So, what we want is whenever I read data from IMU, the sensor need to provide the most recent data. Is there any way to achieve this? 

Thanks,
Ashok.

PaulZC, 2022-09-26:
Keep reading until the FIFO is empty, then use the last reading:

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/104#issuecomment-1257723728