sparkfun/SparkFun_ICM-20948_ArduinoLibrary #108: Biases values

apnogueira, 2022-12-14:
### Subject of the issue
I'm trying to read the bias values from the DMP using the example provided in this library. However I get these values:
[image]

I tried using a second ESP8266 with a different IMU and it had a similar result but with te gyro biases 0 and the CPass biases not zero.

Are these normal values? If so, what format are they on? If not, any idea why?
The only change I made to the example code was to comment lines 241-244 and add `EEPROM.begin(128);` since the ESP8266 EEPROM begin function does not return a bool.

### Your workbench
* What platform are you using?
* Esp8266 D1 
* How is the device wired to your platform?
* I2C

PaulZC, 2022-12-14:
Hi @apnogueira ,

Are you following these instructions carefully?

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/fd948fbc5dbe080eddafe88e3c1f7eadf5baff69/examples/Arduino/Example11_DMP_Bias_Save_Restore_ESP32/Example11_DMP_Bias_Save_Restore_ESP32.ino#L272-L275

The three full rotations (around X, Y and Z) and six hold-stationary's (+X, -X, +Y, -Y, +Z, -Z) are really important.

Best wishes,
Paul

PaulZC, 2022-12-14:
Slow, steady rotations are best. If you need more time, just edit the code accordingly...

Best,
Paul

apnogueira, 2022-12-15:
Hi @PaulZC 

Yes I did. I tried several times, but all with similar results. Are there any chances the esp8622 and esp32 EEPROM's are different to the point this is correct data but with wrong formatting?

PaulZC, 2022-12-15:
What do you see when the example restarts? Do the “restored” values look sensible?

PaulZC, 2023-05-25:
Closing. Please reopen if you need more help with this.

closed 2023-05-25 by PaulZC