sparkfun/SparkFun_ICM-20948_ArduinoLibrary #111: Arduino Nano Initialization of the sensor returned: Data Underflow

Ryam-20151-2, 2023-02-15:
### Subject of the issue
Using an Arduino Nano with the ICM-20948, trying example 1, everything compiles and writes to the board, but the error "Initialization of the sensor returned: Data Underflow" is repeated in the terminal.

### Your workbench
* Arduino IDE
* Arduino Nano V3
* protoboard soldered
* mini-b usb from PC

### Steps to reproduce
Set up Arduino IDE with example 1, set AD0_val = 0
compile and run

### Expected behaviour
It should run the example

### Actual behaviour
it repeats the error message

PaulZC, 2023-02-15:
Hi @Ryam-20151-2 ,

This is usually a sign that you are not using the correct I2C/SPI pins - or the correct I2C address.

Can you please post a photo of the connections between the Nano and the 20948 board. Or post a list of your connections (jumper wires). We need to know if you are using SPI or I2C, and which pins you are using.

Thank you,
Paul

Ryam-20151-2, 2023-02-16:
I see, I am using I2C, since I am using the Nano AD0_VAL should be 0 correct? I am using pins A4 and A5 on the nano. I attached both a schematic and photo of the board (sorry for poor quality, other boards are soldered in already). 
Any things I can test to see if its the wiring or my setup of the library that's incorrect?

Thank you
[image]
[image]

Ryam-20151-2, 2023-02-16:
haha, it seems setting AD0_VAL to 86 did the trick, thanks for the help

PaulZC, 2023-02-16:
Hi @Ryam-20151-2 ,

Thanks for the update. Glad that's working for you.

Just to be clear, I guess you set AD0_VAL to **0** - which sets the I2C address to 0x68 ?

Another Nano user reported that [same issue](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/issues/7#issuecomment-770191170). Strange. I don't really understand why you need to do that on the Nano. There must be something slightly strange about the pin voltages - the Nano is a 5V board - and/or the pull-up resistances on the Adafruit ICM20948 board. Anyway, it's working so I'm not going to worry about it too much...! ;-)

Best wishes,
Paul

closed 2023-02-16 by PaulZC