sparkfun/SparkFun_ICM-20948_ArduinoLibrary #114: mounting orientation?

texasfunambule, 2023-04-14:
The quat9 example seems to work fine (except maybe for compass alignement), yet calculation is all done assuming the pcb orientation is horizontal. In my project the pcb has to be vertical. I tried to change the matrices CPASS_MTX or B2S_MTX in ICM20948.cpp, but there was no way to get the orientation correct. Could not find documentation for B2S_MTX (I sassume "board 2 system"?), or indication if CPASS can be changed or is tuned for the position of the 2 dies carrying the accel/gyro and the compass. Should they be changed both simultaneously? Anybody knows?

ICM20948 on Arduino Due
https://shop.pimoroni.com/products/icm20948

Zzx-J, 2024-01-10:
Same problem here :(  Have you found any solution?

texasfunambule, 2024-01-11:
Hi,

sorry I can't remember why I gave up on the ICM-20948. Finally I switched back to BN085. Part of the problem was having the proper resistors in the right place along the I2C wires and between the wires and ground or Vcc. Maybe it would also make the ICM-20948 work fine. Another issue was that the Arduino Due has a design flaw and would simply not start at power on from cold (even without fix it starts properly when you turn it off for only a few seconds and then back on). Soldering a resistor in the right place solved that issue.

Hope this helps, best of luck with your design!

texasfunambule, 2024-01-11:
about orientation on ICM-20948: i finally figured out that b2s means "bring to see" which must be when the device is oriented so that the screen is facing towards user. I don't remember detail there again except I played with quaternion computation and geometry to get angles. BN085 was fine.

Zzx-J, 2024-01-14:
Hi, thanks for your reply, I decided to do the reorientation using quaternion multiplications too. Actually, I have used BNO085 before (in another project) but i was not happy with its drift so I would like to try ICM this time, what a coincidence.