sparkfun/SparkFun_ICM-20948_ArduinoLibrary #115: Porting to Linux Raspberry Pi 3

pbenaim, 2023-04-21:
Hello every body,

I would like use this very nice lib on one Raspberry Pi 3 in SPI in C++

On RPI3 we have wiringPi and wiringPiSPI to access to SPI bus, but I don't know really how convert this Arduino lib into RPI3.

Maybe somebody have already done that and can share this work.

thanks a lot for your help.

pbenaim, 2023-04-21:
I done :-)

I removed I2C, because is too long if you want use the DMP (ICM_20948_USE_DMP), it's my case.
So only in SPI.

Debugs funtions remplaced with printf...

If somebody, want the modifications, I can send it, by the way that you want. You must reopen the thread...

Thanks again to SparkFun, to publish this library.

closed 2023-04-21 by pbenaim

apnogueira, 2023-06-30:
Hi @pbenaim ! 
I am currently trying to run this library in a Raspberry Pi 4 using SPI. Exactly like you. Could you share your modifications?

pbenaim, 2023-06-30:
Yes, I will do this week-end

apnogueira, 2023-07-02:
Thanks in advance!

pbenaim, 2023-07-02:
Hello André, 

I am sorry and sad at the same time. 

The mother board of my computer died this week end 😢

I have ordered a new one, so no computer for the moment. 😭 I must wait for the delivery... 

So you must wait too... 

But, like I said on github if you replace Debugs funtions with printf, normally all will be fine. 

Maybe you need to remove all I2c references. 

Sorry again. 

I will be back to you when my computer will be alive again... 

Best regards 

Philippe 

Sent from Nine