sparkfun/SparkFun_ICM-20948_ArduinoLibrary #116: ICM interface with ESP32- HSPI pins Programming Issue

iharshadec, 2023-04-21:
I am using ESP32 HSPI pins which is 12,13 and 14 with ICM with Level Shifter to convert between 1.8v and 3.3v
I am not able to program my ESP32 after programming it once with ICM example code.

I am using Arduino + Windows environment for this.

PaulZC, 2023-04-21:
Hello Harshad (@iharshadec ),

Pin 12 is one of the ESP32 boot control (strapping) pins. Pulling pin 12 high will cause the bootloader to fail. Perhaps there is a pull-up resistor in the level shifter circuit?

https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html

You can use pin 12, but you need to ensure it is isolated until the boot is complete.

Best wishes,
Paul

closed 2023-04-21 by PaulZC

iharshadec, 2023-04-21:
I am not sure what is correct way to isolate Pin 12 until boot is done. Any suggestion?

PaulZC, 2023-04-21:
It depends on your project. We have used this circuit successfully to isolate pins 2, 12 and 15. When RESET is low, switch 4 is open, so PWREN is low. Because PWREN is low, ESP2, ESP12 and ESP15 are isolated. PWREN stays low until ESP32 is configured as an output and pulled high.

[image]

iharshadec, 2023-04-28:
Thank you @PaulZC