sparkfun/SparkFun_ICM-20948_ArduinoLibrary #118: USING DMP 9 axis DOF，angle cannot stop immediately

muniye, 2023-05-12:
### Subject of the issue
Describe your issue here.

### Your workbench
* What platform are you using?
* What version of the device are you using? Is there a firmware version?
* How is the device wired to your platform?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue to a gist.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

PaulZC, 2023-05-12:
Hello @muniye ,

I am sorry. I do not have an answer for you. There are many things about the DMP that we do not understand. If you look through the open and closed issues, you will long discussions about the behaviour and performance of the DMP.

Very best wishes,
Paul

closed 2023-05-12 by PaulZC

muniye, 2023-05-12:
OK,Thank you for your response.