sparkfun/SparkFun_ICM-20948_ArduinoLibrary #120: ICM-20948 Quaternion Output Example6 Not Working

hiraz01, 2023-05-24:
I am not able to get Example6_DMP_Quat9_Orientation working with the Sparkfun ICM-20948 board I purchased from DigiKey using the Arduino IDE. Nothing is getting displayed on the Serial port. Not even a startup message. I have successfully got Example1 working with an Arduino Pro Mini(3.3V) and a SAMD21 based Adafruit Bluefruit board (had to use this board since the DMP example sketch does not fit on Arduino Pro Mini).

I uncommented line 29 of the ICM_20948_C.h file but it did not work. I noticed when I go back and try to load the Example 1 sketch, the magnetometer readings remain the same and all zeros. If I go back and comment the define on line 29 the magnetometer works again in Example 1. The odd thing is that the serial output does not display anything for the Example6 code. 

I just installed the most recent library for the sensor using the Arduino IDE today (2023 May 24) and it is version 1.2.11.

Any ideas or help would be greatly appreciated.

hiraz01, 2023-05-25:
Just an update. I stripped out the extraneous code not directly related to the I2C and followed the same Serial startup procedure as Example 1 and it seems to be working now.

I still do not understand why the magnetometer data is not available in Example 1 code when the ICM_20948_C.h code is altered. Is that supposed to be the case?

PaulZC, 2023-05-25:
Hi @hiraz01 ,

Some Adafruit boards use ```SerialUSB``` as the default Serial port, instead of ```Serial```. You may need to change the ```#define SERIAL_PORT``` to match?

The standard examples 1-5 and the DMP examples 6-11 are not compatible with each other. The DMP completely changes the way the ICM 20948 operates. Examples 1-5 will only work if the DMP is disabled in the header file. Examples 6-11 will only work if the DMP is enabled in the header file.

Good luck with your project,
Paul

closed 2023-05-25 by PaulZC