sparkfun/SparkFun_ICM-20948_ArduinoLibrary #123: How to find Drift Angle?

fenilGhoghari, 2023-08-02:
Hi,

I am using ICM20498 with Sparfun library and now i am able to get appropriate coordinators of Accelerometer,  Gyroscope and Magnetometer one Level of calibration is okay and it's working.

Now i have tried roll,Pitch and Yaw example to find Drift Angle for my skateboard but i am not able to found.

So my issue with Drift Angle can you please give some idea of formula to drive Drift Angle using this all parameter.

I have try with the Vector Formation to drive Angle but i not able to drive this i juse able to drive only heading angle so,

Please suggest me some formula or tutorial for this.

Thanks

fenilGhoghari, 2023-08-03:
Hi, @PaulZC  can you please help me to find Drift Angle?

PaulZC, 2023-08-03:
Hi Fenil (@fenilGhoghari ),

I am sorry. I do not understand what "drift angle" is, for a skateboard. Can you send me a better description?

Best wishes,
Paul

fenilGhoghari, 2023-08-03:
Hello pailZC (@PaulZC )

Sorry for the short description actually i have one skateboard and i have placed ICM20948 sensor those provide me raw data like Magnetometer, Gyroscope and Accelerate of the skateboard.

So when my skateboard running in state forward at a time my drift should be zero or when my skateboard change the running direction at a time my drift should be calculated for better drift understand here i am attaching one article.

 [Drift Angle Article](https://www.sciencedirect.com/topics/engineering/drift-angle#:~:text=The%20drift%20angle%20at%20any,Maritime%20Engineering%20Reference%20Book%2C%202008)

Also you can consider Electric Scooter drift

[Video Refrence](https://www.youtube.com/watch?v=A0fgufnpYJs)

Here in video skateboard is moving another direction and change the instead direction so i want to maser this kind of the Drift.

Thanks a lot!!!

Regards,
Fenil

PaulZC, 2023-08-03:
Hi Fenil,

If I understand correctly, the drift angle is the difference between the direction the skateboard is pointing and the direction it is travelling in (its heading)?

I do not think you can measure that with an IMU. I think you will need to combine the IMU with GNSS (GPS)? Or use a dead-reckoning GNSS receiver - which includes an IMU? Or use GNSS Relative Positioning (Moving Base)? See https://content.u-blox.com/sites/default/files/ZED-F9P-MovingBase_AppNote_%28UBX-19009093%29.pdf

Best wishes,
Paul

PaulZC, 2024-01-29:
I think this issue is stale. Closing... Please re-open if you need more help with this.

closed 2024-01-29 by PaulZC