sparkfun/SparkFun_ICM-20948_ArduinoLibrary #127: calibration in imu

jasvinder1079, 2023-09-12:
Hi,
do we need to calibrate the sensor icm-20948 or DMP is automatically do this if we rotate the sensor in all direction during power on?
or we have to make some changes in the code for the calibration.

PaulZC, 2023-09-12:
Hi,

To calibrate the DMP: start the DMP; rotate the sensor around all three axes; and hold it stationary in all six orientations for a few seconds.

Best wishes,
Paul

closed 2023-09-12 by PaulZC

jasvinder1079, 2023-09-13:
OK DMP automatically calibrate the sensor when we turn it on and rotate it in all direction but Without DMP how to calibrate ICM-2948?

PaulZC, 2023-09-13:
You will need to do that using the normal IMU techniques. See:
https://thecavepearlproject.org/2015/05/22/calibrating-any-compass-or-accelerometer-for-arduino/

noega33, 2024-12-21:
Hello Paul,
There is something not clear in my mind.
I'm able to read from Bias registers and write to them data that I've stored on nRF52840  flash memory, but if I load my stored data at start-up, How can I stop the DMP from keeping updating Bias registers and change their content every time the ICM is powered?
Thank you

PaulZC, 2024-12-23:
Hi @noega33 ,

As far as I'm aware, the DMP continually refines the biases during operation. I believe it needs to, as the sensor biases will change - with temperature especially. I don't know if it is possible to disable this. If it is, I don't know how to do it. There are many things we don't understand about the DMP...

Best wishes,
Paul

noega33, 2025-01-12:
Hello everyone & Paul,
Is there any way to change the sensor orientation?
I've seen different ways of doing it (@Zane) using mount matrix (but I could not succeded in adapting it to the sparkfun library.
And the only mount matrix I've seen in the Sparkfun library (seems) only to address the compass.
Any idea 
Thank you
Cheers

PaulZC, 2025-01-13:
Hi @noega33 ,

Please tell us more about why you want to change the sensor orientation. Is this with the DMP - or without? The magnetometer (compass) is a separate die within the chip and its axes are different. When using the DMP, they need to be changed (using a matrix-multiply) to match the axes of the accelerometer+gyro die.

Best,
Paul

noega33, 2025-01-13:
Hi Paul,
more about why I want to change the orientation of the sensor axis.
My project is describe in: https://parformer.fr/
A) in summary, it is a system (patented in France) that currently measures the rotation speed of 4 main golfer body segment's (Pelvis, Thorax, arm and hand), so far this works great. I would like know to measure pelvis and thorax angles (attitude)
B) In my setting, sensors are installed with the Z axis horizontal, and I believe that this it will be better to put the Z axis vertical.
C) I'm using DMP and I've managed to build a sketch using a combination of : Example6_DMP_Quat9_Orientation and Example11_DMP_Bias_Save_Restore_ESP32 (adapted to nrf52840 board (Holyiot 1080 and using two instances of TwoWire)
D) So far, when I use the system (on a desk) with Z axis vertical pitch and roll angles work pretty well, Yaw takes a lot of time to stabilize (wich is a problem I also need to resolve). When I use the system on a normal operation (Z horizontal) angles are more unstable.

E) A separate question: Have you compared DMP self calibration accuracy against manual calibration ? (I'm relating to jremington comment: https://forum.arduino.cc/t/icm20948-help-needed/1110660 : "Whatever the DMP does for calibration, you can do better."

Thank you,
Cheers 
noega33

PaulZC, 2025-01-13:
Hi @noega33 ,

Maybe it is possible to do this with the DMP orientation parameters?

https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/blob/9caa80c79dd510039ec1680f8ac88a2b0f0289c2/src/util/ICM_20948_DMP.h#L285-L289

Regarding the calibration, you would need to ask Jim about this. He has much more experience with sensor calibration than I do.

Good luck with your project,
Paul

noega33, 2025-01-13:
Hi Paul
I dont have any clue how to use the DMP orientation parameters.
I have tried to port the following:
//*********************************************************
// Mounting matrix configuration applied for Accel, Gyro and Mag
static const float cfg_mounting_matrix[9] = {
 1.f, 0, 0,
 0, 1.f, 0,
 0, 0, 1.f
};

static void icm20948_apply_mounting_matrix(a)
{
  int ii;
  for (ii = 0; ii < INV_ICM20948_SENSOR_MAX; ii++)
  {
  inv_icm20948_set_matrix(&myICM[a], cfg_mounting_matrix, (inv_icm20948_sensor)ii);
  }
}

int inv_icm20948_set_matrix(struct inv_icm20948 * s, const float matrix[9], enum inv_icm20948_sensor sensor)
{
	int32_t mounting_mq30[9];
	int result = 0;
	int i;
	for(i = 0; i < 9; ++i)
		mounting_mq30[i] = (int32_t)(matrix[i] * (1 << 30));
	// Convert mounting matrix in char
	DmpDriver_convertion(s->mounting_matrix, mounting_mq30);
	//Apply new matrix
	inv_icm20948_set_chip_to_body_axis_quaternion(s, s->mounting_matrix, 0.0);
	}
	return result;
}
etc. etc...
//****************************************************
without any luck! (libraries are to different)
I'll keep digging :)
Cheers
noega33