sparkfun/SparkFun_ICM-20948_ArduinoLibrary #130: Board Manager 2.0.14 issue with HSPI and ICM-20948 Library

iharshadec, 2023-10-31:
### This library is not working with Latest Board Manager 2.0.14 with HSPI configuration on ESP32
I am using this library with HSPI configuration with ESP32 (PICO-D4) and I was on Board manager 1.0.6
When I updated my arduino and wanted to update to latest board manager 2.0.14 I came to know that my example code is getting stuck in the begin function itself.

So the same library which is working fine with board manager 1.0.6 is now not working with 2.0.14 ESP board manager.

### Your workbench
Windows 11
Arduino 1.8.19
ESP32 Board Manager - 2.0.14

```
 #include "main.h"
 
 SPIClass icmSPI(HSPI);
 //SPIClass icmSPI(VSPI);
 
 SPIClass vspi = NULL;
 SPIClass hspi = NULL;

 volatile ICMInfo_t xICMInfo;

 ICM_20948_SPI ICM;  // If using SPI create an ICM_20948_SPI object

 unsigned char uInitICM()
 {
	//vspi = SPIClass(VSPI);
	hspi = SPIClass(HSPI);

	//vspi.begin(18 /* SCK */, 19 /* MISO */, 23 /* MOSI */, 5 /* SS */);
	hspi.begin(14 /* SCK */, 12 /* MISO */, 13 /* MOSI */, ICM_CS_PIN /* SS */);

	//ICM.begin(ICM_CS_PIN, vspi);
	ICM.begin(ICM_CS_PIN, hspi);
        Serial.println("After begin");

	//ICM.begin(ICM_CS_PIN, vspi, 7000000);
	//ICM.begin(ICM_CS_PIN, hspi, 1000000);

	//ICM.begin(ICM_CS_PIN, VSPI);

	if (ICM.status != ICM_20948_Stat_Ok)
	{
		Serial.println("ICM initialization unsuccessful");
		Serial.println("Check ICM wiring or try cycling power");
		Serial.print("Status: ");
		Serial.println(ICM.statusString());
		return 0;
	}
	return 1;
  }

```

iharshadec, 2023-10-31:
In my case in this lib  This code is hanging at 

> 	_spi->beginTransaction(_spisettings);
> 	_spi->transfer(0x00);
> 	_spi->endTransaction();

	

```

ICM_20948_Status_e ICM_20948_SPI::begin(uint8_t csPin, SPIClass& spiPort, uint32_t SPIFreq)
{
	if (SPIFreq > 7000000)
		SPIFreq = 7000000;

	// Associate
	_spi = &spiPort;
	_spisettings = SPISettings(SPIFreq, ICM_20948_SPI_DEFAULT_ORDER, ICM_20948_SPI_DEFAULT_MODE);
	_cs = csPin;

	// Set pinmodes
	pinMode(_cs, OUTPUT);

	// Set pins to default positions
	digitalWrite(_cs, HIGH);

	// _spi->begin(); // Moved into user's sketch
	// 'Kickstart' the SPI hardware.
	_spi->beginTransaction(_spisettings);
	_spi->transfer(0x00);
	_spi->endTransaction();
	
	Serial.println("This Never Prints");

	// Set up the serif
	_serif.write = ICM_20948_write_SPI;
	_serif.read = ICM_20948_read_SPI;
	_serif.user = (void*)this; // refer to yourself in the user field

	// Link the serif
	_device._serif = &_serif;

	// Perform default startup
	status = startupDefault();
	if (status != ICM_20948_Stat_Ok)
	{
		return status;
	}

	return ICM_20948_Stat_Ok;
}
```

PaulZC, 2024-01-02:
Hi @iharshadec ,

I am sorry. I do not see how we can help you with this. It sounds as if something has changed in the ESP board package. Perhaps the pins are defined differently?

I am happy to leave this open in case someone else can help.

Best wishes,
Paul

iharshadec, 2024-01-02:
Sorry, I missed to update this but it's sorted out with this change.

Removed these lines
```
 SPIClass vspi = NULL;
 SPIClass hspi = NULL;

hspi = SPIClass(HSPI);
```
And this change
`icmSPI.begin(14 /* SCK */, 12 /* MISO */, 13 /* MOSI */, ICM_CS_PIN /* SS */);`

with this it's working fine with either version(Old and New)

closed 2024-01-02 by iharshadec